micropython/micropython #11698: gc.collect() does not work properly

wuhao19920619, 2023-06-04:
I am running MicroPython v1.20.0 on 2023-04-26; ESP32S3 module (spiram octal) with ESP32S3

```
import network
from machine import Pin, I2S
import urequests as requests
import struct
import gc
from debugfun import breakpoint

gc.enable()

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
if wlan.isconnected():
    wlan.disconnect()
wlan.connect('SSID', 'ssidssid')

sck_pin = Pin(12) 
ws_pin = Pin(13)
sdo_pin = Pin(14)

audio_out = I2S(0,
                sck=sck_pin, ws=ws_pin, sd=sdo_pin,
                mode=I2S.TX,
                bits=8,
                format=I2S.MONO,
                rate=44100,
                ibuf=8192)

buf_size = 4096
buf = memoryview(bytearray(buf_size))

while True:
    print("Free RAM:",gc.mem_free())
    text=input('>Say:\n')
    text = str(text).encode("utf-8")
    text = ''.join('{:02x}'.format(x) for x in text)    
    response = requests.get(url=("http://xxxxx/?d=" + text))
    data = response.content
    print("Data size:",len(data))
    i = data.find(b"data")+8
    while i < len(data):
        buf = data[i:i+buf_size] 
        i += buf_size
        audio_out.write(buf)
    del text
    del response
    del data
    del i
    gc.collect()
```

The Free RAM get lower after every loop. I have deleted all variables created in the loop so I think it is supposed to be a constant free memory.

dlech, 2023-06-04:
> buf = data[i:i+buf_size]

This replaces `buf` with a new object that isn't deleted, so is likely the source of the change you are seeing.

I assume you meant to do something like:

    buf[:] = data[i:i+buf_size]

wuhao19920619, 2023-06-04:
Yes you are right. Thank you and sorry for the stupid question. I will close it.

wuhao19920619, 2023-06-04:
Sorry just one more thing, In my original code, when the new `buf `object created, there should be no reference for the old `buf`. I think this should be removed from memory by calling gc.colect() right?

wuhao19920619, 2023-06-04:
I did test and find there is something weird with gc.collect(). Here is what I did in REPL:
```
>>> import gc
>>> gc.mem_free()
7921312
>>> gc.mem_free()
7916960
>>> gc.mem_free()
7912608
>>> gc.mem_free()
7908256
>>> aaa = bytearray(95648)
>>> gc.mem_free()
7421104
>>> print(aaa[-10:])
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
>>> gc.mem_free()
6646624
>>> gc.mem_free()
6259360
>>> gc.mem_free()
5872096
>>> gc.mem_free()
5484832
>>> gc.mem_free()
5097568
>>> gc.mem_free()
4710304
```
After the bytearray created, the memory consumed much more than the size of bytearray. Calling gc.mem_free() after that also used larger memory.

jimmo, 2023-06-05:
@wuhao19920619 In your original question it's worth noting that your use of `memoryview` doesn't do anything. By making @dlech's suggested change you at least re-use the buffer, but it's important to note that there's an additional copy made by the slice of data.

In your original code:

'buf = data[i:i+buf_size]'

is the same as

```
s = make a slice object
tmp = make a copy of data from i to i+buf_size
buf = tmp
```

if you want to minimise the number of copies (and allocations) you also need to make a memoryview out of `data`

i.e.

```
buf[:] = memoryview(data)[i:i+buf_size]
```

Unfortunately you still need to allocate the slice and memoryview instances, but this is very small compared to the buffers.

FYI you can use `binascii.hexlify` or even `str(x, 'utf-8').hex()` to convert a string to hex encoding.

Also instead of `del` it's always better to use a function instead, and let the variables go out of scope. That way you can be sure you don't miss anything. e.g.

```py
def run():
  print("Free RAM:",gc.mem_free())
  text=input('>Say:\n')
  text = str(text).encode("utf-8")
  text = ''.join('{:02x}'.format(x) for x in text)    
  response = requests.get(url=("http://xxxxx/?d=" + text))
  data = response.content
  print("Data size:",len(data))
  i = data.find(b"data")+8
  while i < len(data):
      buf[:] = memoryview(data)[i:i+buf_size] 
      i += buf_size
      audio_out.write(buf)

while True:
    run()
    gc.collect()
```

All that said, I agree with both your initial and follow up questions, your code shouldn't be causing the heap usage to grow.

There's also something very strange about the REPL log you posted, in particular:

```
>>> gc.mem_free()
7908256
>>> aaa = bytearray(95648)
>>> gc.mem_free()
7421104
```

This should allocate a couple hundred bytes for the REPL to compile+evaluate your code, plus a 95648 bytes for the buffer, plus 16 bytes for the bytearray object. But yours is allocating 487152 bytes.

For example, on a pyboard, I see

```
>>> gc.mem_free()
377296
>>> aaa = bytearray(95648)
>>> gc.mem_free()
281312
```

Also the 387264 bytes difference between each subsequent call to mem_free is very strange.

jimmo, 2023-06-05:
Wait what is `from debugfun import breakpoint`  ???

wuhao19920619, 2023-06-05:
It is just a simple tool based on` input() `and `exec()`
[https://github.com/urfdvw/CircuitPython-Debugger](url)

jonnor, 2024-07-13:
Testing on 1.22.2 on ESP32-S3 with no other code on the device, I cannot reproduce the issue of too much memory being committed.

```
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
2086448
>>> aaa = bytearray(95648)
>>> gc.mem_free()
1987856
```

Reduction in available memory by 98592 instead of 95648, seems OK.

@StreakingJerry is this still an issue with MicroPython 1.23 for you? If so, please provide a minimal example of how to reproduce.