micropython/micropython #11699: parsing strings defined as const ... mpy-cross gives SyntaxError

massimosala, 2023-06-04:
Hi

I have many strings depend on one, already defined as const.

Example:

```
PRESETS_DIR			= const("_presets/")
CONFIG_FILE			= const(PRESETS_DIR + const("config.json"))
```

mpy-cross aborts with

> SyntaxError: not a constant

It's not good practice to have to repeat the first constant string in all subsequent declarations.
By changing the first declaration, it seems to me inefficient to have to manually change all the others.

dpgeorge, 2023-06-05:
The parser does not do constant folding on strings (but it does do it on integers).  It would be possible to make it work for strings like in the example above.

jimmo, 2023-06-05:
> It's not good practice to have to repeat the first constant string in all subsequent declarations.
> By changing the first declaration, it seems to me inefficient to have to manually change all the others.

Not quite sure what you mean by this but it's important to note that just not using `const` is totally fine here.

i.e.

```py
PRESETS_DIR			= const("_presets/")
CONFIG_FILE			= PRESETS_DIR + "config.json"
```

Most of the benefit of `const` is about removing the overhead of doing a globals lookup when all you wanted was an integer. (i.e. the globals lookup dominates the overal time taken). Whereas in this case const isn't a particularly relevant optimisation.

i.e.

```
X = const(10)

...
  ...
    ...
      ... # inside some really tight loop
      y += X
```

there is a huge difference between having X be const or not.

massimosala, 2023-06-05:
> Not quite sure what you mean by this but it's important to note that just not using `const` is totally fine here.

Hi Jim, Damien

Take this silly example:

```
PRESETS_DIR			= const("_presets/")
CONFIG_FILE			= const(PRESETS_DIR + "config.json")
CONFIG2_FILE			= const(PRESETS_DIR + "config2.json")
CONFIG3_FILE			= const(PRESETS_DIR + "config3.json")
CONFIG4_FILE			= const(PRESETS_DIR + "config4.json")
```

If I change _PRESETS_DIR_, I have to update every declaration:

```
PRESETS_DIR			= const("_new_presets/")
CONFIG_FILE			= const("_new_presets/config.json")
CONFIG2_FILE			= const("_new_presets/config2.json")
CONFIG3_FILE			= const("_new_presets/config3.json")
CONFIG4_FILE			= const("_new_presets/config4.json")
```

Jim wrotes:

> Most of the benefit of const is about removing the overhead of doing a globals lookup when all you wanted was an integer

I don't understand: do you mean _const_ is useless on strings?

Damien wrotes:

> The parser does not do constant folding on strings (but it does do it on integers). It would be possible to make it work for strings like in the example above.

What do you think ? Is constant folding on strings beneficial or not?
I am focused on small targets, like esp8266.

Thanks for your fast replies.

jimmo, 2023-06-05:
> If I change _PRESETS_DIR_, I have to update every declaration:
> 
> ```
> PRESETS_DIR			= const("_new_presets/")
> CONFIG_FILE			= const("_new_presets/config.json")
> ```

No you don't have to do this. I gave an example above of how you can do it. It just means that all but `PRESETS_DIR` will be non-const. But that's fine.

> > Most of the benefit of const is about removing the overhead of doing a globals lookup when all you wanted was an integer
> 
> I don't understand: do you mean _const_ is useless on strings?

It's not that it's useless, it's that its benefit isn't as much as it is for integers.

Remember, const isn't about saving memory or de-duplicating the strings. That happens anyway. It just makes the code that uses this variable more efficient because instead of looking the variable up in the globals dict when accessing it, the code generator just loads the value directly. But it's still the same string object.

If we had constant folding on strings then that would be great. But it's not going to make a significant difference to your program. You should look at the bytecode disassembly to confirm.

massimosala, 2023-06-13:
Thanks Jim four your explanation.
@dpgeorge  Damien, seeing the other open issues, I think we can code without constant folding on strings.

These details are valuable to add to the docs, where only integers are mentioned:
https://docs.micropython.org/en/latest/reference/speed_python.html#the-const-declaration

Should I open a PR ?