micropython/micropython #11724: ure returns different results than using re (Regular expression)

maxi07, 2023-06-07:
Hello,
this is my function to validate for HEX colors.

```python
try:
    import ure as re
except ModuleNotFoundError:
    import re
def is_valid_color(color: str) -> bool:
    """Validate the color format using a regular expression"""
    try:
        color_pattern = r'^#[0-9a-fA-F]{6}$'
        return re.match(color_pattern, color) is not None
    except Exception as ex:
        print("Failed validating color:", ex)
        return False

print(is_valid_color("#00ff33"))
```

When this code is executed using micropython 1.20.0, the function returns False. When run with Python 3.11, the function returns True. I was expecting a True result, as the regular expression does match with the search string. This can be verified using [Regex101](https://regex101.com/r/c1od0z/1), which does find a match.

dpgeorge, 2023-06-08:
Indeed, counted repetitions are not supported.

We could eventually implement them.  But in the meantime it might be a good idea to raise an exception if they are used.

maxi07, 2023-06-08:
An exception would be great, as returning a not expected result can lead to confusion.

ricksorensen, 2023-06-11:
See open pull request [#8152](https://github.com/micropython/micropython/pull/8152)
Would it be straight forward to insert:
```
case '{":
    return NULL;
```
in `lib/re1.5/compilecode.c` which would create an exception in `extmod/modre.c`?  

There are some other python `re` parameters that are not yet supported in this micropython library, but I am afraid they require more sophisticated architecture decisions as well as implementations.