micropython/micropython #11733: Ports/RP2 - 'make submodules' hangs on mbedtls

mungewell, 2023-06-08:
Clean/Fresh install of MicroPython hangs when I attempt to:
```
simon@the-void:~/pico/micropython/ports/rp2$ make submodules
make -f ../../py/mkrules.mk GIT_SUBMODULES="lib/pico-sdk" submodules
make[1]: Entering directory '/home/simon/pico/micropython/ports/rp2'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/pico-sdk
Synchronizing submodule url for '../../lib/pico-sdk'
make[1]: Leaving directory '/home/simon/pico/micropython/ports/rp2'
GIT_SUBMODULES=$(cmake -B build-PICO/submodules -DECHO_SUBMODULES=1 -DMICROPY_BOARD=PICO -DMICROPY_BOARD_DIR=/home/simon/pico/micropython/ports/rp2/boards/PICO -S . 2>&1 | grep '^GIT_SUBMODULES=' | cut -d= -f2); \
make -f ../../py/mkrules.mk GIT_SUBMODULES="${GIT_SUBMODULES}" submodules
make[1]: Entering directory '/home/simon/pico/micropython/ports/rp2'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/mbedtls lib/tinyusb lib/micropython-lib
Synchronizing submodule url for '../../lib/mbedtls'
Synchronizing submodule url for '../../lib/micropython-lib'
Synchronizing submodule url for '../../lib/tinyusb'
Cloning into '/home/simon/pico/micropython/lib/mbedtls'...
```

The lib/mbedtls directory is created, but is empty.

MicroPython is at HEAD
```
commit d080d427ebcc263ac6ff792271cec1506e5c81fc (HEAD -> master, origin/master, origin/HEAD)
Author: Yaroslav Halchenko <debian@onerussian.com>
Date:   Fri Jun 2 13:21:39 2023 -0400

    top: Add "mis" to list of ignore words for codespell.
    
    Observed with codespell 2.2.5.dev57+gdc7e98d9:
    
        $ codespell
        ./ports/rp2/machine_uart.c:163: mis ==> miss, mist
        ./ports/rp2/machine_uart.c:168: mis ==> miss, mist
        2
```

Log says mbedtls was update a the start of May
```
commit 61b8e1b2d89175d288239376762184c78fdd5c9d
Author: Carlosgg <carlosgilglez@gmail.com>
Date:   Tue May 2 20:49:33 2023 +0100

    lib/mbedtls: Update to mbedtls v2.28.3.
    
    This is the latest release in the 2.28 long-time support branch.
    
    Signed-off-by: Carlos Gil <carlosgilglez@gmail.com>
```

mungewell, 2023-06-08:
Command is running
```
simon      28089  0.0  0.0  23364  3968 pts/0    S+   22:14   0:00 /usr/lib/git-core/git remote-https origin https://github.com/ARMmbed/mbedtls.git
```

However I attempt to visit that URL, github re-writes:
https://github.com/Mbed-TLS/mbedtls

Note added "-"... maybe important.

dpgeorge, 2023-06-09:
Thanks for the report.  I can see that the mbedtls repo has changed name, but GitHub does a redirect so it should be working.  It works on the CI.

Testing it locally it also works:
```
$ git clone https://github.com/micropython/micropython.git
Cloning into 'micropython'...
remote: Enumerating objects: 110435, done.
remote: Counting objects: 100% (372/372), done.
remote: Compressing objects: 100% (143/143), done.
remote: Total 110435 (delta 237), reused 320 (delta 229), pack-reused 110063
Receiving objects: 100% (110435/110435), 57.55 MiB | 11.69 MiB/s, done.
Resolving deltas: 100% (82000/82000), done.

$ cd micropython

$ git submodule update --init lib/mbedtls 
Submodule 'lib/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'lib/mbedtls'
Cloning into '/home/damien/tmp/micropython/lib/mbedtls'...
Submodule path 'lib/mbedtls': checked out '981743de6fcdbe672e482b6fd724d31d0a0d2476'

$ cat .git/modules/lib/mbedtls/config 
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
	worktree = ../../../../lib/mbedtls
[remote "origin"]
	url = https://github.com/ARMmbed/mbedtls.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "development"]
	remote = origin
	merge = refs/heads/development
```
It took about 17 seconds to initialise the mbedtls submodule.

mungewell, 2023-06-09:
Did a pull and tried again, and yes it's working for me. Possibly an issue with GitHub hosting last night...

Closing, Thanks.

closed 2023-06-09 by mungewell