micropython/micropython #11734: Inconsistent outputs for ".format(**locals())"  between CPython and MicroPython

xiaxinmeng, 2023-06-08:
".format(**locals())" causes a keyError on MicroPython, while it can work well on CPython.

MicroPython_exampe.py
```
def load3():
    a, b, c = 2, 3, 4 
    return "{a}{b}".format(**locals())

print(load3())

```

### The expected behavior:
It output '23' on CPython 3.9.0

### The actual output:
```
>> "micropython/ports/unix/build-standard/micropython" 'MicroPython_exampe.py'
Traceback (most recent call last):
  File "/home/xxm/Desktop/MicroPython_exampe.py", line 44, in <module>
  File "/home/xxm/Desktop/MicroPython_exampe.py", line 42, in load3
KeyError: a
```

### Test Environment:
MicroPython v1.20.0-162-gd080d427e on 2023-06-08; linux [GCC 7.5.0] version
Ubuntu 18.04
CPython 3.9.0

robert-hh, 2023-06-08:
See the documentation at https://docs.micropython.org/en/latest/reference/index.html
"MicroPython aims to implement the Python 3.4 standard (with selected features from later versions) "
and  https://docs.micropython.org/en/latest/genrst/index.html, MicroPython differences from CPython.

jimmo, 2023-06-09:
@xiaxinmeng In particular in this case the issue is that MicroPython does not support `locals()` in order to implement a very significant optimisation with the handling of local variables. This is not an issue with `str.format`.

See https://docs.micropython.org/en/latest/genrst/core_language.html#local-variables-aren-t-included-in-locals-result

closed 2023-06-09 by jimmo