micropython/micropython #11735: float('_') should report a ValueError

xiaxinmeng, 2023-06-08:
'__' cannot be converted to float. float('_') should report a ValueError

MicroPython_exampe.py
```
a = float('__')
print(a)
```

### The expected behavior
reports a ValueError as CPython 3.9.0
```
Traceback (most recent call last):
  File "/home/xxm/Desktop/MicroPython_exampe.py", line 49, in <module>
    a = float('__')
ValueError: could not convert string to float: '__'
```

### The actual output:
0.0

### Test Environment:

MicroPython v1.20.0-162-gd080d427e on 2023-06-08; linux [GCC 7.5.0] version
Ubuntu 18.04
CPython 3.9.0

dpgeorge, 2023-06-08:
Thanks for the bug report.

Although this is a difference to CPython, I'm not sure it's worth fixing (not worth the cost in additional code size).

massimosala, 2023-06-13:
> Thanks for the bug report.
> 
> Although this is a difference to CPython, I'm not sure it's worth fixing (not worth the cost in additional code size).

mmh but it can be a source of subtle bugs.

Many developers follow the "try/except" philosophy... and so '_' is accepted as a valid float?!
In cases like input validation (of users or untrusted sources), it's not good.
If indeed this verification increases the code that much, then this non-standard (and dubious) behavior needs to be documented.