micropython/micropython #11736: '/' as a parameter of a function gives a syntax error

xiaxinmeng, 2023-06-08:
If a python program takes / as the parameter of a function, micropython reports a SyntaxError. See the following case. This test case can work normally on CPython.

MicroPython_exampe.py
```
def run_in_thread(func, /, *args, **kwargs):
    pass
```

### The expected behavior:
CPython 3.9.0: work normally

### The actual output:
```
Traceback (most recent call last):
  File "/home/xxm/Desktop/MicroPython_exampe.py", line 55
SyntaxError: invalid syntax
```

### Test Environment:
Unix port,
MicroPython v1.20.0-162-gd080d427e on 2023-06-08; linux [GCC 7.5.0] version
Ubuntu 18.04
CPython 3.9.0

dpgeorge, 2023-06-08:
MicroPython aims to implement CPython 3.4, with selected features from later CPython versions.

Please see the documentation to understand what is and is not implemented: http://docs.micropython.org/en/latest/genrst/index.html

dpgeorge, 2023-06-08:
For this particular feature, see #8480.