micropython/micropython #11737: SyntaxError in f-string f"sys.path:{path!r}"

xiaxinmeng, 2023-06-08:
The following f-string causes an SyntaxError on MicroPython.  

```
path= "path"
print(f"sys.path:{path!r}")
```
### The expected output:
CPython 3.9.0: works normally

### The actual output
```
Traceback (most recent call last):
  File "/home/xxm/Desktop/MicroPython_exampe.py", line 69
SyntaxError: invalid syntax
```

Test Environment:

Unix port,
MicroPython v1.20.0-162-gd080d427e on 2023-06-08; linux [GCC 7.5.0] version
Ubuntu 18.04
CPython 3.9.0

robert-hh, 2023-06-08:
See the documentation at https://docs.micropython.org/en/latest/reference/index.html
"MicroPython aims to implement the Python 3.4 standard (with selected features from later versions) "
and  https://docs.micropython.org/en/latest/genrst/index.html, MicroPython differences from CPython.

jimmo, 2023-06-09:
This has a relatively straightforward fix, see discussion here https://github.com/micropython/micropython/pull/8711