micropython/micropython #11738: Support for Retrain the model on a device in MicroPython

alibukharai, 2023-06-08:
**Description:**
I am currently working on a project using ESP32 and ESP32-S3 boards and utilizing MicroPython.

I am exploring ways to push the capabilities of the ESP32 board to its limits. As part of my project, I would like to investigate the possibility of training and retraining TFLite models or **numpy based small Neural nets** models in real-time using the micropython.

Specifically, I am interested in the following capabilities:

**Training Neural Network models: Is it possible to utilize the ulab to retrain, train or tune  models directly on the ESP32 board? This would involve updating the model weights and biases during runtime based on real-time data. It would be beneficial for scenarios where the model needs to adapt and improve its performance continuously.**

I would appreciate any insights, guidance, or examples on how to achieve these capabilities using the any libaray. Additionally, if these features are not currently supported, I would be interested to know if there are any plans or possibilities of adding such functionality in future updates.

Thank you for your attention and support. I look forward to any feedback or suggestions regarding training and real-time retraining of models using the libraries in MicroPython.

Here are some useful relevant resources
1. [Tensorflow](https://www.tensorflow.org/lite/examples/on_device_training/overview#retrain_the_model_on_a_device)
2. [Article 1](https://arxiv.org/pdf/2206.15472.pdf)
3. [Article 2](https://arxiv.org/pdf/2203.01077.pdf)

jimmo, 2023-06-09:
This is not currently on MicroPython's roadmap and any support for this sort of thing is more likely to be provided by `ulab`. I see you've already asked the same question [there](https://github.com/v923z/micropython-ulab/issues/624) so I'll close this as a duplicate of that.

closed 2023-06-09 by jimmo

jonnor, 2024-09-19:
On-device learning is supported in emlearn-micropython using K Nearest Neighbors: https://github.com/emlearn/emlearn-micropython/