micropython/micropython #11743: [webassembly] import js; js.Class.new(...) not working

WebReflection, 2023-06-08:
/cc @dpgeorge

when running MicroPython as WASM I am incapable to create instances via:

```python
import js

def callback(resolve, reject):
  resolve('OK')

p = js.Promise.new(callback)
p.then(print)
```

While above example uses `Promise`, literally any other class/constructor is incapable of creating instances.

The good news is that I can somehow circumvent this limitation through this code:
```python
from js import Array, Promise, Reflect

def callback(resolve, reject):
  resolve('OK')

p = Reflect.construct(Promise, Array(callback))
p.then(print)
```

the bad news is that such way of creating instances is not super user friendly so I hope we can find a solution to the initial problem (if it's Proxy we're after, the `new` keyword might result into the working approach as returned value after invoke).

dpgeorge, 2023-06-09:
The issue is that MicroPython is not supporting arbitrary arguments to the `Obj.new(...)` call.  This calling pattern is a Pyodide idiom, right?

Anyway, it should be fixed now on my webassembly-add-modjs branch (which will eventually be merged into this repo).

WebReflection, 2023-06-09:
> This calling pattern is a Pyodide idiom, right?

well, `.new(...args)` is a Python thing but in JS you can `new Class(...args)` without any issue so I am not sure what you mean by pyodide idiom.

dpgeorge, 2023-06-09:
> in JS you can `new Class(...args)` without any issue so I am not sure what you mean by pyodide idiom

What I mean is that there is no such thing as `Class.new(...)` in JS.  In JS you do `new Class(...)` but that syntax won't work in Python.  So as far as I understand Pyodide invented the `Class.new(...)` syntax to be able to construct new classes in a simple way.  It could have been a different decision, eg `Class.make_new(...)`, and everything would still be able to work.  So this was a Pyodide invention, as far as I know.

WebReflection, 2023-06-09:
I see … yeah, I’ve never thought about it and indeed Python never needs new as it’s implicit … although I kinda like the new field as that’s not common at all in JS and it seems to be very semantic in foreign JS bindings.

dpgeorge, 2023-06-10:
> although I kinda like the new field as that’s not common at all in JS

But if a JS class did define a custom `new` field, then it would not be accessible from the Python side (because in the MicroPython bindings accessing `new` will go straight to `Reflect.construct`).  Is that an acceptable limitation?

An alternative could be `Class.__new__(...)`, but changing it would require changing Pyodide as well.

WebReflection, 2023-06-10:
> Is that an acceptable limitation?

yes, as any other alterntive would suffer the same issue, so `new` is the most sensible to me and, as JS developer, if I wanted to defined a static `new` field, I am 99.9% sure that'd be exactly the same thing there.

WebReflection, 2023-06-10:
P.S. as after thought, MicroPython *could* instead define `new` at the `Function` level so that'd be inherited and if any `new` is defined instead at class definition no surprise would happen ... JS example:

```js
Function.prototype.new = function (...args) {
  return new this(...args);
};

class UserLand {}

const ul = UserLand.new(1, 2, 3);
ul instanceof UserLand; // true

class Override {
  static new() { return [1, 2, 3]; }
}

Override.new(); // [1, 2, 3]
```

This way everyone should be happy and no surprises could happen.

WebReflection, 2023-06-10:
P.S.2 as I'm incapable to patch native Promise at runtime (see other related bug) I am confident me implementing that `new` in the native `Function.prototype` will fail the same way as unreachable, unavailable, once MicroPython is executed.

Once the other bug is fixed, I would be more than OK in having this `new` convention fixed from runtime users perspective.

dpgeorge, 2023-06-12:
I have kept the `Class.new(...)` idiom but made it so that it will not override any existing `.new()` method defined on the class (by first searching for a JavaScript attribute called "new", than falling back to the Reflect.construct call).

This issue should now be fully fixed.

closed 2023-06-12 by dpgeorge