micropython/micropython #11749: [webassembly] unable to use polyfilled globals

WebReflection, 2023-06-09:
/cc @dpgeorge

It is a very common practice to polyfill (fix incoming standards not fully implemented yet) globals but in MicroPython I need to do some fancy dance that replaces the whole global reference instead of just augmenting it and using it as desired.

Example:

```html
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <title>Pyodide Reflect</title>
  <script type="module" src="https://unpkg.com/@ungap/with-resolvers"></script>
  <script type="module">
    import {loadMicroPython} from '/micropython/micropython.mjs';
    const runtime = await loadMicroPython({url: '/micropython/micropython.wasm'});

    console.log(Promise.withResolvers); // function

    runtime.runPython(`
      from js import Promise

      deferred = Promise.withResolvers() # fail!
      deferred.promise.then(print)
      deferred.resolve('OK')
    `);
  </script>
</head>
</html>
```
Above page would throw an error stating that `Promise.withResolvers()` doesn't exist but the polyfill actually patches it and it works already in every browser/JS engine/runtime.

The polyfill is also [pretty trivial](https://github.com/ungap/with-resolvers/blob/main/index.js) so this MicroPython behavior surprised me ... on the other hand, if I do this instead everything is fine:

```html
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <title>Pyodide Reflect</title>
  <script type="module" src="https://unpkg.com/@ungap/with-resolvers"></script>
  <script type="module">
    import {loadMicroPython} from '/micropython/micropython.mjs';
    const runtime = await loadMicroPython({url: '/micropython/micropython.wasm'});

    // this makes no sense in the JS world
    const {withResolvers} = Promise;
    globalThis.Promise = class extends Promise {
        withResolvers() {
            return withResolvers.call(this);
        }
    }
    // but now the following code works

    runtime.runPython(`
      from js import Promise

      deferred = Promise.withResolvers()
      deferred.promise.then(print)
      deferred.resolve('OK')
    `);
  </script>
</head>
</html>
```

Is there any particular reason once fully replaced the code works while it wouldn't in the first place? Other runtimes seem to work just fine with the initial/first approach, thank you!

WebReflection, 2023-06-09:
P.S. I've managed to reduce/isolate the only thing needed to show the unexpected behavior:

```html
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <title>Pyodide Reflect</title>
  <script type="module" src="https://unpkg.com/@ungap/with-resolvers"></script>
  <script type="module">
    import {loadMicroPython} from '/micropython/micropython.mjs';
    const runtime = await loadMicroPython({url: '/micropython/micropython.wasm'});

    // why is this needed? comment this and see the error
    globalThis.Promise = class extends Promise {};

    runtime.runPython(`
      from js import Promise

      deferred = Promise.withResolvers()
      deferred.promise.then(print)
      deferred.resolve('OK')
    `);
  </script>
</head>
</html>
```

dpgeorge, 2023-06-10:
What is the difference between `globalThis.Promise` and `Promise`, when executing within the `<script type="module">` tag?

Said another way, what happens if you write this:
```js
globalThis.Promise = class extends globalThis.Promise {};
```
?

WebReflection, 2023-06-10:
the thing is that `Promise` is already a well known and defined global in JS and I don't want to override it, or better, there should be no need to do so. Latest example does likely the same thing as my example but neither should be needed, it's quite an obtrusive patch in the wild

dpgeorge, 2023-06-10:
> the thing is that `Promise` is already a well known and defined global in JS and I don't want to override it, or better, there should be no need to do so

Yes I understand.  I'm trying to understand the cause of the problem, which seems to be that there is a difference in JavaScript between writing `globalThis.Promise` and `Promise`.

When you write `Promise` in the global scope, what exactly is it referring to, and why isn't it the same as `globalThis.Promise`?

dpgeorge, 2023-06-10:
The way the MicroPython bindings to JS are extracting `Promise` is like this:
```js
  s = "Promise";
  ...
  if (s in globalThis) {
        let value = globalThis[s];
        if (typeof value == "function") {
            value = value.bind(globalThis);
        }
        <return proxy to value>;
  }
```
So it's extracting `globalThis["Promise"]` and then binding the result to `globalThis`.

Apparently that's not working well with polyfills.  Do you know how to do it correctly, so it does work with polyfills?

dpgeorge, 2023-06-12:
It looks like the problem is the `bind`: `globalThis.Promise.bind(globalThis)` is not a Promise object.

Need to work out how to differentiate between a function and a class.  Only the former should use bind.

dpgeorge, 2023-06-12:
OK, so the root cause of this is that `Promise` is a function, not a class.  And then the MicroPython JS bindings bind a function to its base/self.

Doing `globalThis.Promise = class extends Promise {};` turns the `Promise` entity into a class and fixes the issue, because the binding is no longer done.

dpgeorge, 2023-06-12:
I have fixed the issue by not calling `.bind()` when the base/self is `globalThis`.

The fix is on my `webassembly-add-modjs` branch.  Compiled version available at https://micropython.org/resources/wasm/micropython.mjs (also the .wasm is there).

closed 2023-06-15 by dpgeorge

WebReflection, 2023-06-15:
Apologies I wasn’t receiving notifications about these comments so I’ve seen this only now but thanks for fixing it, I’ll update to latest ASAP

WebReflection, 2023-06-15:
@dpgeorge slightly off topic ... is there any progress or something I can track around MicroPython over a CDN? your files ending up in our repo also pass through linters, spell checkers, and what's not and this way is a bit more painful than it should be to grab and push new versions, thanks!