micropython/micropython #11760: ports/unix: Apple M1 on macOS 13.4 error during compilation of py/emitglue.c with MICROPY_FORCE_32BIT=1

ned-pcs, 2023-06-11:
After reporting #11754 I tried to build using `MICROPY_FORCE_32BIT=1`, and got this error during compilation of `py/emitglue.c`:
```
CC ../../py/emitglue.c
clang -m32 -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS1=1 -DMICROPY_VFS_LFS2=1 -DMICROPY_PY_SSL=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/mbedtls_config.h\" -DMICROPY_SSL_MBEDTLS=1 -I../../lib/mbedtls/include -DMICROPY_PY_BTREE=1 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DLFS1_NO_MALLOC -DLFS1_NO_DEBUG -DLFS1_NO_WARN -DLFS1_NO_ERROR -DLFS1_NO_ASSERT -DLFS2_NO_MALLOC -DLFS2_NO_DEBUG -DLFS2_NO_WARN -DLFS2_NO_ERROR -DLFS2_NO_ASSERT -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I../../shared/readline -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -std=gnu99 -DUNIX -Os -DNDEBUG -fdata-sections -ffunction-sections -Ivariants/standard  -g -U _FORTIFY_SOURCE -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/ffi -DMICROPY_PY_FFI=1 -DMPZ_DIG_SIZE=16  -DMICROPY_ROM_TEXT_COMPRESSION=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMICROPY_MODULE_FROZEN_STR -c -MD -o build-standard/py/emitglue.o ../../py/emitglue.c || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
../../py/emitglue.c:125:9: error: operand must be a register in range [r0, r14] or apsr_nzcv
        "0:"
        ^
<inline asm>:1:16: note: instantiated into assembly here
        0:mrc p15, 0, r15, c7, c10, 3
                      ^
1 error generated.
```
I'm at git commit fd277704c458784a0e25ae87cf7190833661185d.
My system is a Mac Mini M1 (2020); `uname -a` reports:
```
Darwin Neds-mini 22.5.0 Darwin Kernel Version 22.5.0: Tue Apr 25 04:07:22 PDT 2023; root:xnu-8796.121.2~8/RELEASE_ARM64_T8103 arm64
```

dlech, 2023-06-11:
Apple no longer allows running 32-bit apps on macOS so it doesn't make sense to enable `MICROPY_FORCE_32BIT` when targeting macOS.

ned-pcs, 2023-06-11:
Ah yes. Perhaps this should be removed from the Makefile?

closed 2023-06-11 by ned-pcs

dlech, 2023-06-11:
It is still used on Linux with x86_64 architecture.

ned-pcs, 2023-06-12:
I was wondering whether maybe it should be removed from the macOS part of the Makefile.

In  `ports/unix/Makefile`, lines 98-103 contain:
```
ifeq ($(UNAME_S),Darwin)
ifeq ($(MICROPY_FORCE_32BIT),1)
CC = clang -m32
else
CC = clang
endif
```

dpgeorge, 2023-06-13:
It should still be possible to build in 32-bit mode on older macOS, right?

ned-pcs, 2023-06-13:
Pre-Catalina, yes.

dpgeorge, 2023-06-14:
Then I think the option to build in 32-bit mode should stay for now.

ESP32DE, 2023-06-30:
> Then I think the option to build in 32-bit mode should stay for now.

don't touch this 32 bit option :)