micropython/micropython #11797: rp2 PIO: two statemachines cannot share the same pins if they run on differenit PIOs

makismantas, 2023-06-16:
Hello
I spotted the following issue:
When trying to run two different programs in two statemachines sharing the same pin(s), they work fine ONLY if both statemachines belong to the same PIO (0 or 1), eg one in PIO(0) statemachine 0 and the other in PIO(0) statemachine 1.
The situation changes if the two programs are in different PIOs (one in 0 and the other in 1)

test code:

import time
import rp2
from machine import Pin

@rp2.asm_pio(set_init=(rp2.PIO.OUT_LOW))
def prog_0():
    set(pins, 1) [31]
    nop() [31]
    nop() [31]
    nop() [31]
    set(pins, 0) [31]
    nop() [31]
    nop() [31]
    nop() [31]
    irq(0)
    
@rp2.asm_pio(set_init=(rp2.PIO.OUT_LOW))
def prog_1():
    set(pins, 1) [31]
    nop() [31]
    nop() [31]
    nop() [31]
    nop() [31]
    nop() [31]
    nop() [31]
    set(pins, 0) [31]
    nop() [31]
    nop() [31]
    nop() [31]
    nop() [31]
    nop() [31]
    nop() [31]
    irq(1)
    
def irq_handler_0(pio):
    print('pio_0', pio.irq().flags())

def irq_handler_1(pio):
    print('pio_1', pio.irq().flags())
    
pio_0 = rp2.PIO(0)
pio_0.irq(irq_handler_0)
pio_1 = rp2.PIO(1)
pio_1.irq(irq_handler_1)

sm0 = pio_0.state_machine(0, prog_0, freq=2000, set_base=Pin(7))
sm1 = pio_1.state_machine(1, prog_1, freq=2000, set_base=Pin(7))

sm0.active(1)
sm1.active(1)
time.sleep(3)
sm0.active(0)
sm1.active(0)

In the above code, the interrupts are working fine (from both programs), so both programs are running, but the  led does not respond to instructions of the FIRST assigned statemachine (sm0)
The situation is the same, even if I never activate the second statemachine (sm1): the led does not blink at all (by sm0).
If sm1 is in the same PIO, that is if I change the line
sm1 = pio_1.state_machine(1, prog_1, freq=2000, set_base=Pin(7))
to
sm1 = pio_0.state_machine(1, prog_1, freq=2000, set_base=Pin(7))
everything works fine
The need of having programs running in different PIOs is that you can have more memory (32 instructions per PIO).
 
Thanks in advance

rkompass, 2023-06-17:
I'm used to the calling `sm = rp2.StateMachine(0, prog_1, freq=125_000_000, set_base=Pin(7))`
and there the second PIO block is addressed by ids 5..7. You could try that.
Did you try with different output-pins first?
From the data sheet:

3.2.5. Pin Mapping

PIO controls the output level and direction of up to 32 GPIOs, and can observe their input levels. On every system clock
cycle, each state machine may do none, one, or both of the following:
• Change the level or direction of some GPIOs via an OUT or SET instruction, or read some GPIOs via an IN instruction
• Change the level or direction of some GPIOs via a side-set operation
Each of these operations is on one of four contiguous ranges of GPIOs, with the base and count of each range
configured via each state machine’s PINCTRL register. There is a range for each of OUT, SET, IN and side-set operations.
Each range can cover any of the GPIOs accessible to a given PIO block (on RP2040 this is the 30 user GPIOs), and the
ranges can overlap.
For each individual GPIO output (level and direction separately), PIO considers all 8 writes that may have occurred on
that cycle, and applies the write from the highest-numbered state machine. If the same state machine performs a SET
/OUT and a side-set on the same GPIO simultaneously, the side-set is used. If no state machine writes to this GPIO
output, its value does not change from the previous cycle.
Generally each state machine’s outputs are mapped to a distinct group of GPIOs, implementing some peripheral
interface.

So in principle the setting from both machines should be taking effect.

makismantas, 2023-06-17:
@rkompass 
I have tried with 
sm_0 = rp2.StateMachine(0, prog_0, ...)
sm_1 = rp2.StateMachine(5, prog_1, ...)
It is exactly the same. Both machines work but the first assigned (sm_0 in this case) ignores pin settings (if they share pins)
Of course, as mentioned in example, they work ok if they belong to same PIO (0..3 or 4..7 both of them)

rkompass, 2023-06-17:
> RP2040 Datasheet
> 1.4. Pinout Reference 13
> Function
> 1.4.3. GPIO Functions
> Each individual GPIO pin can be connected to an internal peripheral via the GPIO functions defined below. Some internal
> peripheral connections appear in multiple places to allow some system level flexibility. SIO, PIO0 and PIO1 can connect
> to all GPIO pins and are controlled by software (or software controlled state machines) so can be used to implement
> many functions.
> Table 2. General
> Purpose Input/Output
> (GPIO) Bank 0
> 

.... you have to read there.

Every gpio pin can be controlled by one of up to nine sources:

1. SPI (certain channel), 
2. TX or RX UART (certain channel), 
3. SDA or SCL of  I2C (certain channel), 
4. PWM (certain channel), 
5. SIO,
6. PIO0, 
7. PIO1,
8. clock function (only certain pins), or  
9. USB (certain channel).

The general function like pin.value(1) is controlled by the SIO. So if initialized the pin is set to SIO. The other settings are also called alternative functions.
There is a control register for each pin, where bits 0-4 select that function.

If I do in my program:
```
led = Pin(25)
...
sm = rp2.StateMachine(0, blink, freq=2000, set_base=led)
print(led)
```
then the output is `Pin(GPIO25, mode=ALT, pull=PULL_DOWN, alt=PIO0)`, indicating that the GPIO pin was
set to the alternative function **PIO0**.
This is done automatically.
Now, when you setup the second state machine, the alternative function will automatically be set to **PIO1**.
So this is a limitation of the hardware.

makismantas, 2023-06-17:
@rkompass 
Ok, this clears thing up. 
Understood,  thanks a lot! :)

closed 2023-06-17 by makismantas