micropython/micropython #11805: Cannot catch exception SyntaxError: invalid micropython decorator

rkompass, 2023-06-17:
With:
```
try:
    @micropython.blah
    def f_blah(x):
        return x+1
    if f_blah(3) == 4:
        print("working")
    else:
        print("not working")
except SyntaxError:
    print("not implemented")
```
the program halts with
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
SyntaxError: invalid micropython decorator
```
and this is not caught in the exception.

Unlike with CPython, where the exception is a NameError, that is caught.

If this could be caught in Micropython it would contribute to the writing of portable code.
Not all platforms have e.g. @micropython.native.

peterhinch, 2023-06-18:
Trapping `SyntaxError` is not ideal. Another way of querying whether a platform supports native or viper would be good, but I'm not aware of one.

jimmo, 2023-06-19:
Unfortunately I don't think this can be fixed in MicroPython in a practical way.

In CPython the decorator is evaluated when the code is executed. In this case it's not a real decorator, rather an instruction to the compiler. We could raise NameError sure but the code is not actually executing, so the exception handler would have to be around the `import` or the `eval` that was loading this code.

To solve this would require emitting the function as bytecode, and then making it into a "real" decorator which when the code is executed translated the bytecode into native code. (This might work for native and maybe asm_thumb although would be a complete rewrite, but wouldn't work for viper which needs the additional type information).

peterhinch, 2023-06-19:
@jimmo Is it not worth considering some other way to query whether a platform supports native/viper code, perhaps via the `sys` module?

At the moment I have to write in the docs "if this code won't run on your hardware, comment out the decorators". Which is sub-optimal.

rkompass, 2023-06-19:
> In this case it's not a real decorator, rather an instruction to the compiler. 
If in MPs case it's an instruction to the compiler, can't the compiler check for an `except:` before raising the `SyntaxError` ?

rkompass, 2023-06-19:
Thank you for the answer.

If that 
`+        return false;`
means the code does not get an exception and has a bytecode function instead:

That would be counter-intuitive.
For such behavior an exception block with the undecorated function should be supplied, if desired.

Can the compiler compile a NamerError instead?

Considering that the above change is very simple, and we could live with it, is there a natural way to test whether a function is bytecode or viper / native / asm.thumb ?

Thanks for the pointer to sys.implementation._mpy
Is this documented?
Are there also bits for viper and asm.thumb?

peterhinch, 2023-06-20:
> Note that sys.implementation._mpy >> 10 tells you the native architecture of the system, and is 0 if native is not supported.

That fits my needs.

> Are there also bits for viper and asm.thumb?

If Native is unsupported then Viper will be too as it emits native code. I assume that if Native is supported, Viper will also be, but confirmation would be good.

As for asm.thumb you need to check the arch. STM supports ArmV7, RP2 supports ArmV6  which is a subset.

rkompass, 2023-06-20:
Just to complete my thoughts:

If the compiler might compile the function to bytecode like @dpgeorge indicated it is possible,
 and also might include a `raise NameError("unsupported micropython decorator")` at the function start, then it would be perfect. At least it seems to me so:
You have the exception raised, and in case you catch it, you have the bytecode function.

But of course, I'm also happy with the  `sys.implementation._mpy`.

rkompass, 2024-01-07:
I have found a simple workaround:
You have to put the definition with the questionable decorator in a separate file and try to import it:
E.g a file `imp_blah.py` containing
```
@micropython.blah
def f_blah(x):
    return x+1
```
with try/except at import like:
```
try:
    from imp_blah import f_blah
    if f_blah(3) == 4:
        print("working")
    else:
        print("not working")
except SyntaxError:
    print("decorator not implemented")
```
is catched like expected and this may be used for testing the builtin decorators (and separate for example asm_xtensa and asm_thumb).

I consider this issue as solved and close it.

closed 2024-01-07 by rkompass

peterhinch, 2024-01-07:
Very neat.

It would be good if there were a way conditionally to invoke a decorator. This is particularly relevant for `native` where a function will run in the absence of the decorator. Consider this, where `foo` might be substantial:
```python
if native_is_supported:  # Determined as above
    @micropython.native  # Nope. Syntax error.
def foo():
    # Long function definition
```

stinos, 2024-01-07:
It would be rather trivial to make `micopython.native` become a no-op in case where it's not actually available, only drawback being that people might think things are running native even though they aren't.

rkompass, 2024-01-07:
@peterhinch 
I have found that you may catch the compilation of a code string. And that the code string for `compile` may optionally contain the decorator:
```
_sdef = '''
def mywait(x):
    while(True):
        x -= 1
        if x < 0:
            return
'''

try:
    exec(compile("@micropython.native\n"+_sdef, "_", 'exec'))
except:
    exec(compile(_sdef, "_", 'exec'))
```
This may help to avoid the many single files containing various decorated versions of the functions of interest.