micropython/micropython #11827: integrate display driver ILI9342C for esp32

nixilin, 2023-06-20:
I'm using ESP32-S3-BOX and it contains a LCD [official link](https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box/hardware_overview_for_box.md), but I can not use it while I'm plan to use micropython because there's no driver.

* For features for existing ports (e.g. new peripherals or microcontroller features), please describe which port(s) it applies too, and whether this is could be an extension to the machine API or a port-specific module?
ports/esp32

* For drivers (e.g. for external hardware), please link to datasheets and/or existing drivers from other sources.
https://github.com/russhughes/ili9342c_mpy.git 
Looks OK

aguaviva, 2023-10-29:
I've been using that module for a while and it works very well. 

Any chance this can be integrated into Micropython?

robert-hh, 2023-10-29:
> Any chance this can be integrated into Micropython?

What do you mean with "integrated": a) Add it to Micropython-lib, or b) freeze it by default into a firmware? a) may happen, if someone make a PR for it. b) is unlikely to happen. But you can always install the driver yourself or build your own firmware with the driver integrated into it.

aguaviva, 2023-10-29:
>a) may happen, if someone make a PR for it.

What are the requirements to have this driver added into Micropython-lib?

andrewleech, 2023-10-29:
micropython-lib is all pure python, whereas the fast driver linked from @russhughes is a C driver that needs to be compiled in.

There are pure python ones that would be appropriate to submit, they're just not quite as fast.

For these they would need to be in the appropriate drivers folder and have a compatible licence. If you're submitting code originally written by someone else you should ideally get their permission / blessing, or at least document where it came from and meet any attribution requirements of the associated licence.
Otherwise it's just add a matching manifest and probably readme, maybe an example.

peterhinch, 2023-10-29:
For a pure Python driver there is a fair chance that this one ([docs](https://github.com/peterhinch/micropython-nano-gui/blob/master/DRIVERS.md#34-driver-for-ili94xx), [code](https://github.com/peterhinch/micropython-nano-gui/blob/master/drivers/ili94xx/ili9486.py)) will work. It is fairly generic and has been tested with ILI9341, ILI9488, HX8357D and ILI9486.

If you try it please report back.