micropython/micropython #11834: esp32: Slow I2C on S2 after bus failures.

jimmo, 2023-06-21:
After a certain number (usually <10) of failures (e.g. ENODEV due to non-existent address), the clock frequency seems to drop to 1/16 of the requested value.

Unfortunately, this is exactly how `i2c.scan()` is implemented, so the behavior I see is that everything is fine if I _don't_ do a scan first.

`i2c_get_period` continues to report the original frequency. If add a call `i2c_param_config` to re-set the frequency, then all is fine. 

Can replicate on IDF 4.4 and 5.0, and only on S2.

kdschlosser, 2024-03-26:
I have a similar issue running the S3. the I2C doesn't work. If I flash a simple I2C sketch using the Arduino IDE the I2C works fine. I2C doesn't work on MicroPython 1.22.2 using the ESP32-S3

bland328, 2024-03-26:
I'm running into this problem on an ESP32-S3 dev board with built-in display and I2C-based GT911 capacitive touch digitizer.

MicroPython can't find the GT911 when scanning, and hangs for nearly two minutes attempting it before returning an empty list.

As @kdschlosser mentioned, a simple Arduino IDE sketch discovers the device without issue.

kdschlosser, 2024-03-26:
I believe I found the source of the problem...

```
i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = self->sda,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,   <----- HERE
        .scl_io_num = self->scl,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,   <----- AND HERE
        .master.clk_speed = freq,
    };
```

This is an assumption that should not be made. Since a `machine.Pin` object is what gets passed when creating an I2C instance it should be possible to check the pins to see if the pins are being pulled up. If they are then set the pull-up flags and of not don't set the pull-up flags.

What I am working with is a display board with an ESP32-S3 already attached. External pull-ups have been placed on the I2C lines and the addition of the internal pull-ups is likely causing the value to be out of the ideal range for the frequency that is being used. In my use case 4.7K ohm external resistors are being used.

In the ESP-IDF docs there is this blurb...

>Generally speaking, the higher frequency is selected, the smaller resistor should be used (but not less than 1KOhms). This is because high resistor will decline the current, which will lengthen the rising time and reduce the frequency. Usually, range 2KOhms to 5KOhms is what we recommend, but users also might need to make some adjustment depends on their reality.

I am not able to locate what the actual pull-up values are for the ESP32-S3 chip for I2C but I hardly doubt it is the 45K ohm value that is seen for general GPIO use. That value wouldn't be in line with the 2K to 5K ohms they state. The frequency I am running the bus at is 400Khz and the 4.7kOhm external resistors that are used put it right in the window that is appropriate for the speed. If I add a second pullup of say 5K ohm the resistance now becomes 2422ohms which is going to be far too low for a 400Khz transfer speed.

Whether or not this is actually causing the problem it should be fixed because it is not right.

smhodge42, 2024-03-26:
Are you creating a serial port object before creating the I2C one?  See my post #14159 on March 22. The symptom I had was the same, on the same chip.  (Peter Hinch suggested trying a software I2C but I haven't got around to doing that yet.)

kdschlosser, 2024-03-26:
No serial ports are being used other than the repl.

Unless the RGB dispay driver that is built into the ESP-IDF is causing some kind of an issue with the I2C. I would find that quite strange if that is what is causing the problem. 

I can try moving the creation of the I2C to before the display driver is initialized and see if that corrects the problem.

robert-hh, 2024-03-26:
> If I add a second pullup of say 5K ohm the resistance now becomes 2422ohms which is going to be far too low for a 400Khz transfer speed.

Usually a lower pull-up resistor value does not hurt, unless it gets too low for the signal to achieve a low level. For an ESP32 that would be around 100 Ohm. And you should assume that the internal pull-up values are always the same irrespective of how the Pin is used. Because that is part of the GPIO circuitry, which is used by the I2C hardware.

kdschlosser, 2024-03-26:
So now here is my question. why would they provide a pull-up value that is  way outside of what they say should be used? that doesn't make any sense.

I know that I2C can be very touchy when it comes to pull-up resistors and the values. when communicating using 400Khz and max on the I2C bus is 4Mhz to calculate a liner mapping of speed to resistance is pretty straight forward to do. Now I know that capacitance comes into play,  but for sake of example and this is the reason why a value of 4.7K is used for 400Khz

```python
def remap(value, old_min, old_max, new_min, new_max):
    old_range = old_max - old_min
    new_range = new_max - new_min
    new_value = (((value - old_min) * new_range) / old_range) + new_min
    return new_value

# the  5000 and 2000 is the ideal range that is stated in the ESP-IDF documentation
print(remap(400000.0, 0.0, 4000000.0, 5000.0, 2000.0))
```
and the result from that is exactly 4700 or 4.7K of resistance. That's not a coincidence.

If running the the function above except for resistance and using the 100 value you are stating..
```
print(remap(100, 50000.0, 2000.0, 0.0, 40000000.0))
```
Results in a speed of 41583333 which is 10 times higher than the target speed.

If the internal pull-up is 5K along with the 4.7K that is externally attached then we get a resistance of 2422ohms and that resistance is good for 3.9Mhz which is substantially higher than what is needed. 

> The value of the pullup resistor is an important design consideration for I2C systems as an incorrect value can lead to signal loss.

https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.ti.com/lit/pdf/slva689&ved=2ahUKEwiavqvV-ZKFAxV9IzQIHaBDDrcQFnoECBAQAw&usg=AOvVaw1g0JcOo6ZnTZvXfdvGzjSx

Seeing as how a hardware manufacturer like TI states it will lead to signal loss and what is being seen here pretty much falls into that category It may be something that could cause the problem wouldn't you say??

I have no clue what resistance is being used when enabling the pull-ups for I2C. I have not been able to locate any information on that. There are 2 assumptions that can be made, the first is that it would be 45K ohm which is what a GPIO pin's pull-up is. Then you have to think of why even provide the ability to use the internal pull-up if the resistance value isn't even in the ballpark of the 2K to 5K range. It wouldn't make any sense to provide that ability if that is the case.  The second assumption is that the pull-up value is actually in the 2K to 5K range and if that is the case and an external pull-up has been attached that is going to really throw off the needed resistance value which could lead to signal loss.

There should be a way to check a machine.Pin instance from C code to see if the pull-up has been enabled or not. If it hasn't been enabled then don't enable the pull-up in the config for the I2C.

Or add an argument so the user can choose if it is to be enabled or not. All I can tell you is that compiling on Arduino it works without any issues. I believe the Arduino IDE uses the ESP-IDF to handle I2C. So there is something going on in the code for I2C in MicroPython that is causing the problem. Its not a bug in the ESP-IDF and I am using the same pin numbers so it is not a problem in my code. That leaves only one place where the problem can be caused. 

I have looked over the code in MicroPython and that is the only thing that I have seen that could possibly cause the problem.

kdschlosser, 2024-03-26:
This is the full note about pull-up resistors in the ESP-IDF documentation  ESP-IDF

>The clock frequency of SCL will be influenced by the pull-up resistors and wire capacitance (or might slave capacitance) together. Therefore, users need to choose correct pull-up resistors by themselves to make the frequency accurate. It is recommended by I2C protocol that the pull-up resistors commonly range from 1KOhms to 10KOhms, but different frequencies need different resistors.
> 
> Generally speaking, the higher frequency is selected, the smaller resistor should be used (but not less than 1KOhms). This is because high resistor will decline the current, which will lengthen the rising time and reduce the frequency. Usually, range 2KOhms to 5KOhms is what we recommend, but users also might need to make some adjustment depends on their reality.

So the resistors will alter the frequency. so depending on how sensitive the attached device is the internal pull-up combined with an external one can cause issues. if the external one is set to where it should be set.

This is also wrong

```c
int timeout = I2C_SCLK_FREQ / 1000000 * timeout_us;
i2c_set_timeout(self->port, (timeout > I2C_LL_MAX_TIMEOUT) ? I2C_LL_MAX_TIMEOUT : timeout);
```

as seen in

https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32s3/api-reference/peripherals/i2c.html#id6

>All these parameters can be changed to user-defined values by calling dedicated functions given in the table below. Please note that the timing values are defined in APB clock cycles. The frequency of APB is specified in [I2C_APB_CLK_FREQ](https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32s3/api-reference/peripherals/i2c.html#c.I2C_APB_CLK_FREQ).

`I2C_SCLK_FREQ` is a macro that is set in the I2C source file for the ESP32. It reads

```c
#if CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32S3
#define I2C_SCLK_FREQ XTAL_CLK_FREQ
#elif CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2
#define I2C_SCLK_FREQ APB_CLK_FREQ
#else
#error "unsupported I2C for ESP32 SoC variant"
#endif
```

For the ESP32-S3 `XTAL_CLK_FREQ` is define as 
```c
XTAL_CLK_FREQ                               (40*1000000)
```

So you can see the set_timeout function is being done incorrectly.

the functions that are listed that use the APB frequency are

  * i2c_set_period()
  * i2c_set_start_timing()
  * i2c_set_stop_timing()
  * i2c_set_data_timing()
  * i2c_set_timeout()
  * i2c_set_data_mode()

I found a note on the internal pull-ups. While it doesn't give you a value it tells you that turning them on is pointless to do.

>ESP32-S3’s internal pull-ups are in the range of tens of kOhm, which is, in most cases, insufficient for use as I2C pull-ups. Users are advised to use external pull-ups with values described in the I2C specification.

It is the paragraph above where this link brings you to.

https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32s3/api-reference/peripherals/i2c.html#error-handling

I did also want to mention the entire process of reading from and writing to the I2C bus is different than what is being done in MicroPython.

It looks like I2C is going to need an overhaul in order for it to work properly.

robert-hh, 2024-03-27:
Looking into the ESP-IDF, it is not required to set the detailed timing using the above API. Upon installing the driver, the IDF sets reasonable timing values derived from the requested frequency. Only the timeout set by machine_i2c.c seems a little bit too long. The default chosen by the IDF is 10 clock periods.

Going back into the analog world. A I2C clock cycle T has three phases, a) low time Tl, b) low-to-high transition time Tt and c) high time Th. The add to  a) can be reliably set. It just depends on the clock. c) is the minimum time Thminthat the signal must be high. That's a hardware property. b) depends on the R and C values of the board. If you design a specific board, the capacitance C of the circuitry is know, so you can select a pull-up resistor matching the high time. There are I2C hardware designs where the bus is driven actively high to a few clock cycles to speed up the transition. And than there are differences between hardware designs about how they deal with phase b).
- some hardware have a fixed period. So the parameters for time b) must be smaller than T - Tl -Tlmin, and R (and C) must be properly chosen. If for instance R is too high, the signal will not go high within a period T and communication fails. The RP2040 works that way, and maybe the ESP32 as well. For this design, the bus frequency does not vary when you change the R*C value.
- other hardware does not start phase c) until phase b) is finished and the signal is high. That ensures proper operation, only the time T varies with the value of R*C. The SAMD MCUs use that approach.

In any case, lowering the value of the pull-up resistor does not hurt, unless you exceed the current driving capability of the GPIO pin.
And in any case, if the bus never get's high for whatever reason, the bus stops. That's when timeout should trigger. Therefore even a soft pull-up is useful to cover the case when the bus wire is broken. The initial Issue is most likely related to either a firmware problem or a device holding the bus low. Not related to a mismatch in pull-up resistor values.

kdschlosser, 2024-03-27:
I am not sure if the documentation for the IDF has been updated properly. Upon looking at the code in the ESP-IDF they are not following their own directions for transmitting and receiving. If you look at the code for `i2c_master_write_to_device` in the ESP-IDF you will see that they don't call `i2c_cmd_link_create` and instead they call `i2c_cmd_link_create_static` instead. I am not sure of the reasoning for this.

The other thing I am seeing as an issue is when performing a scan a NULL is getting passed as the buffer to be transmitted. it appears like i2c_master_write is not liking this too much. In the current driver for micropython there is no error checking of that function and if there was an error would be returned when the function is called.

I also notice an issue with the timeout being used when calling `i2c_master_cmd_begin` The bus speed is not taken into consideration. So If I had a bus speed of say 50000hz and I was sending 100 bytes it would timeout before all of the bytes could be sent. the set timeout value would be 10.2 milliseconds when 16.16 milliseconds is what would be needed to send the data. I am pretty sure I have my calculations correct for the timeout. I could be wrong tho.

kdschlosser, 2024-03-27:
Now the `i2c_set_timeout` function doesn't work the same for the S3 series of chips. There is a mistake in the documentation about it stating the ADB frequency is what gets used. This is incorrect for this series of MCU's. The timeout value is ` timeout = 2^x * (1/XTAL_CLK_FREQ)`. This is due to the timeout value needing to fit into a 5 bit register.

https://www.esp32.com/viewtopic.php?t=32281#p114099