micropython/micropython #11837: Zephyr port: async event loop implementation starves CPU

bogdanm, 2023-06-21:
When using `asyncio` in MicroPython, the event loop implementation (`asyncio.run`) ends up polling Python objects in a queue. The polling code does this (`extmod/modselect.c:poll_poll_internal`):

```
...
    mp_uint_t start_tick = mp_hal_ticks_ms();
    mp_uint_t n_ready;
    for (;;) {
        // poll the objects
        n_ready = poll_map_poll(&self->poll_map, NULL);
        if (n_ready > 0 || (timeout != (mp_uint_t)-1 && mp_hal_ticks_ms() - start_tick >= timeout)) {
            break;
        }
        MICROPY_EVENT_POLL_HOOK
    }
...
```

The `for(;;)` loop above is a busy wait loop. In the Zephyr port, it consumes a lot of CPU time and it starves other threads. Even when `MICROPY_EVENT_POOL_HOOK` is set to `k_yield` (and thus the loop gives control back to the OS scheduler after each iteration) the code will still take most of the CPU time. Setting `MICROPY_EVENT_POOL_HOOK` to something like `k_msleep(100)` (which waits 100ms before running another iteration of the loop) fixes the starving issue, but it delays the Python thread for no good reason.

I have a possible solution for this, but please let me know first if there is interest to move this forward, since lately I got the impression that the Zephyr port isn't exactly a "fist class citizen" in the world of MicroPython ports.

dpgeorge, 2023-06-22:
We definitely want to bring the zephyr port up to the level that other "main" ports (like rp2, esp32, stm32, mimxrt) are at, and more efficient polling is an important thing to do.

There has already been quite a bit of discussion about how to make asyncio polling more efficient in a general way, for all ports.  See #6125.

But that might be a while until it's finalised, and in the meantime we could consider improving the way zephyr does polling.  Can you give more details on what you are thinking here?

dpgeorge, 2023-06-22:
That approach sounds close to what I was thinking with "uevent": each waitable entity (stream, pin, etc) would register with a central flag/semaphore, and would set/signal that flag/semaphore when it was ready (ie finished waiting, had some data ready, a pin changed, etc).  The main loop would then just wait on this flag/semaphore in the most efficient way that it could.

Your solution is strongly dependent on the user writing C code to give the semaphore so doesn't seem like it would generalise easily at this stage (eg to wait for incoming data on a UART, on zephyr).

Also probably the macro should account for the elapsed time:
```
MICROPY_POLL_SELECT_HOOK(timeout - (mp_hal_ticks_ms() - start_tick));
```

bogdanm, 2023-06-22:
> That approach sounds close to what I was thinking with "uevent": each waitable entity (stream, pin, etc) would register with a central flag/semaphore, and would set/signal that flag/semaphore when [...]

Yes, that'll definitely fix things. I can try to help with the implementation if needed, although at the moment I'm very new to the async world.

> Your solution is strongly dependent on the user writing C code to give the semaphore so doesn't seem like it would generalise easily at this stage (eg to wait for incoming data on a UART, on zephyr).

It looks like `uevent` is already a good generalization, I didn't realize that the main idea was so similar to what I had in mind.

Thanks for the timeout suggestion.