micropython/micropython #11843: docs: non standard behaviour of float()

massimosala, 2023-06-22:
See
https://github.com/micropython/micropython/issues/11735

The official page
https://docs.micropython.org/en/latest/genrst/builtin_types.html#float
should be  updated about the different result of

> float('_')

(and eventually other invalid inputs)
between Python and MicroPython.

robert-hh, 2023-06-26:
Instead of asking that someone should do something, please go ahead and do it. In this case, suggest the change for the documentation by making a Pull Request.

robert-hh, 2023-06-29:
You PR seems strange. It includes 106 commits with 967 changed files, when it was supposed to change some documentation files. Do you have any files which you changed yourself deliberately?

massimosala, 2023-07-06:
Hi

To create a new PR, the web page asks to choose a branch.
I tried to follow the instructions.

Pls help or, if you prefer, feel free to create the PR.