micropython/micropython #11853: ESP32-S3 only sees approx. half the SPI-RAM in latest (ESP-IDF v5.x) builds

ma261065, 2023-06-23:
I have an ESP32-S3 N16R8 device (i.e. 8MB SPI-RAM and 16MB Flash) 

If I flash this build `https://www.micropython.org/resources/firmware/GENERIC_S3_SPIRAM_OCT-20230623-unstable-v1.20.0-243-gd93316fe3.bin` and then check free memory with `gc.mem_free()` I see 4254384

If I flash with this (older) build `https://www.micropython.org/resources/firmware/GENERIC_S3_SPIRAM_OCT-20230621-unstable-v1.20.0-230-g41c91422f.bin` and then check free memory with `gc.mem_free()` I see 8192976

I believe that the major difference between these builds is that the later one is built against ESP-IDF v 5.x

VynDragon, 2023-07-02:
Can confirm same issue here. shows about 4 MB on my ESP32S3R8 on micropython master + esp-idf 5.1, and 8 MB when switching back to 1.20.0 and esp-idf 4.4.

jimmo, 2023-07-03:
This is intentional (although obviously not desirable) and changed due to API differences in IDF 5.x. Is this causing problems for your application (i.e. are you using large frame buffers?).

We are planning an update to the way SPIRAM is managed on ESP32 and planning to make it grow-on-demand. Having it use the entire SPIRAM means that there is nothing left for the IDF heap (e.g. for SSL buffers) and also that the GC has to work a lot harder every collection.

ma261065, 2023-07-03:
Thanks for the answer Jim.

> Is this causing problems for your application (i.e. are you using large frame buffers?)
I think it will cause a problem, as I'm allocating large buffers for audio playback.

What's the ETA for the update to allow the SPIRAM to auto-expand?

> Having it use the entire SPIRAM means that there is nothing left for the IDF heap
Was this not a problem pre ESP-IDF 5.x on these devices? Why does it have to use *all* the SPI-RAM, and not just whatever is left over after some portion is reserved for the heap?

VynDragon, 2023-07-03:
As long as it's configurable it wont cause any issues for me.

nspsck, 2023-07-23:
Hi, the issue is mainly caused by [this piece of code](https://github.com/micropython/micropython/blob/ea1a5e43d08429bed96e251f68f77aa0334bb371/ports/esp32/main.c#L108), where the total heap will be cut in half. You can delete that `/ 2` and your code will probably work again like before.

ma261065, 2023-07-23:
@nspsck thanks for tracking that down.

From reading the code I think the change is caused by a desire to simplify the task heap allocation. There is a whole code block that was deleted prior to that line which used to make a bunch of decisions based on the size of available SPIRAM. Now the logic seems to be that we get half the RAM at most for the heap, regardless, which prior to this change used to only be the case for chips without SPIRAM. The side effect of this is that we are basically throwing away a big chunk of the memory on the chips with larger SPIRAM amounts. 

Given that this issue is tagged for release 1.21, I presume that there will be some changes coming in this logic soon, probably incorporating some of the pending pull requests that deal with Garbage Collector optimisation.

Let's wait and see as, I imagine that the maintainers are flat-out-busy right now working through the side-effects of the big change to ESP-IDF 5.x. And for their hard work we are eternally grateful.

shariltumin, 2023-07-24:
```
$ diff main.c main.c-ORIG
109,110c109
< // void *mp_task_heap = heap_caps_malloc(mp_task_heap_size, caps);
< void *mp_task_heap = heap_caps_malloc(2000*1024, caps)
---
> void *mp_task_heap = heap_caps_malloc(mp_task_heap_size, caps);
```

Set the MP heap hardcoded to 2000*1024=2048000. We should get about 2000KB MP heap.

```
>>>
>>> gc.mem_free()
4273360
```

Still gc.mem_free() shows about half of the 8MB PSRAM free.

```
>>> esp32.idf_heap_info(esp32.HEAP_DATA)
[(32767, 32031, 31744, 32015), (8388608, 6338304, 6291456, 6338304), (265392, 230676, 229376, 230676), (22308, 64, 56, 28), (32768, 16864, 13312, 13496), (8132, 7752, 7680, 7752)]
>>>
```

But esp32.idf_heap_info() reports 'correctly' - (8388608, 6338304, 6291456, 6338304) the 6338304 is about 6MB since MP was set to 2000KB.

gc.mem_free() uses py/gc.c gc_info(), maybe gc.c is broken?

```
>>> import micropython as mp
>>> mp.mem_info()
stack: 704 out of 15360
GC: total: 4274752, used: 2688, free: 4272064
 No. of 1-blocks: 52, 2-blocks: 14, max blk sz: 18, max free sz: 266968
```

mem_info() also reports a total of 4274752, but it should be 2048000.

pillo79, 2023-07-25:
> ```
> $ diff main.c main.c-ORIG
> 109,110c109
> < // void *mp_task_heap = heap_caps_malloc(mp_task_heap_size, caps);
> < void *mp_task_heap = heap_caps_malloc(2000*1024, caps)
> ---
> > void *mp_task_heap = heap_caps_malloc(mp_task_heap_size, caps);
> ```

@shariltumin this only affects the memory allocation on the OS side, not its use by MP. The GC is still initialized a few lines after that with the value in `mp_task_heap_size`, causing lots of trouble :wink:
To have consistent results, you can instead force the value of that variable to your desired number (leave some free room for OS needs though!).

shariltumin, 2023-07-25:
Yes, @pillo79 you are right. @dpgeorge pointed out my mistake.
I tested two MP/IDF heap ratios for the 8MB PSRAM esp32s3 board.

1. 2/6:
```
$ diff main.c main.c-ORIG 
108,109c108
< // size_t mp_task_heap_size = MIN(heap_caps_get_largest_free_block(caps), heap_total / 2);
< size_t mp_task_heap_size = 2000*1024;
---
> size_t mp_task_heap_size = MIN(heap_caps_get_largest_free_block(caps), heap_total / 2);
```
2. 6/2:
```
$ diff main.c main.c-ORIG 
108,109c108
< // size_t mp_task_heap_size = MIN(heap_caps_get_largest_free_block(caps), heap_total / 2);
< size_t mp_task_heap_size = 6000*1024;
---
> size_t mp_task_heap_size = MIN(heap_caps_get_largest_free_block(caps), heap_total / 2);
```
Despite the esp-idf warning (up to 4MB), I manage to allocate more than 4MB of memory block. I do not know the impact on system stability, perhaps more knowledgeable persons can explain.

See issue #12075 for a full discussion.

dpgeorge, 2023-07-28:
@Tangerino can you please describe how your application uses RAM and why it breaks?  Just so we can better understand how to fix the problem.

Tangerino, 2023-07-28:
Sure. 
Devices drivers and network topologies are loaded in RAM, we monitor large solar farms with plenty of devices and electrical variables.
The firmware loads user code to perform custom control.
The firmware loads custom alarms definition.
The device supports WiFi, BLE, ETH and 4G all together.
The device is fully remotely managed
We support a remote shell and many other features.

All this uses a lot of RAM.

Using V1.19 (around 1200 devices in production as of today) I can monitor the RAM usage as in.
[image]

Now using V1.20 with very little functionality
[image]

So, to be 'fair' the application works. But we made a choice to use a device with a bigger RAM in order to do more stuff. Arbitrarily taking half of it has an impact on the overall device capabilities.

Can we optimize the app? Yes, we can, but I was thinking on double the RAM. And the more RAM I add, more RAM I loose.

I'm very surprised with the V1.20 performance! Above I can read 'grow-on-demand'. I think this is the way.

Thank you.

glancia, 2023-07-30:
I work with @Tangerino and I can tell that performance on 1.20 has been multiplied by a factor of 5 to 10 times.  Great job guys!

jimmo, 2023-08-01:
> I work with @Tangerino and I can tell that performance on 1.20 has been multiplied by a factor of 5 to 10 times. Great job guys!

@glancia @Tangerino that's very interesting, but quite surprising. Could you clarify whether you mean the v1.20 release or the currently nightly build. I'm guessing the latter as the v1.20 release didn't have the spiram allocation size change.

How are you measuring this performance improvement?

ma261065, 2023-08-01:
> the v1.20 release didn't have the spiram allocation size change

The current dailies still use the same SPIRAM allocation as 1.20, don't they?

jimmo, 2023-08-01:
> > the v1.20 release didn't have the spiram allocation size change
> 
> The current dailies still use the same SPIRAM allocation as 1.20, don't they?

@ma261065 No. That's what this issue is all about -- in the move to IDF 5.x (which was done a few weeks ago), the allocation strategy was changed. v1.20 (and nightlies until a few weeks ago) use the old way, and recent nighties use the new way.

ma261065, 2023-08-01:
Ahh, sorry - I had remembered that the change happened with 1.20. My bad.

dpgeorge, 2023-08-02:
> * system boot -> 6s on 1.20 vs 18s on 1.19 vs 30-40s on 1.18

This could be due to recent improvements with u-module naming, weak links, and reducing the number of times the filesystem is accessed when importing a module.

jimmo, 2023-08-02:
> @jimmo , it was 1.20 release, still cutting RAM by 2. The execution is not breaking because we're not stressing too much, but RAM free came as low as 30kb.

@glancia Can you confirm for sure that it's the v1.20 release? The problem is that the half-spiram change (and the improvements to import which would improve startup time) were not merged until _after_ the v1.20 release. So have you backported the RAM changes to v1.20?

Also @Tangerino asked in a different issue about the removal of `zlib`, but that was also post-v1.20.

(The slightly confusing thing is that when you build from source, or use a nightly build, the version number printed in the REPL is the _previous_ release -- i.e. we don't increment the version number until the next release -- see https://github.com/micropython/micropython/issues/12127).

> 
> We're not doing any scientific measurements just some operations performance comparisons, running the same application version and same application size:
> 
>     * system boot -> 6s on 1.20 vs 18s on 1.19 vs 30-40s on 1.18

As Damien pointed out in the previous comment, this could definitely be explained with the recent import changes.

> 
>     * a simple operation -> 0.5s on 1.20 vs 2.5s on 1.19

This is very interesting. Great news! But yeah, the last major series of performance improvements were merged just before the v1.18 release.

ma261065, 2023-08-02:
> The slightly confusing thing is that when you build from source, or use a nightly build, the version number printed in the REPL is the previous release -- i.e. we don't increment the version number until the next release -- see https://github.com/micropython/micropython/issues/12127

You are quite correct. v1.20 was released on 27th April 2023, but the change to use IDF 5.x came with commit https://github.com/micropython/micropython/commit/6a9db521eda2f62766749d84244c70fd2073d4a0 on 23rd June 2023

glancia, 2023-08-02:
@Tangerino please confirm which version you used

Tangerino, 2023-08-04:
I'm using the nightly source code.
The boot time is much faster but the overall performance is much better.
I did not make any measurements but GC time went to 0.250 msec to 0,12 msec.

PR #12141 (2023-08-15): py/gc & esp32: Add MICROPY_GC_SPLIT_HEAP_AUTO to automatically grow heap on esp32
When `MICROPY_GC_SPLIT_HEAP_AUTO` flag is set, the heap is automatically extended with new areas on demand, and shrunk if a heap area becomes empty during a GC pass or soft reset. Python code doesn't have to do anything, if an allocation would otherwise fail then a new heap area is added to contain it.

Enable this new flag for esp32 port.

Tested on ESP32 `GENERIC_SPIRAM` and `GENERIC_S3` configurations.

When this option is enabled `micropython.mem_info()` is extended with an extra piece of data, "max new split [... cut]
closed 2023-08-15 by dpgeorge