micropython/micropython #11855: Sponsorships: payment method

massimosala, 2023-06-23:
Hi

I want to contribute but here
https://github.com/sponsors/micropython/sponsorships
the only payment method available is credit card.

Sorry, but I stopped typing my credit card into websites for several years now.

I think it is deterrent for other people as well.
Can you add Paypal?

Massimo

dlech, 2023-06-23:
For reference:

GitHub stopped supporting Paypal as a payment option in February. https://github.blog/changelog/2023-01-23-github-sponsors-will-stop-supporting-paypal/

It is possible to add buttons for other platforms: https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/displaying-a-sponsor-button-in-your-repository

massimosala, 2023-06-23:
Mmh I see ... but .. are we forced to pay on github?

If Damien is registered on Paypal (he can always do it), then publish the email with which he registered - or contact me privately - and I can proceed with the monthly donation.

massimosala, 2023-06-27:
up
@dpgeorge

massimosala, 2024-06-18:
The parties seem not interested to receive funds with Paypal.

closed 2024-06-18 by massimosala