micropython/micropython #11856: littlefs: upgrade

massimosala, 2023-06-23:
MP uses Littlefs 2.3:
https://github.com/micropython/micropython/tree/master/lib/littlefs

This version was released in December 2020:
https://github.com/littlefs-project/littlefs/releases

In the meantime, **several bugs have been fixed**, performance improved and new versions released.

The latest version is 2.6.1. From its notes:
_This release bumps the on-disk minor version of littlefs from lfs2.0 -> lfs2.1.
This is the first on-disk minor version bump for littlefs, and is untested waters, so do open a GitHub issue if this creates any problems._

Maybe it's premature to jump from 2.3 to 2.6, but could we at least upgrade to 2.5.1?

dlech, 2023-06-23:
It was updated to v2.5.0 in #8674

massimosala, 2023-06-23:
> It was updated to v2.5.0 in #8674

Hi David

That's great news!

So I propose for MP 1.21 to upgrade from 2.5.0 to 2.5.1.
For my own experience, it is better to follow eternal libraries releases.
Stay behind, then when you have to update are pains.

@jimmo @dpgeorge  what do you think ?

---

BTW, someone has to update
https://github.com/micropython/micropython/tree/master/lib/littlefs

massimosala, 2023-06-27:
up

dlech, 2023-06-27:
Hi @massimosala. We appreciate your ideas and enthusiasm for MicroPython. But adding comments that don't add any useful information can be annoying. Every comment you make sends a notification to 729 people who probably don't want to be bothered with "up".

MicroPython is a large and complex project and can seem to move quite slow in some aspects. It is quite common for months or sometimes even years to pass before we come back to some ideas. 

Taking this issue, for example, if it is just "best practice" but not actually causing a bug in real-world use, then it is likely to remain low priority.

massimosala, 2023-06-27:
> Hi @massimosala. We appreciate your ideas and enthusiasm for MicroPython. But adding comments that don't add any useful information can be annoying. Every comment you make sends a notification to [...]

Hi David.

From littlefs 2.5.0 to 2.5.1 there is some work done. And it is the file-system, not an optional library.

I am learning about your resources and project management, I cannot subvert your priorities.
In the future I will try not to insist on open topics.

mattytrentini, 2024-06-19:
I'm going to close this ticket. For reference, the version of [littlefs that MicroPython uses is currently v2.8.1](https://github.com/micropython/micropython/tree/master/lib/littlefs).

Littlefs maintains a clear list of changes for each [Release](https://github.com/littlefs-project/littlefs/releases). MicroPython reviews those changes periodically but prioritises _stability_ over latest features in terms of migrating to new littlefs versions. That is, of course, balanced with staying _close enough_ to the latest version to ensure that upgrading isn't overly painful. 

If there is a desire or need to migrate to a _specific_ littlefs version then please raise a ticket detailing why it's necessary. Ideally, prove that the upgrade is desirable with benchmarks for your use case. 

Validating that a littlefs upgrade is free of issue on all ports is complex and time-consuming!

closed 2024-06-19 by mattytrentini