micropython/micropython #11857: What happened to the ssd1306 driver?

PsuFan, 2023-06-23:
Can someone help me find this driver? Not sure where it went. Trying to use with ESP32 / i2c.

https://github.com/micropython/micropython/blob/master/drivers/display/ssd1306.py

Thanks!

ricksorensen, 2023-06-23:
[https://github.com/micropython/micropython-lib/blob/7128d423c2e7c0309ac17a1e6ba873b909b24fcc/micropython/drivers/display/ssd1306/ssd1306.py#L30](https://github.com/micropython/micropython-lib/blob/7128d423c2e7c0309ac17a1e6ba873b909b24fcc/micropython/drivers/display/ssd1306/ssd1306.py#L30]

check `micropython-lib`

PsuFan, 2023-06-23:
okay thanks! not sure why it was moved or not built in but okay...

closed 2023-06-23 by PsuFan

dpgeorge, 2023-06-24:
You can install it via `mpremote mip install ssd1306`.

TecDroiD, 2023-08-02:
I just tried to use this and got an ENODEV error

does anybody have an idea what this has to say?

`
>>> import ssd1306
>>> from machine import I2C
>>> i2c = I2C(0)
>>> i2c.scan()
[60]
>>> display = ssd1306.SSD1306_I2C(128, 64, i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ssd1306.py", line 1, in __init__
  File "ssd1306.py", line 1, in __init__
  File "ssd1306.py", line 1, in init_display
  File "ssd1306.py", line 1, in write_cmd
OSError: [Errno 19] ENODEV
`

peterhinch, 2023-08-02:
ENODEV means that the hardware has not been detected.

PsuFan, 2023-08-02:
Weird, are you using a 1306 or 1309?

PsuFan, 2023-08-02:
> it's been worse :D on lilygo ESP32-Mini, there is a Pin called CMD which I misread as GND. So basically, the device was recognized because i2c pins were enough to power the bus driver circuit but it didn't power the display circuit itself.

Yeah that label is on a lot of boards!