micropython/micropython #11861: RP 2W Bluetooth : Incomplete Adv_Data

dobodu, 2023-06-24:
Advertissement Data seem to be incomplete after _IRQ_SCAN_RESULT while using BLE

I've been trying to check the data in two way : 

Pico with latest built ==> 0x0319400202010A020A000303F0FF
LightBlue on phone  ==> 0x0319400202010A020A000303F0FF07094F4244424C45

In this example the "07" bytes long "09=Name" corresponding to 4F4244424C45 does not exist on the Pico

dpgeorge, 2023-06-25:
It could be that the "name" part of the advertising data is in the scan response data, which is sent separately and requires passing `True` in to the 4th argument to `BLE.gap_scan()`.

dobodu, 2023-06-25:
> It could be that the "name" part of the advertising data is in the scan response data, which is sent separately and requires passing `True` in to the 4th argument to `BLE.gap_scan()`.

Thank you for the advice,

I tried : 

`ble.gap_scan(2000, 30000, 30000, True) `

but the  _IRQ_SCAN_RESULT, wich is the only one transmitting adv_data within data field, still misses the name (0x09 for ADV_TYPE_NAME).

While this is annoying (no way to show understandable name (UUID or MAC are not user friendly), the connection can still be made based on the UUID.

dpgeorge, 2023-06-27:
Please try running the following code on your Pico W, while the other device is advertising:
```py
import time, bluetooth

ble = bluetooth.BLE()

def irq_handler(e, d):
    if e == 5:
        mac = "--:--:--:--:" + d[1][-2:].hex(":")
        print(e, (d[0], mac, d[2], d[3], bytes(d[4])))
    else:
        print(e, d)

ble.irq(irq_handler)
ble.active(True)
ble.gap_scan(1000, 6250, 6250, True)
time.sleep_ms(1100)
```
And paste here the output.

dobodu, 2023-07-01:
Thank you Dpgeorges,

Here the output : 

`5 (0, '--:--:--:--:22:33', 0, -63, b'\x03\x19@\x02\x02\x01\n\x02\n\x00\x03\x03\xf0\xff')
5 (0, '--:--:--:--:22:33', 4, -63, b'\x07\tOBDBLE')`

The code can reach the name of BLE device, my code was wrong !

I close the issue !

closed 2023-07-01 by dobodu