micropython/micropython #11872: esp32: Cannot create an UART or I2C object with the most recent daily build.

robert-hh, 2023-06-26:
This is the sequence of commands at REPL. The board is a "classic" ESP32 with SPIRAM.

```
MicroPython v1.20.0-243-gd93316fe3 on 2023-06-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from machine import UART, Pin
>>> u=UART(1, rx=4, tx=5)
E (17470) uart: uart_param_config(726): Invalid src_clk

>>> from machine import I2C
>>> i=I2C(1, sda=Pin(14), scl=Pin(15))
E (262380) i2c: i2c_driver_delete(433): i2c driver install error

>>> import platform
>>> platform.platform()
'MicroPython-1.20.0-xtensa-IDFv5.0.2-with-newlib4.1.0'
>>> 
```

dpgeorge, 2023-06-27:
Thanks for the report, I can confirm the issue.  I have a fix, to always set `source_clk` in the uart config struct.

dpgeorge, 2023-06-27:
Fixed by 5a9a0df3407cb195ff1f61cf9cba0fc8700c54c6

closed 2023-06-27 by dpgeorge