micropython/micropython #11874: Building firmware ESP IDF 5.0 Errors

MATTYGILO, 2023-06-26:
I am building with the latest esp32s3 micropython code which is now ESP IDF 5.0

I am getting these errors

"""
idf/main/CMakeFiles/__idf_main.dir
/MyPathmicropython/ports/esp32/ppp_set_auth.c.obj
cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C
In function 'gen_numname',
    inlined from 'dir_register' at /MyPathmicropython/lib/oofatfs/ff.c:2513:13:
/MyPathmicropython/lib/oofatfs/ff.c:1988:11: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
 1988 |     ns[i] = '~';
      |     ~~~~~~^~~~~
/MyPathmicropython/lib/oofatfs/ff.c: In function 'dir_register':
/MyPathmicropython/lib/oofatfs/ff.c:1959:10: note: at offset -1 into destination object 'ns' of size 8
 1959 |     BYTE ns[8], c;
      |          ^~
[ 70%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/
MyPathmicropython/ports/esp32/uart.c.obj
cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C
[ 70%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/
MyPathmicropython/ports/esp32/usb.c.obj
cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C
[ 70%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/
MyPathmicropython/ports/esp32/usb_serial_jtag.c.obj
cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C
/MyPathmicropython/ports/esp32/usb.c:35:10: fatal error: tinyusb.h: No such file or directory
   35 | #include "tinyusb.h"
      |          ^~~~~~~~~~~
compilation terminated.
"""

MATTYGILO, 2023-06-26:
@dpgeorge I think you need to include `tinyusb` in
[micropython/ports/esp32/esp32_common.cmake](https://github.com/micropython/micropython/blob/92c7532d8e461f3fd9ea44f4d024ee3ce156e5e8/ports/esp32/esp32_common.cmake#L104)

For the error to stop

robert-hh, 2023-06-26:
Which board did you try to build? No problem here with GENERIC_S3_SPIRAM.

MATTYGILO, 2023-06-26:
Currently using GENERIC_S3_SPIRAM_OCT with custom sdkconfig, which may be part of the reason for the error, just updated to the newest micropython from a previous version using v4.4, just need to modify stuff so it then uses 5.0

MATTYGILO, 2023-06-26:
For the other error ChatGPT suggests increasing the buffer size
```
static void gen_numname (
    BYTE* dst,          /* Pointer to the buffer to store numbered SFN */
    const BYTE* src,    /* Pointer to SFN */
    const WCHAR* lfn,   /* Pointer to LFN */
    UINT seq            /* Sequence number */
)
{
    BYTE ns[12], c;  // Increased buffer size to 12
```
However I'm uncertain if that is the correct logic
https://github.com/micropython/micropython/blob/92c7532d8e461f3fd9ea44f4d024ee3ce156e5e8/lib/oofatfs/ff.c#L1959

robert-hh, 2023-06-26:
ChatGPT just parrots what others think, but don't know if it's true.

robert-hh, 2023-06-26:
Nevertheless, it seems that the code has a problem if the upper 4 bits of seq are not 0. Then` ns[-1]` will be overwritten. That may be another variable on the stack, like `c`, which is not used anymore.

robert-hh, 2023-06-26:
Today I've built for GENERIC, GENERIC_SPIRAM, GENERIC_C3, GENERIC_C3_USB, GENERIC_S3_SPIRAM and UM_FEATHER_S2. No problems. And Damien had built it for all 28 variants of the board.

MATTYGILO, 2023-06-26:
So I think it is because of my custom sdkconfig.cmake because I have these settings:
```
 set(CONFIG_USB_OTG_SUPPORTED "y")
 set(CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE "256")
 set(CONFIG_USB_HOST_HW_BUFFER_BIAS_BALANCED "y")
```

MATTYGILO, 2023-06-26:
Maybe it is because

[```](https://github.com/micropython/micropython/blob/92c7532d8e461f3fd9ea44f4d024ee3ce156e5e8/ports/esp32/boards/GENERIC_S3_SPIRAM_OCT/mpconfigboard.cmake#L5)

https://github.com/micropython/micropython/blob/92c7532d8e461f3fd9ea44f4d024ee3ce156e5e8/ports/esp32/boards/sdkconfig.usb#L1

The reference uses the USB and then tinyusb is not available

mattytrentini, 2023-06-27:
Are you using IDF 5.0.2?

dpgeorge, 2023-06-27:
As @robert-hh says, all esp32 boards defined in this repo build correctly with IDF 5.0.2.

If you're using tinyusb then you need a `idf_component.yml` with:
```
## IDF Component Manager Manifest File
dependencies:
  idf:
    version: ">=5.0.2"
  espressif/esp_tinyusb: "~1.0.0"
```

Tangerino, 2023-06-27:
Any help here, please.
`make` works but `make BOARD-GENERIC_SPIRAM` don't.

`make` uses the right make path `/opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E make_directory /......`

But the above errors displays `/opt/homebrew/Cellar/cmake/3.25.1/bin/cmake: No such file or directory`

Is there any updated guide?

Tangerino, 2023-06-27:
I manually removed all `build_*` folders, ran `make erase` and now it compiles!
What is the step I've missed before?

MATTYGILO, 2023-06-27:
So I no longer get compile errors I've just added some bounds

```
static void gen_numname(
    BYTE* dst,
    const BYTE* src,
    const WCHAR* lfn,
    UINT seq
)
{
    BYTE ns[8], c;
    int i, j;  // Change UINT to int to allow negative values
    WCHAR wc;
    DWORD sr;

    mem_cpy(dst, src, 11);

    if (seq > 5) {
        sr = seq;
        while (*lfn) {
            wc = *lfn++;
            for (i = 0; i < 16; i++) {
                sr = (sr << 1) + (wc & 1);
                wc >>= 1;
                if (sr & 0x10000) sr ^= 0x11021;
            }
        }
        seq = (UINT)sr;
    }

    i = 7;
    do {
        c = (BYTE)((seq % 16) + '0');
        if (c > '9') c += 7;
        if (i >= 0) {  // Check if i is within bounds
            ns[i] = c;
        }
        i--;
        seq /= 16;
    } while (seq);
    if (i >= 0) {  // Check if i is within bounds
        ns[i] = '~';
    }

    for (j = 0; j < i && dst[j] != ' '; j++) {
        if (dbc_1st(dst[j])) {
            if (j == i - 1) break;
            j++;
        }
    }
    do {
        if (j < 8) {  // Check if j is within bounds
            dst[j++] = (i < 8) ? ns[i++] : ' ';
        } else {
            break;
        }
    } while (j < 8);
}
```

MATTYGILO, 2023-06-27:
Ok so including idf components with tiny usb fixes those errors now getting these errors:

```
boards/MICROLITE_S3_SPIRAM_OCTAL/managed_components/espressif__esp_tinyusb/include/tusb_cdc_acm.h:51:5: error: expected specifier-qualifier-list before 'cdc_line_coding_t'
   51 |     cdc_line_coding_t const *p_line_coding; /*!< New line coding value */
      |     ^~~~~~~~~~~~~~~~~
submodules/micropython/ports/esp32/usb.c:41:27: error: 'CONFIG_TINYUSB_CDC_RX_BUFSIZE' undeclared here (not in a function); did you mean 'CFG_TUD_CDC_RX_BUFSIZE'?
   41 | static uint8_t usb_rx_buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE];
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           CFG_TUD_CDC_RX_BUFSIZE
[ 71%] Building C object esp-idf/main/CMakeFiles/__idf_main.dirsubmodules/micropython/ports/esp32/gccollect.c.obj
submodules/micropython/ports/esp32/usb.c: In function 'usb_tx_strn':
submodules/micropython/ports/esp32/usb.c:86:12: error: implicit declaration of function 'tud_cdc_n_connected'; did you mean 'tud_connected'? [-Werror=implicit-function-declaration]
   86 |     while (tud_cdc_n_connected(CDC_ITF) && len && esp_timer_get_time() < timeout) {
      |            ^~~~~~~~~~~~~~~~~~~
      |            tud_connected
cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C
submodules/micropython/ports/esp32/usb.c:90:9: error: implicit declaration of function 'tud_cdc_n_write_flush'; did you mean 'tinyusb_cdcacm_write_flush'? [-Werror=implicit-function-declaration]
   90 |         tud_cdc_n_write_flush(CDC_ITF);
      |         ^~~~~~~~~~~~~~~~~~~~~
      |         tinyusb_cdcacm_write_flush
At top level:
submodules/micropython/ports/esp32/usb.c:41:16: warning: 'usb_rx_buf' defined but not used [-Wunused-variable]
   41 | static uint8_t usb_rx_buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE];
      |                ^~~~~~~~~~
```

dpgeorge, 2023-06-28:
@Tangerino if you had `build-*` directories left over from a IDF 4.x build, then they must be fully deleted before building with IDF 5.x.

MATTYGILO, 2023-06-28:
So fixed the other errors by just creating a new board settings afresh based on micropython default octal board. But, I'm getting this error still

```
/home/runner/work/chorus_device/chorus_device/chorusware/submodules/micropython/lib/oofatfs/ff.c:1990:15: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
 1990 |         ns[i] = '~';
      |         ~~~~~~^~~~~
/home/runner/work/chorus_device/chorus_device/chorusware/submodules/micropython/lib/oofatfs/ff.c: In function 'dir_register':
/home/runner/work/chorus_device/chorus_device/chorusware/submodules/micropython/lib/oofatfs/ff.c:1959:10: note: at offset -1 into destination object 'ns' of size 8
 1959 |     BYTE ns[8], c;
      |          ^~
```

domix75, 2024-01-13:
I'm trying to build a fw for esp32s3 with 32MB. I cloned esp-idf and micropython (branch 1.22-release)
I manage to build esp32 generic
but when I set the target to esp32s3, I get the same error as @MATTYGILO reports just above : 
```
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/network_common.c.obj
In file included from micropython/ports/esp32/usb.c:36:
micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include/tusb_cdc_acm.h:51:5: error: expected specifier-qualifier-list before 'cdc_line_coding_t'
   51 |     cdc_line_coding_t const *p_line_coding; /*!< New line coding value */
      |     ^~~~~~~~~~~~~~~~~
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/machine_bitstream.c.obj
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/network_lan.c.obj
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/network_ppp.c.obj
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/network_wlan.c.obj
micropython/ports/esp32/usb.c:41:27: error: 'CONFIG_TINYUSB_CDC_RX_BUFSIZE' undeclared here (not in a function); did you mean 'CFG_TUD_CDC_RX_BUFSIZE'?
   41 | static uint8_t usb_rx_buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE];
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           CFG_TUD_CDC_RX_BUFSIZE
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/mpnimbleport.c.obj
micropython/ports/esp32/usb.c: In function 'usb_tx_strn':
micropython/ports/esp32/usb.c:95:12: error: implicit declaration of function 'tud_cdc_n_connected'; did you mean 'tud_connected'? [-Werror=implicit-function-declaration]
   95 |     while (tud_cdc_n_connected(CDC_ITF) && len && esp_timer_get_time() < timeout) {
      |            ^~~~~~~~~~~~~~~~~~~
      |            tud_connected
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/modsocket.c.obj
micropython/ports/esp32/usb.c:99:9: error: implicit declaration of function 'tud_cdc_n_write_flush'; did you mean 'tinyusb_cdcacm_write_flush'? [-Werror=implicit-function-declaration]
   99 |         tud_cdc_n_write_flush(CDC_ITF);
      |         ^~~~~~~~~~~~~~~~~~~~~
      |         tinyusb_cdcacm_write_flush
micropython/ports/esp32/usb.c: At top level:
micropython/ports/esp32/usb.c:41:16: warning: 'usb_rx_buf' defined but not used [-Wunused-variable]
   41 | static uint8_t usb_rx_buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE];
      |                ^~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/build.make:3088 : esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/usb.c.obj] Erreur 1
make[2]: *** Attente des tâches non terminées....
[ 97%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/modesp.c.obj
make[1]: *** [CMakeFiles/Makefile2:5645 : esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/all] Erreur 2
make: *** [Makefile:136 : all] Erreur 2
make failed with exit code 2, output of the command is in the micropython/ports/esp32/build/log/idf_py_stderr_output_171694 and micropython/ports/esp32/build/log/idf_py_stdout_output_171694

```

projectgus, 2024-01-17:
@domix75 MicroPython isn't building quite correctly here, as the output shows it's using a plain `build` directory:

> make failed with exit code 2, output of the command is in the micropython/ports/esp32/build/log/idf_py_stderr_output_171694 and micropython/ports/esp32/build/log/idf_py_stdout_output_171694

This usually happens if you invoke `idf.py` or CMake directly. MicroPython esp32 port needs to build via its own makefiles, i.e. by running `make BOARD=ESP32_GENERIC` for plain ESP32 and `make BOARD=ESP32_GENERIC_S3` for esp32s3.

This performs a bunch of extra configuration wrangling, and uses a separate build directory each time (`build-BOARDNAME`).

If you've made this change and still see an error, please post the exact command(s) you've run and the full build output (can use https://gist.github.com for this).

domix75, 2024-01-21:
Thanks
when I do that, (make BOARD=ESP32_GENERIC_S3) it works fine indeed, but I have this problem when I modify boards/ESP32_GENERIC_S3/sdkconfig.board to create a firmware for a 32Mo card (ESP32-S3-DevKitC-1-N32R8V)

$ cat boards/ESP32_GENERIC_S3/sdkconfig.board 
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y

CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE_32MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions-32MiB.csv"

here is the trace:
https://gist.github.com/domix75/17e830d9110a58a3571b5f3dc3690001

thanks again

projectgus, 2024-01-22:
@MATTYGILO Are you still seeing the original problem that you opened this issue for? 

If yes, then could you please post the board configuration you're using that reproduces the problem?

projectgus, 2024-02-20:
@domix75 My best guess would be that there was an existing filesystem in the flash left over from running MicroPython with an 8MB partition layout (this is the default S3 size). You can try reformatting, using the [steps here](https://docs.micropython.org/en/latest/reference/filesystem.html#esp32).

If that doesn't work, please open a new Discussion with as many details as possible and can resume from there.

jonnor, 2024-08-24:
Issue does not seem to be reproducible. Proposing to close this.

closed 2024-08-24 by robert-hh