micropython/micropython #11882: ESP32: target_copile options are not populated to module

vikramdattu, 2023-06-27:
As commented at below project, `target_compile_options` passed to the interface target, which is interfaced with `usermod` target, are not getting populated in `.cc` files which are sources of this interface target.

https://github.com/mocleiri/tensorflow-micropython-examples/pull/120#issuecomment-1608933468 

micro_python tip: [9b48634](https://github.com/micropython/micropython/tree/9b486340da22931cde82872f79e1c34db959548b)
The interface target cmake: https://github.com/vikramdattu/tensorflow-micropython-examples/blob/368fe328af641bf776559babaa6f485b5584ee9b/micropython-modules/microlite/micropython.cmake#L450 

Any pointer would be appreciated. I have tried every possible way of passing compile_options but nothing is passed! :(

vikramdattu, 2023-06-29:
An update to the issue:
As per my understanding the build system of `micropython` has an issue, especially when adding new module.

The build system does pre-processing for itself, which does not collect all the compile_flags for the step. It only collects ones from idf_components.
More details here: https://github.com/mocleiri/tensorflow-micropython-examples/pull/120#issuecomment-1612467330 

Please take a look at the `ports/esp32/main/CMakeList.txt`, you can notice the following code snippet, which confirms the same.

```
# Collect all of the include directories and compile definitions for the IDF components.

foreach(comp ${IDF_COMPONENTS})
    micropy_gather_target_properties(__idf_${comp})
endforeach()
```

To address the problem for now, I made an adjustment by adding the `micropy_gather_target_properties(microlite)` call in the `microlite/micropython.cmake` file.
This modification successfully resolves the error. Please note that including `micropython/py/py.cmake` is necessary for the aforementioned macro to function correctly.

cnadler86, 2024-10-12:
I am struggling with a similar problem. Did you found a solution it could you please provide a little more inside on how to solve this?
Thank you ver much