micropython/micropython #11925: No access to web server on the Raspberry Pi Pico W from external networks

CHPeter, 2023-07-03:
Hello
WEB server Raspberry Pi Pico W is ideally accessible from a local network and does not respond to a request from an external network - for example http://217.65.242.xx: 34911.
Port 34911 on the router translates to port 80.
If I register port 80 in Firewall NAT instead of port 34911, everything works.
Please help me.
I use:
MicroPython v1.20.0 2023-04-26
Last firmware from https://rpf.io/pico-w-firmware.

Thank you.
Peter.

[pico_w_network.docx](https://github.com/micropython/micropython/files/11938079/pico_w_network.docx)

evilhamsterman, 2023-07-31:
This is not a bug in micropython, this is how you have setup your NAT in your firewall. You have to map your <external ip>:<external port> to the webserver's <internal ip>:<internal port> The type of server doesn't matter it could be a Pico, it could be an Apache server the same applies.

If your firewall is set to forward external TCP port 34911 to the internal IP address 172.16.99.76:80 then it depending on your firewall, you may have to also create a firewall rule to allow your firewall to accept connections on port 34911 from one or any external IPs. 

A few general notes though
1. If you want to post code don't post it as a docx file, either post it directly in the ticket you can see how here https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code, or create a [gist](gist.github.com) and link to it in the ticket
2. Make sure you remove any sensitive information from your code like your SSID, password, and your network configuration
3. Port forwarding in general should be done with considerable care, even shifting your port to a different one than default is just obfuscation. For accessing stuff on your local network it's much better to use a VPN type tool. Many firewalls have support for them or you can use something like Tailscale with a subnet router https://tailscale.com/kb/1019/subnets/

jonnor, 2024-08-24:
As described above, this is not a problem in MicroPython. Proposing to close this issue.

closed 2024-08-24 by robert-hh