micropython/micropython #11941: ESP32 serie: SPI 2 fails silently when not using a miso pin, and additionally crashes the MCU when the ESP is specifically a ESP32S3

VynDragon, 2023-07-05:
On 1.20.0 and master, with ESP-IDF 5.1 and 4.4.5, interacting with SPI bus 2 silently fail without mosi pin assigned, this is not the case for SPI 1.
### On ESP32S3, this additionaly causes the ESP to throw a panic and reset.

to reproduce on any, initialize the SPI 2 with parameters similar to SPI 1:
```display_spi = machine.SPI(2,baudrate=27000000,sck=machine.Pin(18, machine.Pin.OUT),mosi=machine.Pin(13, machine.Pin.OUT))```
Any attempts to use the bus after this will result in nothing.

Additionaly, any SPI command on ESP32S3's SPI2 without initializing the full set of pins (sck, mosi, and miso) will instantly crash the ESP and yield no exploitable error message or log.

for example:`machine.SPI(2)` will result in a panic screen and reset

### Expected result
Either a error message like SoftSPI, or the ability to use only 2 pins on regular ESP32 like SPI 1
Additionaly, no crash and panic on ESP32S3.

VynDragon, 2023-07-05:
Additionally, see #11918

Meteors27, 2023-07-16:
same issue

dpgeorge, 2023-07-25:
Improvements to SPI on esp32 have been made in commits aab8061dceb2ea4c6e87a876f48969e4a5a85c94, 162dd022b188b3e26738bb692417480a0c5546c4 and b2adfc8077435419c5a5eecb2818cba44f1d3f8f

The above `display_spi` example code works on all esp32 boards.  As does constructing `machine.SPI(2)`.

So I'll close this issue now.  Please reopen it if there is still a problem.

closed 2023-07-25 by dpgeorge