micropython/micropython #11960: Cannot allocate memory

ruihua-code, 2023-07-07:
**socket server** and **ble gap_scan** working memory cannot be allocated,how to solve, thank you

**firmware version:** v1.20.0
**dev board:** ESP-32-WROOM (CP2102)

**Error ouput:**
`E (466152) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 60 bytes)`
`E (483392) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 108 bytes)`
`E (490932) MDNS: Cannot allocate memory (line: 216, free heap: 28 bytes)`
`E (490932) MDNS: Cannot allocate memory (line: 216, free heap: 88 bytes)`
`E (528322) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 80 bytes)`
`E (528322) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 108 bytes)`
`E (528322) MDNS: Cannot allocate memory (line: 3088, free heap: 88 bytes)`
`E (528322) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 160 bytes)`
`E (542992) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 128 bytes)`
`E (558052) MDNS: Cannot allocate memory (line: 216, free heap: 76 bytes)`
`E (558052) MDNS_Networking: Cannot allocate memory (line: 131, free heap: 228 bytes)`
`E (558052) MDNS: Cannot allocate memory (line: 3088, free heap: 228 bytes)`
`

**micropython.mem_info() print:**
`stack: 860 out of 4096`
`GC: total: 108032, used: 4960, free: 103072`
`No. of 1-blocks: 93, 2-blocks: 26, max blk sz: 19, max free sz: 4227`

**ble scan code:**
```python
import bluetooth
from time import sleep
from micropython import const
import zrh_thread_var
import gc
import micropython

_IRQ_SCAN_RESULT = const(5)
_XIAO_MI_MAC = const('D9:6E:B5:74:A8:E3')

def do_ble_central():
    bt = bluetooth.BLE()
    bt.active(True)
    def scan_callback(event, data):
        if event == _IRQ_SCAN_RESULT:
            _, addr, _, rssi, _ = data            
            device_address = ":".join("{:02X}".format(b) for b in addr)
            if _XIAO_MI_MAC == device_address:
                print("find ok", device_address, rssi)                

    bt.gap_scan(0, 2000000, 2000000)
    bt.irq(scan_callback)

    while True:        
        sleep(2)
        gc.collect()
        print("scaning...")
        micropython.mem_info()
        if zrh_thread_var.ble_central_runing == False:
            bt.gap_scan(None)
            bt.active(False)
            print("stop scan......")
            break
```

ruihua-code, 2023-07-08:
ESP-IDF v4.4.4

jimmo, 2023-07-12:
The memory it's talking about here is the IDF heap, not the MicroPython heap.

> socket server and 

> E (490932) MDNS: Cannot allocate memory (line: 216, free heap: 28 bytes)

I don't see any socket server in your example code? Or anything that configures the networking interfaces (which would enable MDNS)

> ESP-IDF v4.4.4

Can you try using a nightly build (that uses IDF 5.0)

closed 2023-08-24 by ruihua-code