micropython/micropython #11961: WebAssembly/PyScript: the _mp_js_hook function doesn't work in web-workers.

ntoll, 2023-07-07:
Hi @dpgeorge - I think I've found a bug, but I can't find the relevant code in this repository, and I can only assume it's on a private branch somewhere (otherwise I'd have created a PR and checked my proposed fix). Please can you confirm/integrate the following?

In `micropython.mjs` (as served from: https://cdn.jsdelivr.net/npm/@micropython/micropython-webassembly-pyscript@1.20.0-268/micropython.mjs), when running in a web worker that requires access to the WASM filesystem I get an error thrown on line 3995 in the following function:

```js
function _mp_js_hook() {
          if (typeof window === 'undefined') {
              var mp_interrupt_char = Module.ccall('mp_hal_get_interrupt_char', 'number', ['number'], ['null']);
              var fs = require('fs');
  
              var buf = Buffer.alloc(1);
              try {
                  var n = fs.readSync(process.stdin.fd, buf, 0, 1);
                  if (n > 0) {
                      if (buf[0] == mp_interrupt_char) {
                          Module.ccall('mp_sched_keyboard_interrupt', 'null', ['null'], ['null']);
                      } else {
                          process.stdout.write(String.fromCharCode(buf[0]));
                      }
                  }
              } catch (e) {
                  if (e.code === 'EAGAIN') {
                  } else {
                      throw e;
                  }
              }
          }
      }
```

The line throwing the error is `var fs = require('fs');` with the reported error being `TypeError: require is not a function`.

This is correct, because `require` is a `node` based function and not available in the browser. Looking at the logic of the function, I notice this line:

```js
if (typeof window === 'undefined') {
    ....
```

I believe this should be changed to:

```js
if (ENVIRONMENT_IS_NODE) {
    ...
```

The code enclosed in this function is, AFAICT, for `node` based things, BUT when this code is run within a web worker the `window` is `undefined` - causing a false positive.

The use case that highlighted this problem is the "worker" example (https://github.com/ntoll/tortle/tree/main/examples/workers) in my `tortle` module (a re-write of `turtle`) which uses web-workers to create... well, turtles all the way down. :wink: The script to be evaluated on the web worker by MicroPython has an `import turtle` statement that needs to read the `turtle.py` module added to the runtime's filesystem.

Hope this makes sense. Just ask if not. Thank you!

ntoll, 2023-07-11:
@dpgeorge now that I've found the right branch :wink:, this PR should fix the issue: https://github.com/dpgeorge/micropython/pull/18.

dpgeorge, 2023-07-12:
This is fixed on that branch, now available as https://www.npmjs.com/package/@micropython/micropython-webassembly-pyscript/v/1.20.0-295

TODO: that patch should be merged to this repo.

ntoll, 2023-07-12:
Thank you @dpgeorge !

dpgeorge, 2023-07-13:
Fixed in this repo by 14c2b641312c7faec20b521b910a6354280ca068

closed 2023-07-13 by dpgeorge