micropython/micropython #11965: Evaluate Profile-Guided Optimization (PGO)

zamazan4ik, 2023-07-08:
Hi!

Recently I checked Profile-Guided Optimization (PGO) improvements on multiple projects. The results are [here](https://github.com/zamazan4ik/awesome-pgo/).

Since PGO showed measurable improvements in compiler/interpreter-like workloads (CPython, Clang, Clangd, clang-format, GCC, Rustc, etc.) I think it could be useful to check PGO for the Micropython project too.

We need to perform PGO benchmarks on Micropython. And if it shows improvements - add a note about possible improvements in Micropython performance with PGO. Providing an easier way (e.g. a build option) to build scripts with PGO can be useful for the end-users too.

jonnor, 2024-09-01:
Anyone interested in doing some PGO benchmarks? It is most relevant on actual hardware platforms, since on host ports like Unix it is already way fast enough and lots of memory available.

stinos, 2024-09-02:
The interpreter etc are the same for all ports and running with PGO on unix is probably going to be simpler overall so it still makes sense to find areas to optimize even though it might be fast enough already.
Question is rather: what benchmark should be used for the PGO?

jonnor, 2024-09-02:
The interpreter is the same, but the hardware is vastly different. Including things like memory access, which influences low-level benchmarks a lot. So much that I would hesitate to conclude that on-device behavior will be better if x64 perf is better. But it is of course a starting point :)
There are at least two benchmarks included in MicroPython in the tests/ directory - perf_bench, internal_bench.

stinos, 2024-09-03:
Yes, it's not about making conclusions from x64, because just as you say one cannot conclude on-device behavior from an x64 test, one cannot conclude on-device behavior on device X from PGO on device Y. Essentially we'd have to run PGO on at least a couple of devices..

zamazan4ik, 2024-09-06:
I can suggest taking a look at how PGO is integrated in CPython right now (it has been for a long time). Which they run for collecting PGO profiles, how frequently, for which builds, etc. I guess it would be easier to do than inventing the PGO policy from scratch. At the very least, you can find some inspiration with CPython's PGO-related scripts.