micropython/micropython #11970: machine.Timer wdt timeout panic

shariltumin, 2023-07-08:
The machine.Timer module seems to be broken with firmware compiled with ESP-IDF-5.0.2, previous firmware compiled with ESP-IDF-4.x.x was fine.

```
$ mv ~/Downloads/esp32-20230627-unstable-v1.20.0-261-g813d559bc.bin firmware.bin

$ strings firmware.bin | grep IDF
MicroPython-1.20.0-xtensa-IDFv5.0.2-with-newlib4.1.0

>>> 
MPY: soft reboot
MicroPython v1.20.0-261-g813d559bc on 2023-06-27; ESP32 module with ESP32
Type "help()" for more information.
>>> from machine import Timer
>>> tim0 = Timer(0)
>>> tim0.init(period=5000, mode=Timer.ONE_SHOT, callback=lambda t:print(0))
>>> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
```

mattytrentini, 2023-07-08:
Not sure if it's useful, but the relevant section in the Espressif IDF Migration Guide appears to be [Timer Group Driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/peripherals.html#timer-group-driver). The MicroPython implementation is in [esp32/machine_timer.c](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_timer.c).

mattytrentini, 2023-07-08:
Also see #11948.

dpgeorge, 2023-07-14:
Fixed by 7d66ae603d91157069d7cdee065919a61f678fb9

closed 2023-07-14 by dpgeorge

Walkline80, 2023-07-14:
It's still remains two migration advise in power_on logs after timer driver switch to hal, here's the part of logs

```doc
I (410) cpu_start: Application information:
I (413) cpu_start: Compile time:     Jul 14 2023 08:54:13
I (419) cpu_start: ELF file SHA256:  df996d7a61421412...
I (425) cpu_start: ESP-IDF:          v5.0.2
I (430) cpu_start: Min chip rev:     v0.3
I (435) cpu_start: Max chip rev:     v0.99
I (440) cpu_start: Chip rev:         v0.3
I (444) heap_init: Initializing. RAM available for dynamic allocation:
I (452) heap_init: At 3FCA13D0 len 0003B340 (236 KiB): D/IRAM
I (458) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DIRAM
I (465) heap_init: At 50000830 len 000017D0 (5 KiB): RTCRAM
I (472) spi_flash: detected chip: generic
I (476) spi_flash: flash io: dio

W (480) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (490) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
```

I think both of rmt and ADC also need to be updated :)