micropython/micropython #11974: W5500 Driver is extremely unstable (It is proof of concept quality)

GM-Script-Writer-62850, 2023-07-09:
Firmware: https://micropython.org/download/W5500_EVB_PICO/ - [v1.20.0 (2023-04-26) .uf2](https://micropython.org/resources/firmware/W5500_EVB_PICO-20230426-v1.20.0.uf2) 

For reference this is the test file
```
$ ls -l /var/www/admin/ok.txt 
-rw-r--r-- 1 root root 3 Jul  7 00:56 /var/www/admin/ok.txt
```

I get far better stability out of the [v2.0.0 build from wiznet](https://github.com/Wiznet/RP2040-HAT-MicroPython/releases) (still unstable, but not nearly this bad)

Sample code:
```
from machine import Pin
from time import sleep
from dht import DHT22
from threadsafe import Message # https://github.com/peterhinch/micropython-async/tree/master/v3/threadsafe
import _thread
import urequest # https://github.com/micropython/micropython-lib/issues/546
import network
import uasyncio
from gc import mem_free

print("Hello World")

#hard reset sensor
Pin(7, Pin.OUT, value=0)
Pin(8, Pin.OUT, value=0)
sleep(1)

#setup sensor
Pin(7, Pin.OUT, value=1)#This pin powers the sensor
sleep(2)
sensor=DHT22(Pin(8))

#LED=Pin(25, Pin.OUT)#debug LED (onboard)

nic=network.WIZNET5K()
nic.active(True)
err=1
while err:
	try:
		nic.ifconfig("dhcp")
		err=0
	except Exception as e:
		print("Network error:",e)
print("Network:",nic.ifconfig())

unblock_thread_lock=uasyncio.Lock()
async def unblock(func, *args, **kwargs):#https://github.com/peterhinch/micropython-async/blob/master/v3/docs/THREADING.md#4-taming-blocking-functions
	def wrap(func, message, args, kwargs):
		message.set(func(*args, **kwargs)) # Run the blocking function.
	msg = Message()
	await unblock_thread_lock.acquire()
	print(mem_free(),"- Before _thread")
	_thread.start_new_thread(wrap, (func, msg, args, kwargs))
	print(mem_free(),"- After _thread")
	msg=await msg
	print(mem_free(),"- After _thread Message")
	unblock_thread_lock.release()
	return msg

def readDHT(pin):
	print(mem_free(),"- From core 1")
	pin.measure()

def request(url):
	print(mem_free(),"- From core 1")
	# just need url to be sent, do not care about a reply, just yolo it
	urequest.get("http://10.0.0.69:8080/ok.txt")

print()
async def main():
	while True:
		await uasyncio.sleep(5)# simulate time for other stuff to work
		print(mem_free(),"- Before urequest")
		uasyncio.create_task(unblock(request,"http://10.0.0.69:8080/ok.txt")) # test page: 2 byte text file
		await uasyncio.sleep(5)# simulate time for other stuff to work
		print(mem_free(),"- Before DHT22")
		await unblock(readDHT,sensor)
		print(mem_free(),"- After")
		print(sensor.temperature(),"C;",sensor.humidity(),"%")
		print('---------------------')
uasyncio.run(main())
```
Results in hardlock, output in thonny
```
Hello World
('10.0.0.180', '255.255.255.0', '10.0.0.1', '10.0.0.1')

172256 - Before urequest
171872 - Before _thread
167696 - After _thread
167680 - From core 1
164496 - After _thread Message
164464 - Before DHT22
164112 - Before _thread
159936 - After _thread
159920 - From core 1
159712 - After _thread Message
159712 - After
24.2 C; 53.0 %
---------------------
159680 - Before urequest
159296 - Before _thread
155120 - After _thread
155104 - From core 1
151952 - After _thread Message
151920 - Before DHT22
151568 - Before _thread
147392 - After _thread
147376 - From core 1
147168 - After _thread Message
147168 - After
24.2 C; 53.0 %
---------------------
147136 - Before urequest
146752 - Before _thread
142576 - After _thread
142560 - From core 1
139408 - After _thread Message
139376 - Before DHT22
139024 - Before _thread
134848 - After _thread
134832 - From core 1
134624 - After _thread Message
134624 - After
24.2 C; 52.9 %
---------------------
134592 - Before urequest
134208 - Before _thread
130032 - After _thread
130016 - From core 1
126864 - After _thread Message
126832 - Before DHT22
126480 - Before _thread
122304 - After _thread
122288 - From core 1
122080 - After _thread Message
122080 - After
24.2 C; 52.8 %
---------------------
122048 - Before urequest
121664 - Before _thread
117488 - After _thread
117472 - From core 1
114320 - After _thread Message
114288 - Before DHT22
113936 - Before _thread
109760 - After _thread
109744 - From core 1
109536 - After _thread Message
109536 - After
24.2 C; 52.8 %
---------------------
109504 - Before urequest
109120
```
------------------------------------------------
Sample code 2:
```
from machine import Pin
from time import sleep
from dht import DHT22
import urequest # https://github.com/micropython/micropython-lib/issues/546
import network
import uasyncio
from gc import mem_free

print("Hello World")

#hard reset sensor
Pin(7, Pin.OUT, value=0)
Pin(8, Pin.OUT, value=0)
sleep(1)

#setup sensor
Pin(7, Pin.OUT, value=1)#This pin powers the sensor
sleep(2)
sensor=DHT22(Pin(8))

#LED=Pin(25, Pin.OUT)#debug LED (onboard)

nic=network.WIZNET5K()
nic.active(True)
err=1
while err:
	try:
		nic.ifconfig("dhcp")
		err=0
	except Exception as e:
		print("Network error:",e)
print("Network:",nic.ifconfig())

def readDHT(pin):
	print(mem_free(),"- From core 0")
	pin.measure()

def request(url):
	print(mem_free(),"- From core 0")
	# just need url to be sent, do not care about a reply, just yolo it
	urequest.get("http://10.0.0.69:8080/ok.txt")

print()
async def main():
	while True:
		await uasyncio.sleep(5)# simulate time for other stuff to work
		print(mem_free(),"- Before urequest")
		request("http://10.0.0.69:8080/ok.txt")
		await uasyncio.sleep(5)# simulate time for other stuff to work
		print(mem_free(),"- Before DHT22")
		readDHT(sensor)
		print(mem_free(),"- After")
		print(sensor.temperature(),"C;",sensor.humidity(),"%")
		print('---------------------')

uasyncio.run(main())
```
Results in crash, output in thonny:
```
Hello World
Network: ('10.0.0.180', '255.255.255.0', '10.0.0.1', '10.0.0.1')

179168 - Before urequest
179168 - From core 0
176224 - Before DHT22
176224 - From core 0
176224 - After
24.1 C; 51.7 %
---------------------
176192 - Before urequest
176192 - From core 0
173248 - Before DHT22
173248 - From core 0
173248 - After
24.1 C; 51.6 %
---------------------
173216 - Before urequest
173216 - From core 0
170272 - Before DHT22
170272 - From core 0
170272 - After
24.0 C; 51.6 %
---------------------
170240 - Before urequest
170240 - From core 0
167296 - Before DHT22
167296 - From core 0
167296 - After
24.0 C; 51.6 %
---------------------
167264 - Before urequest
167264 - From core 0
164320 - Before DHT22
164320 - From core 0
164320 - After
24.0 C; 51.8 %
---------------------
164288 - Before urequest
164288 - From core 0
Traceback (most recent call last):
  File "<stdin>", line 58, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 49, in main
  File "<stdin>", line 41, in request
  File "/home/chad/ownCloud/PICO/garage PiCO/lib/urequest.py", line 183, in get
  File "/home/chad/ownCloud/PICO/garage PiCO/lib/urequest.py", line 86, in request
OSError: [Errno 12] ENOMEM
```
---------------------------------
Sample Code 3:
```
from machine import Pin
from time import sleep
from dht import DHT22
import urequest # https://github.com/micropython/micropython-lib/issues/546
import network
import uasyncio
from gc import mem_free

print("Hello World")

nic=network.WIZNET5K()
nic.active(True)
err=1
while err:
	try:
		nic.ifconfig("dhcp")
		err=0
	except Exception as e:
		print("Network error:",e)
print("Network:",nic.ifconfig())

def request(url):
	print(mem_free(),"- From core 0")
	# just need url to be sent, do not care about a reply, just yolo it
	urequest.get("http://10.0.0.69:8080/ok.txt")

print()
async def main():
	while True:
		await uasyncio.sleep(5)# simulate time for other stuff to work
		print(mem_free(),"- Before urequest")
		request("http://10.0.0.69:8080/ok.txt")
		print(mem_free(),"- After")
		print('---------------------')

uasyncio.run(main())

```
Results in crash, output in thonny:
```
Hello World
Network: ('10.0.0.180', '255.255.255.0', '10.0.0.1', '10.0.0.1')

179664 - Before urequest
179664 - From core 0
176720 - After
---------------------
176720 - Before urequest
176720 - From core 0
173776 - After
---------------------
173776 - Before urequest
173776 - From core 0
170832 - After
---------------------
170832 - Before urequest
170832 - From core 0
167888 - After
---------------------
167888 - Before urequest
167888 - From core 0
164944 - After
---------------------
164944 - Before urequest
164944 - From core 0
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 32, in main
  File "<stdin>", line 25, in request
  File "/home/chad/ownCloud/PICO/garage PiCO/lib/urequest.py", line 183, in get
  File "/home/chad/ownCloud/PICO/garage PiCO/lib/urequest.py", line 86, in request
OSError: [Errno 12] ENOMEM
```

GM-Script-Writer-62850, 2023-07-09:
Sample code:
```
from machine import Pin
from time import sleep
from dht import DHT22
import _thread
from threadsafe import Message
import urequest # https://github.com/micropython/micropython-lib/issues/546
import network
import uasyncio

print("Hello World")

nic=network.WIZNET5K()
nic.active(True)
err=1
while err:
	try:
		nic.ifconfig("dhcp")
		err=0
	except Exception as e:
		print("Network error:",e)
cfg=nic.ifconfig()
print("Network:",cfg)

Pin(7, Pin.OUT, value=1)# This pin powers the sensor
sleep(2)
sensor=DHT22(Pin(8))

led=Pin(25,Pin.OUT,value=0)#onboard LED

unblock_thread_lock=uasyncio.Lock()
async def unblock(func, *args, **kwargs):#https://github.com/peterhinch/micropython-async/blob/master/v3/docs/THREADING.md#4-taming-blocking-functions
	def wrap(func, message, args, kwargs):
		message.set(func(*args, **kwargs)) # Run the blocking function.
	msg = Message()
	await unblock_thread_lock.acquire()
	_thread.start_new_thread(wrap, (func, msg, args, kwargs))
	msg=await msg
	unblock_thread_lock.release()
	return msg

def request(url):
	# just need url to be sent, do not care about a reply, just yolo it
	led.on()
	r=urequest.get("http://10.0.0.69:8080/ok.txt")
	print(r.status_code)
	r.close()
	led.off()

print()
async def main():
	await unblock(sensor.measure)
	print(sensor.temperature(),"C;",sensor.humidity(),"%")
	lp=0
	while True:
		await uasyncio.sleep(1)# simulate time for other stuff to work
		await unblock(request,"http://10.0.0.69:8080/ok.txt") # test page: 2 byte text file
		print('----------',lp,'-----------')
		lp+=1

uasyncio.run(main())
```
output in thonny (another hard lock):
```
Hello World
Network: ('10.0.0.180', '255.255.255.0', '10.0.0.1', '10.0.0.1')

23.7 C; 52.2 %
200
---------- 0 -----------
200
---------- 1 -----------
200
---------- 2 -----------
200
---------- 3 -----------
200
---------- 4 -----------
200
---------- 5 -----------
200
---------- 6 -----------
200
---------- 7 -----------
200
```

GM-Script-Writer-62850, 2023-07-09:
Looks like is i run `gc.collect()` after `sensor.measure()` it does not lockup? Is this a out memory overflow error?

andrewteta, 2023-10-10:
I didn't do nearly as much testing as you, but I did run into some issues while using the Pico Wiznet W5500 board that looked really similar. ENOMEM, lockups, etc. until I strategically placed a gc.collect() in my loop. Now my code is very stable. So it definitely seems memory related.

GM-Script-Writer-62850, 2023-10-10:
tried `gc.collect()` in many places to no avail, the lockup is random, running my code a single core does not crash, if you run sensor.measure() on core 1 (the second core)  if you do so much as run it a single time after enabling the NIC it it will hard lock, not even disabling the NIC while using that on core 1 helps, my PICO W never does this

jonnor, 2024-08-24:
Is this issue still present in MicroPython 1.23 ?

GM-Script-Writer-62850, 2024-08-24:
Currently running a test, i did notice a update that mentioned a possible hard-lock when using core 2 in the change logs, i have been hoping that would fix this issue.

ok i think i let it run long enough, over 3300 well over 100x long than it would run before

dpgeorge, 2024-08-27:
Sounds like this has been fixed, so I'll close this issue.

Feel free to reopen and provide further details on how to reproduce the problem, if there's still a problem.

closed 2024-08-27 by dpgeorge

GM-Script-Writer-62850, 2024-08-27:
I have 2 days of uptime and counting on 2 W5500 PICOs w/ production code, given how bad stability was before, the situation is at the very least an order of magnitude better

i am guessing this is what fixed it based on the change logs:
https://github.com/micropython/micropython/releases/tag/v1.22.1