micropython/micropython #11976: ESP-DSP

street-grease-coder, 2023-07-09:
There's an entire section of esp-dsp already implemented as a component, making things like kalman filter super easy to access, and I would guess better in performance than implementations in micropython. 
Is there any option to include it micropython builds currently, and if no, would you consider implementing it? 
https://github.com/espressif/esp-dsp

ricksorensen, 2023-07-10:
There is a module that can be compiled into many micropython implementations, including ESP-32: [ulab](https://github.com/v923z/micropython-ulab).  
It is a (smallish) subset of numpy and scipy - it does have vector operations, linear algebra, FFT, etc .. see docs.  It does not have filter design functions nor a kalman filter implementation.  

Maybe ulab it self can take advantage of some of the functions in the ESP-DSP sdk.

jonnor, 2024-07-03:
This could be done as one or more dynamic native modules (.mpy), or as a user C module. So these features could be implemented separately from core MicroPython. The mentioned ulab is an example of a user C module for DSP type functionality. And https://github.com/emlearn/emlearn-micropython is an example of dynamic native modules, that has FFT, IIR filters etc.

People are also interested in modules for ARM CMSIS DSP: https://github.com/orgs/micropython/discussions/10200