micropython/micropython #11977: WLAN functions fail on Pico Pi W core1

ayjaym, 2023-07-09:
Reproducible on 1.20 and latest nightly build (as of today)
Attempting to set up the Wifi module on core1 fails at the line ap.active(True) below which hangs. Code works fine on core0
Guarding code with a lock does not resolve the issue.

    ssid = 'MicroPython-AP'
    password = '123456789'
    ap = network.WLAN(network.AP_IF)
    ap.config(essid=ssid, password=password)
    print("status = ", ap.status())
    ap.active(True)
    print(ap.ifconfig())

Chaostheorie, 2023-07-29:
Hi, I can reproduce this issue only, and only if, I don't set any country codes beforehand. Otherwise your code snippet works. 

To set the country codes (adjust to your current country):
```python
import rp2
import network

country = "DE"

rp2.country(country)
network.country(country)

# --- <networking code>
ssid = 'MicroPython-AP'
password = '123456789'
ap = network.WLAN(network.AP_IF)
ap.config(essid=ssid, password=password)
print("status = ", ap.status())
ap.active(True)
print(ap.ifconfig())
```

ayjaym, 2023-07-29:
Well, darn it, I would never have thought to try that! I'm away from home tonight but will definitely check this out tomorrow. Of course unfortunately I've hit another independent thread showstopper which is that an input pin irq asserted on core 0 with any thread running on core 1 causes a lockup. And I think this is also true if the irq is asserted on core 1 with core 0 running something else. But fixing the wlan issue this way would certainly help, thanks so much for a very clever idea!

ayjaym, 2023-08-08:
HIya, I finally got round to trying this. You are correct, it no longer hangs on configuration but it still doesn't receive UDP datagrams, although the same code works perfectly if running on Core 0. So we still have show-stopping issues. Plus, an unrelated issue, IRQs asserted on Core 0 will lock up the machine if anything's running on Core 1. I had no idea Micropython threading support was so immature. This is incredibly disappointing as my whole open source project is stalled and I can't share it with the community. Ah well, thanks so much for your assistance!

Chaostheorie, 2023-08-11:
Happy to hear it at least the start worked. Could you please share a snippet/ example of the UDP-related code? I'm not very experienced with MicroPython's networking code but would love to try fix any issues. 

My current project also builds on the Pico W so any bugs fixed would be nice.

ayjaym, 2023-08-11:
> @ayjaym Regarding IRQs on dual core you might like to read this.
>
> tl;dr Python (including MicroPython) relies on the global interpreter lock (GIL) to guarantee the integrity of objects. On dual core there is no GIL: there are two separate Python instances whose shared state is not GIL protected. Great care needs to be taken where data is shared to avoid corruption or crashes.

Thanks and of course you are quite correct about the GIL but my test cases involve either a thread with just a while True loop on core 1 for the IRQ hang on core 0 scenario or just running the WLAN code on core 1 with nothing running on core 0 (which is my second issue I have raised separately), where UDP datagrams are never received but the same code or core 0 works perfectly.I can't immediately see how the GIL could be an issue in these two scenarios, though of course perhaps there is some subtle issue I'm not aware of.

ayjaym, 2023-08-11:
[rtpmidi.zip](https://github.com/micropython/micropython/files/12323270/rtpmidi.zip)

The attached zip file is a full working (but stripped down) implementation of the RTP Midi protocol (which uses UDP datagrams on two ports) and demonstrates the issue. The rtpmidi.py file is expected to be in the /lib folder on the Pico and rtpmiditest.py is a test harness which transmits and receives MIDI messages.

You can either run the rtpmidi library on core 0 or core 1 from the rtpmiditest code, as you can see at the bottom of the file.

If you download the free RTPMidi utility for Windows and create a remote peer on 192.168.4.1 port 5004 you will see that when the code is run on core 0, and you then connect to the Pico AP that appears, then you can subsequently connect with RTPMidi and a latency figure appears showing that connection was successful.

[If you were in a DAW like Reaper you could then send and receive MIDI messages but the RTPMidi protocol itself sends a regular flow of sync messages between the Windows host and the Pico so that's not necessary to just test it.]

However if the same code is run from core 1, the poll calls never receive the UDP datagrams.

Note: this code is intended for release in an open source project but it may be useful even as it is - it is a stripped down RTPMidi implementation that allows wireless MIDI to a Pico Pi. Note that commented out references to an encoder were intended to add checks for an IRQ-driven rotary encoder (from another library). However an unrelated multicore issue means that IRQs asserted on core 0 with anything at all running on core 1 (even just a while true: loop) will hang, and the library author has confirmed this. Frankly, Micropython multi-core support is very immature and I was very surprised to see this given that the RP2040 and the ESP32 have been available now for several years.

I would love to fix this, although the IRQ issue is unfortunately an equally big showstopper. I need the two cores because MIDI is a time-sensitive protocol. In the project, core 1 will do the wireless MIDI and the rotary encoder because being IRQ-driven that's quick. Unfortunately it looks like Micropython's 'schedule' doesn't (yet) have thread affinity because what you would ideally want to do is schedule all the callbacks on core 0. This ensures that time-critical code runs independently on core 1 and then code like reading the touchscreen (which requires about 20ms, during which we take a set of averaged readings via several ADC reads), and writing to the OLED display, can be done on core 0 along with any code the user writes as the project is a generally programmable MIDI interface.

peterhinch, 2023-08-11:
@ayjaym 

> Plus, an unrelated issue, IRQs asserted on Core 0 will lock up the machine if anything's running on Core 1. I had no idea Micropython threading support was so immature.

This isn't the case in general. If you can provide a simple repro we'll investigate, but the following script works on the Pico (link pins 16 and 17 to test):
```python
import uasyncio as asyncio
import machine
import _thread
import time
p0 = machine.Pin(17,  mode=machine.Pin.IN)
p1 = machine.Pin(16,  mode=machine.Pin.OUT)

def callback(p):
    print("callback")
 
_ = p0.irq(trigger=machine.Pin.IRQ_RISING | machine.Pin.IRQ_FALLING, handler=callback)

async def twiddle():
    while True:
        p1(not p1())
        await asyncio.sleep_ms(600)

def foo():
    while True:
        time.sleep(1)
        print("foo")

_thread.start_new_thread(foo,())

asyncio.run(twiddle())
```
Outcome:
```
callback
callback
foo
callback
callback
foo
callback
foo
callback
callback
foo
callback
callback
foo
```

ayjaym, 2023-08-11:
Mike Teachman, the author of the rotary encoder library, was able to reproduce my issue with his library, see here

https://github.com/miketeachman/micropython-rotary/issues/27#issuecomment-1668885053

ayjaym, 2023-08-11:
and can I just say I would be incredibly grateful if somehow these issues were resolved because my hardware is working perfectly, I want to share the project with the community and if I could only get multicore stuff working it would be awesome. 
[I would also have liked MIDI over USB but settled for getting it working over WiFi because as I understand it the USB MIDI work in Micropython is quite a big task and it's not clear when that'll get done. And CircuitPython, which does have it, doesn't support threads and has timing issues, so that wasn't an option either]

peterhinch, 2023-08-11:
@ayjaym Please see https://github.com/miketeachman/micropython-rotary/issues/27#issuecomment-1675169433

ayjaym, 2023-08-11:
Yes, just seen that. I'm pretty sure I first saw the problem with code running on core 0 though, but I'll give it a test. Thanks. Should it not work though with core 0 idle?.

peterhinch, 2023-08-12:
If the code on core 0 runs to completion, MicroPython attempts to return to the REPL. I'm unsure what state it's in at that point with core 1 still running but I don't think it's pretty. In a real application core 0 will be busy until the application terminates or is cancelled.

scruss, 2023-11-03:
> However if the same code is run from core 1, the poll calls never receive the UDP datagrams.

It's my understanding that code that accesses the wifi hardware must run on core 0. I can't find any official documentation that supports this — and given the large CloudFlare outage, it's hard to find *anything* on the internet today.