micropython/micropython #11988: corrupted json, no error raised

kr-g, 2023-07-11:
the following json is processed without any error report using mpy v1.20.0

```
import json

j1 = """
{
    "a" : [["c", ["d"]]],
    
    "b": {
            "bb": 0
    }
}
"""

o1 = json.loads(j1)

# here one closing ] is missing in line with "a"
j2 = """
{
    "a" : [["c", ["d"]],
    
    "b": {
            "bb": 0
    }
}
"""

o2 = json.loads(j2)  # <== this should raise an error

```

in cpython following error is raised 

```
Traceback (most recent call last):
  File "/home/benutzer/repo/misc/mp_async_pin/bug.py", line 26, in <module>
    o2 = json.loads(j2)
  File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.10/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 5 column 8 (char 40)

```

jimmo, 2023-07-12:
Thanks @kr-g 

This is a "known issue" of sorts -- see https://github.com/micropython/micropython/blob/master/extmod/modjson.c#L118 -- in order to minimise code size and memory usage the additional state that would be required to handle this is not tracked.

Whether this is the right trade-off to make... I'm not so sure. Silently giving incorrect parsing seems a bit scary. It might be worth spending some time seeing how much it would cost to improve this.

kr-g, 2023-07-12:
this also works without any error raised...

```
# here scenario of "half" written file on disc
j3 = """
{
    "a" : [["c", ["d"]]],
    
    "b": {
            "bb": 0
    }

"""

o3 = json.loads(j3)  # <== this should raise an error

```

kr-g, 2023-07-12:
somehow last example was expectable since it is more or less same as second

yourcaptain, 2023-07-24:
> 

I tried it and didn't see any error. (mpy v1.20.0, via port unix)

kr-g, 2023-07-25:
due to lack of time i m closing this.
feel free to merge or start from scratch with another solution, 
or leave it out completely by decision.

closed 2023-07-25 by kr-g