micropython/micropython #11990: WebAssembly/PyScript - function argument count mismatch.

ntoll, 2023-07-11:
Hi @dpgeorge - another odd bug from the version of MicroPython served from here: https://cdn.jsdelivr.net/npm/@micropython/micropython-webassembly-pyscript@1.20.0-268/micropython.mjs

Consider the following example PyScript application: https://3fd38942-d35c-4bf6-9b8a-37cec059f572.pyscriptapps.com/a49fcee5-50ee-452a-94e5-2da033723255/latest/

Clicking the "Start Camera" button sets up the video element properly. :tada:

The problem happens when the "Take photo" button is clicked.

```
File "<stdin>", line 18, in photo_click
TypeError: function expected at most 2 arguments, got 5
MicroPython exception: Traceback (most recent call last):
  File "<stdin>", line 18, in photo_click
TypeError: function expected at most 2 arguments, got 5
Uncaught (in promise) PythonError: Traceback (most recent call last):
  File "<stdin>", line 18, in photo_click
TypeError: function expected at most 2 arguments, got 5
```

Here's the code that's called:

```
def photo_click(e):
    video = js.document.querySelector("#video")
    canvas = js.document.querySelector("#canvas")
    context = canvas.getContext('2d')
    js.console.log(context)
    context.drawImage(video, 0, 0, canvas.width, canvas.height)  # This is line 18
    image_data_url = canvas.toDataURL('image/jpeg')
    js.console.log(image_data_url)
```

Clearly, MicroPython is, in some way, calling `context.drawImage` with the wrong number of args. However, when I look in MDN, for a description of the function  (https://developer.mozilla.org/en-US/docs/Web/API/CanvasRenderingContext2D/drawImage) I notice that the five arg call signature is valid, and that the assertion that the function is expecting but two arguments is strange because the function signatures described in the docs don't include any with just two arguments. I suspect something going awry in the FFI that translates between a function with overloaded numbers of arguments (but this is very much only a hunch).

You can see the source code for the complete project here: https://pyscript.com/view/3fd38942-d35c-4bf6-9b8a-37cec059f572/a49fcee5-50ee-452a-94e5-2da033723255/latest/ (see `main.py`).

ntoll, 2023-07-11:
FWIW - this code works with Pyodide.

dpgeorge, 2023-07-11:
Thanks for the report.  I'll have to investigate it.

The code for all of this is (still) on a branch here: https://github.com/dpgeorge/micropython/tree/webassembly-add-modjs

dpgeorge, 2023-07-11:
I could easily reproduce the issue by calling any JS function (from Python) with 3 or more arguments.  The issue was that only up to 2 arguments was supported (!).

That's now fixed (on the branch referenced above), and available at https://www.npmjs.com/package/@micropython/micropython-webassembly-pyscript/v/1.20.0-269

closed 2023-07-11 by dpgeorge