micropython/micropython #12015: esp32 build error ESP_IDF v5.2.0

heikokue, 2023-07-15:
I got two build errors:
1. `/home/heiko/esp/esp-idf/components/app_update/include/esp_ota_ops.h:16:10: fatal error: esp_bootloader_desc.h: No such file or directory`
2. `/home/heiko/esp/micropython/ports/esp32/uart.c:70:24: error: 'uart_periph_signal' undeclared (first use in this function)`

To fix the first error `esp_bootloader_format` has to be added to `esp32_common.cmake`
To fix the second error `#include "soc/uart_periph.h"` has to be added to `uart.c`

I will create a pull request

ma261065, 2023-07-16:
Isn't 5.1 the latest ESP-IDF release?

https://github.com/espressif/esp-idf/releases/tag/v5.1

mattytrentini, 2023-07-16:
On master, IDF v5.0.2 is the latest supported build. See [README.md](https://github.com/micropython/micropython/blob/master/ports/esp32/README.md) which will continue to be updated with the latest supported version.

If your fixes can resolve those issues _and continue to work on v5.0.2_ (and, ideally, also v5.1.0) that would be a welcome fix! However, v5.2.0 has not yet been released and is still under active development. It's not appropriate for us to use for mainline MicroPython builds.

heikokue, 2023-07-17:
OK, builds with v5.0.2 and v5.1 are working

mattytrentini, 2023-07-18:
> OK, builds with v5.0.2 and v5.1 are working

Does building with v5.1 work on master? Or does it require any changes?

If it's the former, I can try deploying to an ESP32-C6 (which _requires_ v5.1)...

IhorNehrutsa, 2023-07-20:
I'm testing **PR esp32/PWM: Reduce inconsitencies between ports #10854** on ESP IDF 5.0.2, 5.0.3, 5.1
See https://github.com/micropython/micropython/pull/10854#issuecomment-1639482899

alingse, 2023-08-18:
+1 same error

riwalker, 2023-08-26:
+1 same error, on 5.1.1, with esp32c6
adding the line fixes the issue: #include "soc/uart_periph.h" has to be added to uart.c

chaseadam, 2024-04-02:
support has been added for 5.2.0 https://github.com/micropython/micropython/commit/d6176c1f5e5556e25944ce52874ea2f2d82ccde5

jonnor, 2024-07-13:
This should have been fixed in the commit above, which is included in MicroPython 1.23.0. So I believe this issue can be closed.

closed 2024-07-13 by robert-hh