micropython/micropython #12024: esp8266-12f 4M flash need be supported

woodgreat, 2023-07-16:
esp8266-12f 4M flash need be supported . My develop board main MCU is esp8266-12f ,But I checked the download page of our main site ,MicroPython.org ,we just support flash 2M of 8266 board firmware ,So can use add 4M version firmware ? thank you !

woodgreat, 2023-07-19:
I am use thonny to fire thw frimware into ESP8266 ,so I tried many ways to solve the problem till one : 
adjust the setting  with our 2M support frimware(esp8266-20230426-v1.20.0.bin)

[image]

then It work !
...but the question still here ,We need a frimware point to the specific MCU chips ,    : )

ccarson1, 2025-01-22:
woodgreat this was very helpful!