micropython/micropython #12033: Negative float/double constants fail.

travisbarney, 2023-07-17:
Should negative float/double constants work?

Negative const example (fails):

>>> MYCONST = const(-2.96285213128e-2)
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: not a constant

------------------------------

Positive const example (works):

>>> MYCONST = const(2.96285213128e-2)
>>> print(MYCONST)
0.0296285213128
>>> 

MicroPython v1.20.0-283-g7d66ae603-dirty on 2023-07-17

jonnor, 2024-09-01:
Reproduced on a recent MicroPython (commit 9e12fc9). This seems like a bug! Fixes welcomed

To be clear, the use of scientific notation is not important. 

```
MYCONST = const(-2.96285213128)
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: not a constant
```

dpgeorge, 2024-09-02:
The lexer treats `-` as a separate token to the number, so `-1.2` is actually parsed and compiled as `negate(1.2)`, and then the float is negated at runtime by the VM.

Hence, it's not actually a constant!

For integers it's the same thing, except the parser has some special logic to do the negation at compile time.

We could probably extend the parser so it negates floats at compile time as well.