micropython/micropython #12034: UnicodeError when printing ssid of WLAN interface

maxi07, 2023-07-17:
Hi,
I am having trouble reading the connected SSID or the rssi with strange results. I am using a raspberry pico with micropython 1.20.0 `MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico W with RP2040`
This is my REPL using micropython

```python
>>> import network
>>> sta_if_wlan = network.WLAN(network.STA_IF)
>>> sta_if_wlan.isconnected()
True
>>> sta_if_wlan.config('ssid')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeError: 

>>> sta_if_wlan.config('channel')
11
>>> sta_if_wlan.status('rssi')
536956352
>>> sta_if_wlan.status('rssi')
-61
>>> sta_if_wlan.config('ssid')
MyNetwork48
>>> sta_if_wlan.config('ssid')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeError: 
```

The commands are executed without big time difference. Sometimes it executes correctly, sometimes it returns other results than expected. As nothing really changes during the commands and only a few seconds pass, I cannot reproduce this behaviour every time.

dpgeorge, 2023-07-18:
This is quite strange, especially getting invalid rssi values.

What is the actual, correct SSID of the network that it's connected to?

Can you please try the latest unstable/nightl firmware from https://micropython.org/download/rp2-pico-w/?

maxi07, 2023-07-20:
Still occurs from time to time, but is there any way to debug this issue further?
The provided SSID is the actual SSID. I cannot really reproduce the issue as it randomly occurs.

dpgeorge, 2023-07-20:
If you can rebuild from source, make this change:
```diff
--- a/extmod/network_cyw43.c
+++ b/extmod/network_cyw43.c
@@ -393,7 +393,7 @@ STATIC mp_obj_t network_cyw43_config(size_t n_args, const mp_obj_t *args, mp_map
                 if (self->itf == CYW43_ITF_STA) {
                     uint8_t buf[36];
                     cyw43_ioctl(self->cyw, CYW43_IOCTL_GET_SSID, 36, buf, self->itf);
-                    return mp_obj_new_str((const char *)buf + 4, nw_get_le32(buf));
+                    return mp_obj_new_bytes((const char *)buf + 4, nw_get_le32(buf));
                 } else {
                     size_t len;
                     const uint8_t *buf;

```

That will prevent the error.  Then report here what the return value is when it's not the correct SSID.

jonnor, 2024-09-01:
Information requested above, needed to be able to debug/fix the issue.