micropython/micropython #12037: Data loss in string transfer through sys.stdout.write

pchala, 2023-07-18:
Env: MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico with RP2040

I have noticed constant data loss when transfer string objects with help of sys.stdout.write() method.

MCU code:
```
import sys

BUF_SIZE = 10000

while True:
    sys.stdin.readline()
    line = bytes(BUF_SIZE).hex() + '\n'
    sys.stdout.write(line)
```
PC code:
```
import serial
connection = serial.Serial('COM6', timeout=3)
for i in range(1000):
    connection.write(b'\n')
    res = connection.readline()
    print(i, len(res))
```

If BUF_SIZE set to 8000 everything will work fine.

In case of sending bytes:
```
import sys

BUF_SIZE = 40000

while True:
    sys.stdin.readline()
    line = bytes(BUF_SIZE)
    sys.stdout.write(line)

```
transfer is ok even for 40000 bytes.

robert-hh, 2023-07-18:
Questions:
- In the BUF_SIZE=10000  you have used bytes(x).hex(). So you would send 20001 bytes. In the BUF_SIZE=40000 you have just sent the bytes object.
- can you tell whether the data is not sent or not received?
- are the differences between the various sizes repeatable?
- has changing the timeout at the receive side any effect?
- makes using sys.stdout.buffer.write() any difference?

pchala, 2023-07-18:
> In the BUF_SIZE=10000 you have used bytes(x).hex().
> So you would send 20001 bytes. In the BUF_SIZE=40000 you have just sent the bytes object.

Yes, in case of sending bytes I have tested up to 50k and no issues are observed.

>can you tell whether the data is not sent or not received?

Some data lost at receiver.

For example output of test looks like:
(py310) λ python client.py
```
0 20002
1 20002
2 20002
3 19874
4 19874
5 19874
6 20002
7 20002
8 19874
9 19874
10 20002
11 19874
12 20002
13 19874
14 19874
15 19874
16 20002
17 19874
18 20002
19 20002
20 19874
21 20002
```

> are the differences between the various sizes repeatable?

Yes, see above (128 bytes)

> has changing the timeout at the receive side any effect?

No, looks exactly same

>makes using sys.stdout.buffer.write() any difference?

No, looks exactly same. I have check even sys.stdout.buffer.buffer.write

dpgeorge, 2023-07-19:
What is the PC side, is it a Windows, Linux, Mac or other?  Windows is known to have bad serial drivers.

dpgeorge, 2023-07-19:
Running the above test (on Linux with /dev/ttyACM0 instead of COM6), using `mpremote run --no-follow mcu.py` to run the code on the rp2, it works OK for me.  The PC side outputs:
```
$ python pc.py
0 20002
1 20002
2 20002
...
997 20002
998 20002
999 20002
```

pchala, 2023-07-19:
> What is the PC side, is it a Windows, Linux, Mac or other? Windows is known to have bad serial drivers.

Yes it is Windows 10. Still strange that sending bytes has no issues while sending string does.

robert-hh, 2023-07-19:
> Still strange that sending bytes has no issues while sending string does.

Can you try sending a non-zero filled bytes object like `b'0123456789'*2000`?
Still strange that the data loss is always exactly 128 bytes. Looks like a full buffer in the USB link is missing.

pchala, 2023-07-19:
I have add checks
MCU code:
```
import sys

BUF_SIZE = 10 * 1024

buf = bytes()

for i in range(BUF_SIZE//0x100):
    buf += i.to_bytes(1, 'big') * 0x100

while True:
    sys.stdin.readline()
    line = buf.hex() + '\n'
    sys.stdout.write(line)
```
PC
```
import serial
connection = serial.Serial('COM6', timeout=5)

def chunk(buf, chunk_size):
    for i in range(0, len(buf), chunk_size):
        yield buf[i:i + chunk_size]

for i in range(10):
    connection.write(b'\n')
    res = str(connection.readline().strip(), 'ascii')
    print(i, len(res))
    for i, s in enumerate(chunk(res, 0x200)):
        if f'{i:0>2x}' * 0x100 != s:
            print((i, s))
            break
```

Result, corrupted stuff just after 8k:
(py310) λ python client.py
0 20352
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
1 20352
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
2 20352
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
3 20416
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202121212121212121212121212121212121212121212121212121212121212121')
4 20352
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
5 20288
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
6 20352
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
7 20352
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')
8 20352
(32, '2020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121 [... cut ...]
(32, '20202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202021212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121212121')

fabse-hack, 2024-10-06:
do you try sys.stdout.flush() ?

fabse-hack, 2024-10-06:
do you try sys.stdout.flush() ??