micropython/micropython #12048: STM32H5 machine.ADC core temperature sensor not working

renestraub, 2023-07-20:
# Description

While testing with an STM32H5 Nucleo board I noticed the internal ADC channels are not working as expected. External (I/O) pins return expected values. Internal ADC input are not reporting reasonable values.

I understand the STM32H5 port is under work and some features not working. I hope my input helps to fix a problem. Let me know if I can help with more testing.

Problem description:
* ADC(ADC.CORE_TEMP) returns incorrect values. 
``` py
>>> from machine import ADC
>>> coretemp=ADC(ADC.CORE_TEMP)
```

Expected result:
* Values should lie around 30°C calibration point, or slightly higher.
``` py
>>> from machine import mem16
>>> mem16[0x08FFF814]
760
>>> mem16[0x08FFF818]
1014
```

Observed result:
* ADC value incorrect, indicates wrong core temperature
* Value is far below 30°C calibration point
``` py
>>> coretemp.read_u16() >> 4
173
```

See detailed log below.

The issue could be in machine_adc.c, adc_config_channel(). ADC object shows `channel` is an encoded number containing additional information besides the ADC channel number. For `ADC.CORE_TEMP` channel is `0xc3210000`. Using this channel number value leads to incorrect loading of the SMPR register. 

Note: The same kind of problem will be present for ADC.CORE_VREF, ADC.CORE_VBAT.

``` py
>>> coretemp=ADC(ADC.CORE_TEMP)
>>> coretemp
<ADC1 channel=3273719808>
>>> hex(3273719808)
'0xc3210000'
```

For G4 cores the channel number is extracted via `__LL_ADC_CHANNEL_TO_DECIMAL_NB`. Maybe this is required for H5 cores as well.

``` sh
STATIC void adc_config_channel(ADC_TypeDef *adc, uint32_t channel, uint32_t sample_time) {
    [removed]

    #if defined(STM32G4)
    channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
    adc->DIFSEL &= ~(1 << channel); // Set channel to Single-ended.
    #endif
    adc->SQR1 = (channel & 0x1f) << ADC_SQR1_SQ1_Pos | (1 - 1) << ADC_SQR1_L_Pos;
    __IO uint32_t *smpr;
    if (channel <= 9) {
        smpr = &adc->SMPR1;
    } else {
        smpr = &adc->SMPR2;
        channel -= 10;
    }
    *smpr = (*smpr & ~(7 << (channel * 3))) | sample_time << (channel * 3); // select sample time

    #endif
}
```

# Version Information

``` sh
% git lg1
* 7d66ae603 - (9 days ago) esp32/machine_timer: Switch from legacy driver to timer HAL. - Damien George (HEAD -> master, origin/master, origin/HEAD)
* 671b35cea - (2 weeks ago) py/builtinimport: Fix built-in imports when external import is disabled. - Jim Mussared
* 606ec9bfb - (9 weeks ago) py/compile: Fix async for's stack handling of iterator expression. - Damien George

MicroPython v1.20.0-282-g671b35cea-dirty on 2023-07-14; STM32H573I-DK with STM32H573IIK3Q
Type "help()" for more information.
```

# Log

``` py
>>> from machine import ADC

### Working external inputs, PF12 taken as example
>>> adcext=ADC(machine.Pin('F12'))
>>> adcext
<ADC1 channel=6>

# Tied to 3.3 V
>>> adcext.read_u16()
65471
>>> adcext.read_u16()
65519
>>> adcext.read_u16()
65503
>>> adcext.read_u16()
65519

# Tied to GND
>>> adcext.read_u16()
48
>>> adcext.read_u16()
0
>>> adcext.read_u16()
16

### Test with core temperature

>>> coretemp=ADC(ADC.CORE_TEMP)
>>> coretemp
<ADC1 channel=3273719808>
>>> hex(3273719808)
'0xc3210000'

# Check low (30°C) and high (130°C) calibration points
>>> from machine import mem16
>>> mem16[0x08FFF814]
760
>>> mem16[0x08FFF818]
1014

# Read sensor, should be between two calibration points, actually close to 30° value
# Right align 12 bit ADC value

>>> coretemp.read_u16() >> 4
173
>>> coretemp.read_u16() >> 4
176
>>> coretemp.read_u16() >> 4
178
>>> coretemp.read_u16() >> 4

# Computes to a core temperature of -199 °C...

>>> (178-760)/((1014-760)/(130-30))+30
-199.1339
```

renestraub, 2023-07-25:
Hello @yn386 

thanks for the quick feedback. I’ll give your changes a try as soon as I am back from holiday next week. 

Can you comment on my observation in the machine ADC driver? I am almost certain the call to __LL_ADC_CHANNEL_TO_DECIMAL_NB in adc_config_channel() is also needed for the H5.

yn386, 2023-07-26:
> I am almost certain the call to __LL_ADC_CHANNEL_TO_DECIMAL_NB in adc_config_channel() is also needed for the H5.

Yes, I think so too.
As you explained above, `__LL_ADC_CHANNEL_TO_DECIMAL_NB` is needed for H5 MCUs.

Also, the ADC channel should be configured as Single-ended.
Now we would be able to implement `adc_config_channel` as follows (same as G4 MCUs):

```c
#if defined(STM32G4) || defined(STM32H5)
channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
adc->DIFSEL &= ~(1 << channel); // Set channel to Single-ended.
#endif
```

renestraub, 2023-07-29:
Hello @yn386,

I gave machine.ADC implementation a try and the changes work. However there is some additional complexity in the handling of the channels, which I am not sure is intended. 

The internal ADC channels come from the STM32 LL driver, which defines a literal with the actual channel in bit position 26. This is why we have to use __LL_ADC_CHANNEL_TO_DECIMAL_NB. So far so good.

``` C
/** @defgroup ADC_HAL_EC_CHANNEL  ADC instance - Channel number
  * @{
  */
/* Note: VrefInt, TempSensor and Vbat internal channels are not available on  */
/*        all ADC instances (refer to Reference Manual).                      */
#define ADC_CHANNEL_0                      (LL_ADC_CHANNEL_0)              /*!< External channel (GPIO pin) ADCx_IN0  */
...
#define ADC_CHANNEL_19                     (LL_ADC_CHANNEL_19)             /*!< External channel (GPIO pin) ADCx_IN19 */
#define ADC_CHANNEL_VREFINT                (LL_ADC_CHANNEL_VREFINT)        /*!< Internal channel VrefInt: Internal
                                           voltage reference, channel specific to ADC1. */
#define ADC_CHANNEL_TEMPSENSOR             (LL_ADC_CHANNEL_TEMPSENSOR)     /*!< Internal channel Temperature sensor,
                                           channel specific to ADC1. */
#define ADC_CHANNEL_VBAT                   (LL_ADC_CHANNEL_VBAT)           /*!< Internal channel Vbat/4: Vbat voltage
                                           through a divider ladder of factor 1/4 to have channel voltage always below
                                           Vdda, channel specific to ADC2. */
#define ADC_CHANNEL_VDDCORE                (LL_ADC_CHANNEL_VDDCORE)        /*!< Internal channel Vddcore, channel
                                           specific to ADC2. */
```

``` C
STATIC const mp_rom_map_elem_t machine_adc_locals_dict_table[] = {
    { MP_ROM_QSTR(MP_QSTR_read_u16), MP_ROM_PTR(&machine_adc_read_u16_obj) },

    { MP_ROM_QSTR(MP_QSTR_VREF), MP_ROM_INT(ADC_CHANNEL_VREF) },
    { MP_ROM_QSTR(MP_QSTR_CORE_VREF), MP_ROM_INT(ADC_CHANNEL_VREFINT) },
    #if defined(STM32G4)
    { MP_ROM_QSTR(MP_QSTR_CORE_TEMP), MP_ROM_INT(ADC_CHANNEL_TEMPSENSOR_ADC1) },
    #else
    { MP_ROM_QSTR(MP_QSTR_CORE_TEMP), MP_ROM_INT(ADC_CHANNEL_TEMPSENSOR) },
    #endif
    #if defined(ADC_CHANNEL_VBAT)
    { MP_ROM_QSTR(MP_QSTR_CORE_VBAT), MP_ROM_INT(ADC_CHANNEL_VBAT) },
    #endif
};
```

However, when a machine.ADC object is created via a Pin object or by specifying an integer __LL_ADC_CHANNEL_TO_DECIMAL_NB must not be used. This requires some ugly construct like this.

``` C
    #elif defined(STM32H5)
    if (channel > 19) {
        // Internal channels CORE_VREF, CORE_TEMP, CORE_VBAT are encoded as literals with
        // ADC channel in bits 26..30.
        channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
    }
    adc->DIFSEL &= ~(1 << channel); // Set channel to Single-ended.
    #endif
```

I am not sure if the mix between integer defined channels and channel literals from `lib/stm32lib/STM32H5xx_HAL_Driver/Inc/stm32h5xx_ll_adc.h` is actually intended. Do you have more insights in this code or should we get George in the discussion?

renestraub, 2023-07-29:
@yn386 
I found a problem with the STM32H5 (and most likely G4) channel constants in the machine.ADC driver. I opened issue #12115. Maybe you want to have a look.

yn386, 2023-07-29:
@renestraub

Oh, you're right. `__LL_ADC_CHANNEL_TO_DECIMAL_NB` must not be used with an integer.

In fact, `__LL_ADC_CHANNEL_TO_DECIMAL_NB` is needed for the internal sensors only.
Therefore, adc_config_channel should implement as follows (it worked with G4 MCU):
```patch
diff --git a/ports/stm32/machine_adc.c b/ports/stm32/machine_adc.c
index fee3d698d..77166f274 100644
--- a/ports/stm32/machine_adc.c
+++ b/ports/stm32/machine_adc.c
@@ -354,7 +354,9 @@ STATIC void adc_config_channel(ADC_TypeDef *adc, uint32_t channel, uint32_t samp
         #endif
     }
     #if defined(STM32G4)
-    channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
+    if (__LL_ADC_IS_CHANNEL_INTERNAL(channel)) {
+        channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
+    }
     adc->DIFSEL &= ~(1 << channel); // Set channel to Single-ended.
     #endif
     adc->SQR1 = (channel & 0x1f) << ADC_SQR1_SQ1_Pos | (1 - 1) << ADC_SQR1_L_Pos;
```
 I will have a look at your issue #12115.

yn386, 2023-07-30:
I had a look at your issue #12115.

> MP_ROM_INT() handles only signed 31 bit integers.

I have a suggestion for MCUs that use bit 31 to define the internal sensors channel as a wordaround for machine.ADC.

At least for G4 and H5 MCUs, the definition of internal sensors use only bit 31-16 and does not use bit 15-0.

Therefore, we can get actual channel by shifing right 16 bits at the definition (`machine_adc_local_dict_table`), and shifting left 16 bits at `machine_adc_make_new`.

The workaround implemetation is as follows (worked with G4).
```patch
diff --git a/ports/stm32/machine_adc.c b/ports/stm32/machine_adc.c
index fee3d698d..3e980214d 100644
--- a/ports/stm32/machine_adc.c
+++ b/ports/stm32/machine_adc.c
@@ -353,8 +353,10 @@ STATIC void adc_config_channel(ADC_TypeDef *adc, uint32_t channel, uint32_t samp
         adc_common->CCR |= ADC_CCR_VBATEN;
         #endif
     }
-    #if defined(STM32G4)
-    channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
+    #if defined(STM32G4) || defined(STM32H5)
+    if (__LL_ADC_IS_CHANNEL_INTERNAL(channel)) {
+        channel = __LL_ADC_CHANNEL_TO_DECIMAL_NB(channel);
+    }
     adc->DIFSEL &= ~(1 << channel); // Set channel to Single-ended.
     #endif
     adc->SQR1 = (channel & 0x1f) << ADC_SQR1_SQ1_Pos | (1 - 1) << ADC_SQR1_L_Pos;
@@ -500,6 +502,13 @@ STATIC mp_obj_t machine_adc_make_new(const mp_obj_type_t *type, size_t n_args, s
 
     machine_adc_obj_t *o = mp_obj_malloc(machine_adc_obj_t, &machine_adc_type);
     o->adc = adc;
+    #if defined(STM32G4) || defined(STM32H5)
+    // This is the workaround for MCUs that use bit 31 to define the internal sensor channel
+    // because MP_ROM_INT() handles only signed 31 bit integers.
+    if (channel > 19) {
+        channel <<= 16;
+    }
+    #endif
     o->channel = channel;
     o->sample_time = sample_time;
 
@@ -517,6 +526,17 @@ STATIC const mp_rom_map_elem_t machine_adc_locals_dict_table[] = {
     { MP_ROM_QSTR(MP_QSTR_read_u16), MP_ROM_PTR(&machine_adc_read_u16_obj) },
 
     { MP_ROM_QSTR(MP_QSTR_VREF), MP_ROM_INT(ADC_CHANNEL_VREF) },
+    #if defined(STM32G4) || defined(STM32H5)
+    // This is the workaround for MCUs that use bit 31 to define the internal sensor channel
+    // because MP_ROM_INT() handles only signed 31 bit integers.
+    { MP_ROM_QSTR(MP_QSTR_CORE_VREF), MP_ROM_INT(ADC_CHANNEL_VREFINT >> 16) },
+    #if defined(STM32G4)
+    { MP_ROM_QSTR(MP_QSTR_CORE_TEMP), MP_ROM_INT(ADC_CHANNEL_TEMPSENSOR_ADC1 >> 16) },
+    #else
+    { MP_ROM_QSTR(MP_QSTR_CORE_TEMP), MP_ROM_INT(ADC_CHANNEL_TEMPSENSOR >> 16) },
+    #endif
+    { MP_ROM_QSTR(MP_QSTR_CORE_VBAT), MP_ROM_INT(ADC_CHANNEL_VBAT >> 16) },
+    #else
     { MP_ROM_QSTR(MP_QSTR_CORE_VREF), MP_ROM_INT(ADC_CHANNEL_VREFINT) },
     #if defined(STM32G4)
     { MP_ROM_QSTR(MP_QSTR_CORE_TEMP), MP_ROM_INT(ADC_CHANNEL_TEMPSENSOR_ADC1) },
@@ -526,6 +546,7 @@ STATIC const mp_rom_map_elem_t machine_adc_locals_dict_table[] = {
     #if defined(ADC_CHANNEL_VBAT)
     { MP_ROM_QSTR(MP_QSTR_CORE_VBAT), MP_ROM_INT(ADC_CHANNEL_VBAT) },
     #endif
+    #endif
 };
 STATIC MP_DEFINE_CONST_DICT(machine_adc_locals_dict, machine_adc_locals_dict_table);
```

This is just the workaround for machine.ADC and there may be a better fix for this issue...

renestraub, 2023-07-30:
@yn386 I would go for a solution that decouples the "magic" channels numbers used in machine_adc.c from the underlying STM32 driver library.

How about defining an enum for internal channels and convert to the STM32 channel literals when needed (i.e. when configuring the channel). This way we have full control over the used values and don't depend on the STM32 drivers. We can then also choose more meaningful values, instead of the odd numbers resulting from the bitfield encoding. As an added benefit it removes the difference of the G4s ADC_CHANNEL_TEMPSENSOR_**ADC1** definition from other MCUs ADC_CHANNEL_TEMPSENSOR.

Something like this.

``` C
typedef enum {
    MACHINE_ADC_CH_FIRST = 256,   // Marker for beginning of internal channel values

    MACHINE_ADC_CH_VREF = MACHINE_ADC_CH_FIRST,
    MACHINE_ADC_CH_VREFINT,
    MACHINE_ADC_CH_TEMP_SENSOR,
    MACHINE_ADC_CH_VBAT,

    MACHINE_ADC_CH_MAX   // Last internal channel entry + 1
} machine_adc_channel_t;
```

yn386, 2023-07-30:
@renestraub
It looks a better fix!
Also, there is no need to change other implementation by checking whether the conversion is needed or not at `machine_adc_make_new`.

renestraub, 2023-08-02:
@yn386 I updated machine_adc.c to work with the STM32H5. Would you mind to have a look at my changes. If you agree I'll create a PR. It would be perfect if you have a STM32G4 target to check for regressions.

https://github.com/micropython/micropython/compare/master...renestraub:mp-stm32h5-adc:stm32-h5-machine-adc?diff=split

yn386, 2023-08-04:
@renestraub

Sorry for the delay response. I checked with NUCLEO-G474RE.
`adc_ll_channel` seems to be incorrect when `MACHINE_ADC_INT_CH_TEMPSENSOR` was specified with STM32G4 MCUs.

https://github.com/renestraub/mp-stm32h5-adc/blob/d3f7cc4a4a75239c70ffcf4b26be51c9877cb25d/ports/stm32/machine_adc.c#L125-L131

should be
```patch
--- a/ports/stm32/machine_adc.c
+++ b/ports/stm32/machine_adc.c
@@ -127,8 +127,8 @@ STATIC uint32_t adc_ll_channel(uint32_t channel_id) {
             adc_ll_ch = ADC_CHANNEL_TEMPSENSOR_ADC1;
             #else
             adc_ll_ch = ADC_CHANNEL_TEMPSENSOR;
-            break;
             #endif
+            break;
         #if defined(ADC_CHANNEL_VBAT)
         case MACHINE_ADC_INT_CH_VBAT:
             adc_ll_ch = ADC_CHANNEL_VBAT;
```
Other channels worked.

renestraub, 2023-08-04:
@yn386, that stupid bug should not have happened. Thanks for capturing it. I updated the PR accordingly.