micropython/micropython #12049: os.uname() dont work on unix

CardealRusso, 2023-07-20:
what is wrong?

dlech, 2023-07-20:
What did you try? Which version of MicroPython are you using? What error did you get?

CardealRusso, 2023-07-20:
> What did you try? Which version of MicroPython are you using? What error did you get?

This version https://github.com/micropython/micropython/releases/tag/v1.20.0
On x64 bit linux

```
MicroPython v1.20.0 on 2023-07-20; linux [GCC 12.2.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import os
>>> os.uname()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'uname'
>>> os
<module 'uos'>
```

dpgeorge, 2023-07-21:
`os.uname()` is not implemented on the unix port of MicroPython.

Please try using `sys.version` and `sys.implementation` instead.  They should have the information you need.

CardealRusso, 2023-07-21:
> `os.uname()` is not implemented on the unix port of MicroPython.
> 
> Please try using `sys.version` and `sys.implementation` instead. They should have the information you need.

Hello, I need to get system architecture similar to "uname -m" (i686, x86_64, arm etc). Since it seems impossible (or at least I couldn't efficiently) run a shell command and save the output of the command to a variable, what alternative do you recommend?

jimmo, 2023-07-21:
See https://github.com/micropython/micropython/pull/12056 to add the `platform` module to more ports.

CardealRusso, 2023-07-21:
As a temporary workaround, I'm getting the /proc/sys/kernel/arch file that returns me the value I want, but I wonder if this is correct.

mattytrentini, 2023-07-24:
Now that #12056 is merged, you can use platform to query something that may be useful:

```python
MicroPython v1.20.0-320-g975a68744-dirty on 2023-07-24; linux [GCC 12.1.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import platform
>>> platform.platform()
'Linux-1.20.0-x86_64--with-glibc2.31'
>>>
```

closed 2023-07-24 by CardealRusso