micropython/micropython #12053: Event.wait() exits instead of wait

stephanelsmith, 2023-07-21:
I'm seeing in certain configurations, awaiting Event().wait() causes a silent exit.  Minimally, I would expect this bit of code to hang. On Python3 it does, but on micropython it exits without message on both unix and esp32 ports. Weird, eh?

```
import asyncio
from asyncio import Event

async def main():
    print('start')
    event = Event()
    await event.wait()
    print('done')

asyncio.run(main())
```

jimmo, 2023-07-25:
@stephanelsmith You're right this should hang, but it's triggering an edge case where there are no runnable tasks and no tasks blocked on I/O. So the loop is just terminating with no more work to do.

It should be straightforward to at least detect this case, but no real program should ever do this. If the event is being set by something outside of asyncio (e.g. an IRQ or scheduler) then it should be using ThreadSafeFlag instead of Event, which will correctly handle this.

stephanelsmith, 2023-07-29:
Yes, I agree this is an edge case.  I came up with it when writing from test cases for the async queue task_done/join feature.

This seems like a reasonable no fix.  Will close!

closed 2023-07-29 by stephanelsmith