micropython/micropython #12083: Hbridge - motor control

IhorNehrutsa, 2023-07-24:
This means that Hbridge is built from discrete components and requires two pairs of complementary PWMs to drive it (ie. 4 PWM outputs).
[image]
**Note: H-bridges integrated circuits not discussed in this issue!**
One Hbridge allows connect one DC motor:
[image] 
Two Hbridges allows connect one stepper motor:
[image]
Three half-Hbridges allows connect one BLDC motor:
[image]

A minimal API is:
```
motor = Hbridge(a=Pin(), _a=Pin(), b=Pin(), _b=Pin(), c=Pin(), _c=Pin(), freq=, duty=, deadtime=)
```
a, _a - required, complementary PWM outputs (usually center-aligned with max deadtime as default)
b, _b - optional, complementary PWM outputs
c, _c - optional, complementary PWM outputs
freq - Hz, frequency for all PWM's
duty - load current - 0..100%
deadtime - Dead-time is a blanking time period (upper & lower transistors in off-state simultaneously) of half-bridge power stage. Both transistors are switched off for short period of time to prevent both transistors conducting simultaneously thus causing a short circuit from supply to ground.
```
motor.duty() change load current
```
```
motor.deinit() close all switches, free internal modules
```

The ESP32 port allows to use MCPWM to build an Hbridge.

rkompass, 2023-07-24:
Hello Ihor,

I was missing some feedback when I posted my code contributions on the older [motor driver discussion](https://github.com/orgs/micropython/discussions/10983). (Feeling floating; was it too much / too fast / too complicated or the opposite?)
Apart from that I learnt that it is important for a qualified discussion of PWM to be able to have the main use cases in mind. The availability of a PWM invert argument and defined states at duty = 0 and duty = 65536 for example helped to reduce the motor driver code and avoid permanent re-inits.

PWM and motor drivers are closely connected. Nice that you bring stepper and BLDC motors into discussion.
With the ordinary DC motor driver we required only one PWM channel and one other pin for direction, or two PWM channels, but only one active at a time. No synchronicity needed.

With stepper and BLDC motor coordination synchronous action of the pwm channels is necessary if I understand it correctly.

How about new names for these? Perhaps like W-Bridge, 2H-Bridge, SynPWM or MultiPWM?

Given that the single channels are a kind of PWM, can you define the synchronicity requirements?

jonnor, 2024-07-13:
There seems to be a very good discussion thread going in https://github.com/orgs/micropython/discussions/10983
So I am suggesting to close this issue in favor of that one.

projectgus, 2024-08-27:
I think I agree that the discussion is the best place to continue this.

@IhorNehrutsa are you still interested in this, and what outcome were you hoping for?

I can see that it might be very useful to have some motor control packages in micropython-lib (similar to your https://github.com/micropython/micropython-lib/pull/720) - but in the case of micros with hardware motor driver peripherals like ESP32 MCPWM then some lower level support might be useful. Is ESP32 the context for this enhancement suggestion?

jonnor, 2024-09-01:
Closed in favor of discussion: https://github.com/orgs/micropython/discussions/10983 - please provide follow up comments there :)

closed 2024-09-01 by jonnor