micropython/micropython #12084: PWM: Reduce inconsistencies between ports.

IhorNehrutsa, 2023-07-24:
Let me continue the discussion that was started here: **PWM: Reduce inconsitencies between ports.**#10817
I express my deepest respect for all who have contributed to bringing the API to a unified form, especially @robert-hh.
I consider this discussion unfinished.
I suggest minimal and clear API (only 3 states are possible for the Pin() output: low, high, pwm wave):
```
pin = Pin(4)
pwm = PWM(pin, freq=5000, duty_u16=32768 | duty_ns=1000 , invert=0)
```
Create a PWM object, initialize it internal. Setting the PIN to PWM mode at that moment, PWM starts immediately.
pin - required
freq - Hz, required (5000 default if omitted)
duty_u16 or duty_ns - required( 50% default if omitted), both duty_u16 and duty_ns args are unacceptable
invert - optional, default is 0

pwm.Init() can be thrown away to drop code size.
```
freq = pwm.freq()         # get current frequency
pwm.freq(1000)            # set PWM frequency from 1Hz to MCU hardware limit as fast as possible
```

```
duty_u16 = pwm.duty_u16() # get current duty cycle, range 0-65536
pwm.duty_u16(2**16*3//4)  # set duty cycle from 0 to 65536 as a ratio duty_u16/65536 as fast as possible, (now 75%)
pwm.duty_u16(0)  # set output to low, if invert=0 and to high, if invert=1
pwm.duty_u16(2**16) #  set output to high, if invert=0 and to low, if invert=1
```

```
pwm.deinit() # free internal modules, Pin state is undefined !!! 
```

I find the start()/stop() methods unnecessary.
I consider it unnecessary inside the deinit() to set PIN outputs as predefined(IN, OUT, PULL_UP, PULL_DOWN, OPEN COLLECTOR, DAC etc) or as they were before PWM initialization.

All other features that the MCU hardware allows (complementary PWM,  align, deadtime etc) please move to **Hbridge - motor control** #12083.

IhorNehrutsa, 2023-07-24:
TODO in ports:
- ESP2866:

1. Add invert arg by pwm_set_channel_invert()
2. Correct duty cycle range to 0-65536

robert-hh, 2023-07-24:
Thanks for the pointer. Indeed, the value range could be limited. ATM it accepts any value, but out-of-band values work as one could assume. Values < 0 behave like 0, and values > 65536 like 65536.
I did intentionally not add the invert method. Could be done. But the esp8266's PWM is anyhow very limited, so it seemed not worth the effort.

rkompass, 2023-07-24:
> I did intentionally not add the invert method. Could be done. But the esp8266's PWM is anyhow very limited, so it seemed not worth the effort.

It would allow a motor driver to be board-agnostic if it was there. But if it's to complicated: One could try that argument and catch the exception..

robert-hh, 2024-11-04:
Sorry. I'm not willing to continue this discussion and testing PR 10854 over and over again. Besides the start-up behavior, it is an excellent contribution and compatible with the other ports.

IhorNehrutsa, 2024-11-04:
@robert-hh Sorry, I didn't mean to offend you.

yoann-darche, 2024-11-04:
From my perspective, after working with the ESP32 port, here is my position:

When the user writes ```p = PWM(Pin(2))```, he expects to see some activity on Pin 2 that resembles a PWM signal. Since neither frequency nor duty cycle was specified, MicroPython assigns default values for these parameters.

If the user wants to update the parameters, he can use ```p = PWM(Pin(2), freq=500)```, which sets the frequency to 500 Hz on Pin 2.

This works because the GPIO pin is the _ineternal ID_, not the Python object itself. Reusing the same pin in code simply updates the parameters associated with that pin.

robert-hh, 2024-11-04:
> Sorry, I didn't mean to offend you.

Do not worry. I'm just tired about this discussion. I think all is said, but maybe not by everyone. It's up to Damien to decide, which way to go. 
Besides that, Damien (kind of) asked me to change back the duty_u16() value, at which the output is at constant high level to 65535. That is in #15153. You may adapt the code in your PR #10854. Maybe it's just setting the cap value. Since ATM it generates proper pulses for duty_16(65535), duty_u16(1) and duty_u16(32768), the setting for duty_u16(65535) will be lost.
Actually this device has the best resolution for PWM with just 500ps difference between high and low pulse at duty_u16(32768) and 1 kHz frequency.

IhorNehrutsa, 2024-11-04:
@robert-hh Ok. I have something to say about the **duty_u16 0-65535 vs 0-65536**, but after decision Question 1).

robert-hh, 2024-11-07:
Tests of this PR against the script from PR #16147 improve. With the most recent push the special if.. for ESP32 is not needed any more. It is now possible to specify freq and duty_u16 separately. Intermittent problems exist still with duty_16 == 0 and 65535. With duty_u16 == 65536 the result is even worse.

Test platform: MicroPython v1.25.0-preview.20.g9d9471f65 on 2024-11-07; Generic ESP32 module with ESP32

Max duty 65535:
```
freq=50    duty_u16=0     : True True : 0 True
freq=50    duty_u16=6553  : True True : 0 True 1 True
freq=50    duty_u16=16383 : True True : 0 True 1 True
freq=50    duty_u16=32767 : True True : 0 True 1 True
freq=50    duty_u16=49151 : True True : 0 True 1 True
freq=50    duty_u16=58981 : True True : 0 True 1 True
freq=50    duty_u16=65535 : True True :        1 False
freq=100   duty_u16=0     : True True : 0 True
freq=100   duty_u16=6553  : True True : 0 True 1 True
freq=100   duty_u16=16383 : True True : 0 True 1 True
freq=100   duty_u16=32767 : True True : 0 True 1 True
freq=100   duty_u16=49151 : True True : 0 True 1 True
freq=100   duty_u16=58981 : True True : 0 True 1 True
freq=100   duty_u16=65535 : True True :        1 False
freq=500   duty_u16=0     : True True : 0 True
freq=500   duty_u16=6553  : True True : 0 True 1 True
freq=500   duty_u16=16383 : True True : 0 True 1 True
freq=500   duty_u16=32767 : True True : 0 True 1 True
freq=500   duty_u16=49151 : True True : 0 True 1 True
freq=500   duty_u16=58981 : True True : 0 True 1 True
freq=500   duty_u16=65535 : True True :        1 True
freq=1000  duty_u16=0     : True True : 0 True
freq=1000  duty_u16=6553  : True True : 0 True 1 True
freq=1000  duty_u16=16383 : True True : 0 True 1 True
freq=1000  duty_u16=32767 : True True : 0 True 1 True
freq=1000  duty_u16=49151 : True True : 0 True 1 True
freq=1000  duty_u16=58981 : True True : 0 True 1 True
freq=1000  duty_u16=65535 : True True :        1 True
freq=2000  duty_u16=0     : True True : 0 True
freq=2000  duty_u16=6553  : True True : 0 True 1 True
freq=2000  duty_u16=16383 : True True : 0 True 1 True
freq=2000  duty_u16=32767 : True True : 0 True 1 True
freq=2000  duty_u16=49151 : True True : 0 True 1 True
freq=2000  duty_u16=58981 : True True : 0 True 1 True
freq=2000  duty_u16=65535 : True True :        1 True
freq=5000  duty_u16=0     : True True : 0 True
freq=5000  duty_u16=6553  : True True : 0 True 1 True
freq=5000  duty_u16=16383 : True True : 0 True 1 True
freq=5000  duty_u16=32767 : True True : 0 True 1 True
freq=5000  duty_u16=49151 : True True : 0 True 1 True
freq=5000  duty_u16=58981 : True True : 0 True 1 True
freq=5000  duty_u16=65535 : True True :        1 True
freq=10000 duty_u16=0     : True (0, 65520) : 0 False
freq=10000 duty_u16=6553  : True True : 0 True 1 True
freq=10000 duty_u16=16383 : True True : 0 True 1 True
freq=10000 duty_u16=32767 : True True : 0 True 1 True
freq=10000 duty_u16=49151 : True True : 0 True 1 True
freq=10000 duty_u16=58981 : True True : 0 True 1 True
freq=10000 duty_u16=65535 : True True :        1 True
```

Max duty 65536:
```
freq=50    duty_u16=0     : True True : 0 True
freq=50    duty_u16=6553  : True True : 0 True 1 True
freq=50    duty_u16=16384 : True True : 0 True 1 True
freq=50    duty_u16=32768 : True True : 0 True 1 True
freq=50    duty_u16=49152 : True True : 0 True 1 True
freq=50    duty_u16=58982 : True True : 0 True 1 True
freq=50    duty_u16=65536 : True True : 0 True 1 (20000, -2)
freq=100   duty_u16=0     : True True : 0 True
freq=100   duty_u16=6553  : True True : 0 True 1 True
freq=100   duty_u16=16384 : True True : 0 True 1 True
freq=100   duty_u16=32768 : True True : 0 True 1 True
freq=100   duty_u16=49152 : True True : 0 True 1 True
freq=100   duty_u16=58982 : True True : 0 True 1 True
freq=100   duty_u16=65536 : True True : 0 True 1 (10000, -1)
freq=500   duty_u16=0     : True True : 0 True
freq=500   duty_u16=6553  : True True : 0 True 1 True
freq=500   duty_u16=16384 : True True : 0 True 1 True
freq=500   duty_u16=32768 : True True : 0 True 1 True
freq=500   duty_u16=49152 : True True : 0 True 1 True
freq=500   duty_u16=58982 : True True : 0 True 1 True
freq=500   duty_u16=65536 : True True : 0 True 1 (2000, -1)
freq=1000  duty_u16=0     : True True : 0 True
freq=1000  duty_u16=6553  : True True : 0 True 1 True
freq=1000  duty_u16=16384 : True True : 0 True 1 True
freq=1000  duty_u16=32768 : True True : 0 True 1 True
freq=1000  duty_u16=49152 : True True : 0 True 1 True
freq=1000  duty_u16=58982 : True True : 0 True 1 True
freq=1000  duty_u16=65536 : True True : 0 True 1 (1000, -1)
freq=2000  duty_u16=0     : True True : 0 True
freq=2000  duty_u16=6553  : True True : 0 True 1 True
freq=2000  duty_u16=16384 : True True : 0 True 1 True
freq=2000  duty_u16=32768 : True True : 0 True 1 True
freq=2000  duty_u16=49152 : True True : 0 True 1 True
freq=2000  duty_u16=58982 : True True : 0 True 1 True
freq=2000  duty_u16=65536 : True True : 0 True 1 (500, -1)
freq=5000  duty_u16=0     : True True : 0 True
freq=5000  duty_u16=6553  : True True : 0 True 1 True
freq=5000  duty_u16=16384 : True True : 0 True 1 True
freq=5000  duty_u16=32768 : True True : 0 True 1 True
freq=5000  duty_u16=49152 : True True : 0 True 1 True
freq=5000  duty_u16=58982 : True True : 0 True 1 True
freq=5000  duty_u16=65536 : True True : 0 True 1 (200, -1)
freq=10000 duty_u16=0     : True True : 0 True
freq=10000 duty_u16=6553  : True True : 0 True 1 True
freq=10000 duty_u16=16384 : True True : 0 True 1 True
freq=10000 duty_u16=32768 : True True : 0 True 1 True
freq=10000 duty_u16=49152 : True True : 0 True 1 True
freq=10000 duty_u16=58982 : True True : 0 True 1 True
freq=10000 duty_u16=65536 : True True : 0 True 1 (100, -1)
```

robert-hh, 2024-11-07:
Indeed PR #10854.

robert-hh, 2024-11-07:
> New PWM parameters take effect in the next PWM cycle.

That is a common behavior but adding this sleep does not change the results. Even time.sleep(10/freq) does not pass all test cases. You can run the test yourself.

IhorNehrutsa, 2025-02-01:
Closed in favor **esp32/PWM: Reduce inconsitencies between ports.** #10854

closed 2025-02-01 by IhorNehrutsa