micropython/micropython #12097: Unix port using gcc with clang flags

brettcannon, 2023-07-26:
I'm trying to build the Unix port, but I keep getting `gcc: error: unrecognized command-line option ‘-Qunused-arguments’`.

Looking at online docs, it looks like [clang has that option](https://clang.llvm.org/docs/ClangCommandLineReference.html#cmdoption-clang-qunused-arguments), but [gcc does not](https://gcc.gnu.org/onlinedocs/gcc/Option-Index.html).

I have `CC` set to `clang`(both globally and when I run `make`, i.e., `CC=clang make`), but the Makefile seems to insist on using gcc and so I can't work around this by trying to use the appropriate compiler. https://github.com/micropython/micropython/blob/master/ports/unix/README.md also explicitly calls out gcc as the compiler to use.

brettcannon, 2023-07-26:
Full output:
```sh
❯ CC=clang V=1 make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make -C /home/brettcannon/Repositories/micropython/micropython/mpy-cross
make[1]: Entering directory '/home/brettcannon/Repositories/micropython/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-standard/genhdr/qstr.i.last
GEN build/genhdr/qstr.i.last
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
Traceback (most recent call last):
  File "/home/brettcannon/Repositories/micropython/micropython/ports/unix/../../py/makeqstrdefs.py", line 211, in <module>
    preprocess()
  File "/home/brettcannon/Repositories/micropython/micropython/ports/unix/../../py/makeqstrdefs.py", line 76, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/multiprocessing/pool.py", line 873, in next
    raise value
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/brettcannon/Repositories/micropython/micropython/ports/unix/../../py/makeqstrdefs.py", line 60, in run
    return subprocess.check_output(args.pp + flags + files)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['gcc', '-E', '-Wno-unused-value', '-Wno-empty-body', '-Qunused-arguments', '-Wno-deprecated-declarations', '-DMICROPY_VFS_FAT=1', '-DMICROPY_VFS_LFS1=1', '-DMICROPY_VFS_LFS2=1', '-DMICROPY_PY_SSL=1', '-DMBEDTLS_CONFIG_FILE="mbedtls/mbedtls_config.h"', '-DMICROPY_SSL_MBEDTLS=1', '-I../../lib/mbedtls/include', '-DMICROPY_PY_BTREE=1', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DLFS1_NO_MALLOC', '-DLFS1_NO_DEBUG', '-DLFS1_NO_WARN', '-DLFS1_NO_ERROR', '-DLFS1_NO_ASSERT', '-DLFS2_NO_MALLOC', '-DLFS2_NO_DEBUG', '-DLFS2_NO_WARN', '-DLFS2_NO_ERROR', '-DLFS2_NO_ASSERT', '-I../../lib/berkeley-db-1.xx/PORT/include', '-I.', '-I../..', '-Ibuild-standard', '-I../../shared/readline', '-Wall', '-Werror', '-Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-Wdouble-promotion', '-Wfloat-conversion', '-std=gnu99', '-DUNIX', '-Os', '-DNDEBUG', '-fdata-sections', '-ffunction-sections', '-Ivariants/standard', '-g', '-U', '_FORTIFY_SOURCE', '-DMICROPY_USE_READLINE=1', '-DMICROPY_PY_TERMIOS=1', '-DMICROPY_PY_SOCKET=1', '-DMICROPY_PY_THREAD=1', '-DMICROPY_PY_THREAD_GIL=0', '-I/home/linuxbrew/.linuxbrew/Cellar/libffi/3.4.4/include', '-DMICROPY_PY_FFI=1', '-DMPZ_DIG_SIZE=16', '-DMICROPY_ROM_TEXT_COMPRESSION=1', '-DNO_QSTR', '../../py/mpstate.c', '../../py/malloc.c', '../../py/gc.c', '../../py/pystack.c', '../../py/qstr.c', '../../py/vstr.c', '../../py/mpprint.c', '../../py/unicode.c', '../../py/mpz.c', '../../py/reader.c', '../../py/lexer.c', '../../py/parse.c', '../../py/scope.c', '../../py/compile.c', '../../py/emitcommon.c', '../../py/emitbc.c', '../../py/asmbase.c', '../../py/asmx64.c', '../../py/emitnx64.c', '../../py/asmx86.c', '../../py/emitnx86.c', '../../py/asmthumb.c', '../../py/emitnthumb.c']' returned non-zero exit status 1.
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
make: *** [../../py/mkrules.mk:111: build-standard/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-standard/genhdr/qstr.i.last'
make: *** Waiting for unfinished jobs....
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
Traceback (most recent call last):
  File "/home/brettcannon/Repositories/micropython/micropython/mpy-cross/../py/makeqstrdefs.py", line 211, in <module>
    preprocess()
  File "/home/brettcannon/Repositories/micropython/micropython/mpy-cross/../py/makeqstrdefs.py", line 76, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/multiprocessing/pool.py", line 873, in next
    raise value
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/brettcannon/Repositories/micropython/micropython/mpy-cross/../py/makeqstrdefs.py", line 60, in run
    return subprocess.check_output(args.pp + flags + files)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
[... cut ...]
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
make[1]: *** [../py/mkrules.mk:111: build/genhdr/qstr.i.last] Error 1
make[1]: *** Deleting file 'build/genhdr/qstr.i.last'
make[1]: Leaving directory '/home/brettcannon/Repositories/micropython/micropython/mpy-cross'
make: *** [../../py/mkrules.mk:168: ../../mpy-cross/build/mpy-cross] Error 2
```

brettcannon, 2023-07-26:
Swapping the arguments around gets farther:
```sh
❯ make CC=clang
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make -C /home/brettcannon/Repositories/micropython/micropython/mpy-cross
make[1]: Entering directory '/home/brettcannon/Repositories/micropython/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-standard/genhdr/qstr.split
GEN build-standard/genhdr/moduledefs.split
GEN build-standard/genhdr/root_pointers.split
GEN build-standard/genhdr/compressed.split
GEN build/genhdr/qstr.i.last
In file included from ../py/emitinlinextensa.c:33:
In file included from ../py/emit.h:30:
In file included from ../py/scope.h:30:
In file included from ../py/emitglue.h:30:
In file included from ../py/bc.h:30:
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
In file included from ../py/mpstate.c:27:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
In file included from ../py/objenumerate.c:30:
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
In file included from ../py/emitcommon.c:29:
In file included from ../py/emit.h:30:
In file included from ../py/scope.h:30:
In file included from ../py/emitglue.h:30:
In file included from ../py/bc.h:30:
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 1 error generated.
'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
In file included from ../py/argcheck.c:30:
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
In file included from ../py/objzip.c:31:
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
In file included from ../py/objpolyiter.c:29:
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:32:
../py/mpthread.h:38:10: fatal error: 'mpthreadport.h' file not found
#include <mpthreadport.h>
         ^~~~~~~~~~~~~~~~
In file included from ../py/emitnxtensawin.c:18:
In file included from ../py/emitnative.c:49:
In file included from ../py/emit.h:30:
[... 648 lines cut ...]
Traceback (most recent call last):
  File "/home/brettcannon/Repositories/micropython/micropython/mpy-cross/../py/makeqstrdefs.py", line 211, in <module>
    preprocess()
  File "/home/brettcannon/Repositories/micropython/micropython/mpy-cross/../py/makeqstrdefs.py", line 76, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/multiprocessing/pool.py", line 873, in next
    raise value
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/brettcannon/Repositories/micropython/micropython/mpy-cross/../py/makeqstrdefs.py", line 60, in run
    return subprocess.check_output(args.pp + flags + files)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/python@3.11/3.11.4_1/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['clang', '-E', '-Wno-unused-value', '-Wno-empty-body', '-Qunused-arguments', '-Wno-deprecated-declarations', '-DMICROPY_VFS_FAT=1', '-DMICROPY_VFS_LFS1=1', '-DMICROPY_VFS_LFS2=1', '-DMICROPY_PY_SSL=1', '-DMBEDTLS_CONFIG_FILE="mbedtls/mbedtls_config.h"', '-DMICROPY_SSL_MBEDTLS=1', '-I../../lib/mbedtls/include', '-DMICROPY_PY_BTREE=1', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DLFS1_NO_MALLOC', '-DLFS1_NO_DEBUG', '-DLFS1_NO_WARN', '-DLFS1_NO_ERROR', '-DLFS1_NO_ASSERT', '-DLFS2_NO_MALLOC', '-DLFS2_NO_DEBUG', '-DLFS2_NO_WARN', '-DLFS2_NO_ERROR', '-DLFS2_NO_ASSERT', '-I../../lib/berkeley-db-1.xx/PORT/include', '-I.', '-I../..', '-Ibuild-standard', '-I../../shared/readline', '-Wall', '-Werror', '-Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-Wdouble-promotion', '-Wfloat-conversion', '-std=gnu99', '-DUNIX', '-Os', '-DNDEBUG', '-fdata-sections', '-ffunction-sections', '-Ivariants/standard', '-g', '-U', '_FORTIFY_SOURCE', '-DMICROPY_USE_READLINE=1', '-DMICROPY_PY_TERMIOS=1', '-DMICROPY_PY_SOCKET=1', '-DMICROPY_PY_THREAD=1', '-DMICROPY_PY_THREAD_GIL=0', '-I/home/linuxbrew/.linuxbrew/Cellar/libffi/3.4.4/include', '-DMICROPY_PY_FFI=1', '-DMPZ_DIG_SIZE=16', '-DMICROPY_ROM_TEXT_COMPRESSION=1', '-DMICROPY_QSTR_EXTRA_POOL [... cut ...] c', '../py/unicode.c', '../py/mpz.c', '../py/reader.c', '../py/lexer.c', '../py/parse.c', '../py/scope.c', '../py/compile.c']' returned non-zero exit status 1.
make[1]: *** [../py/mkrules.mk:111: build/genhdr/qstr.i.last] Error 1
make[1]: *** Deleting file 'build/genhdr/qstr.i.last'
make[1]: Leaving directory '/home/brettcannon/Repositories/micropython/micropython/mpy-cross'
make: *** [../../py/mkrules.mk:168: ../../mpy-cross/build/mpy-cross] Error 2
make: *** Waiting for unfinished jobs....
```

dlech, 2023-07-26:
Maybe you need to manually build `mpy-cross` first?

    make -C mpy-cross/ CC=clang
    make -C port/unix CC=clang

> and `$CC` is still set to `clang` in my shell, so I'm surprised I have to manually specify this

In `py/mkenv.mk` we have `CC = ...`. This would need to be changed to `CC ?= ...` to allow the environment variable to override the default.

brettcannon, 2023-07-27:
> Maybe you need to manually build `mpy-cross` first?

That did it!

> You might also want to delete all of the `build` directories first to make sure cruft isn't left behind from the bad build attempts.

I was running `make clean` constantly. Does that not take care of it? I manually deleted the `mpy-cross/build` directory after reading this.

dlech, 2023-07-27:
Yeah, `make -C ... clean` should do just that.

We've gone back and forth over the years on automatically compiling `mpy-cross` for other targets when it hasn't been built yet, but I think there is some room for improvement still. For example, in the Linux kernel, there is `HOST_CC` separate from `CC` that is used to build tools for the host architecture no mater the target architecture, so maybe we could implement something like that.