micropython/micropython #12101: Latest github release  deflate.DeflateIO.write  Method does not exist

joyanhui, 2023-07-27:
Latest github release
```
    stream = io.BytesIO()
    stream.write(msgpack_data)
    with deflate.DeflateIO(stream, deflate.GZIP) as d:
        d.write(msgpack_data) 
```
DeflateIO.write  Method does not exist

dpgeorge, 2023-07-27:
Compression is not enabled by default on the esp32 port.  If you can build firmware yourself then, to enable compression, you need to change the config to use:
```
#define MICROPY_CONFIG_ROM_LEVEL (MICROPY_CONFIG_ROM_LEVEL_FULL_FEATURES)
```

joyanhui, 2023-07-27:
> Compression is not enabled by default on the esp32 port. If you can build firmware yourself then, to enable compression, you need to change the config to use:
> 
> ```
> #define MICROPY_CONFIG_ROM_LEVEL (MICROPY_CONFIG_ROM_LEVEL_FULL_FEATURES)
> ```

Thanks for your help.

closed 2023-07-27 by joyanhui

jimmo, 2023-07-28:
@joyanhui FYI this is detailed at https://docs.micropython.org/en/latest/library/deflate.html -- we're going to start doing this for all modules (and where applicable, optional methods/classes).