micropython/micropython #12104: Arduino Nano ESP32: deepsleep uses more power than lightsleep

tacker66, 2023-07-27:
With a battery powered Arduino Nano ESP32 (6V at VIN) I get the following power readings (clock speed 240MHz; no other activies running; measuring device is a Voltcraft VC-20 in "mA" mode):

- sleep_ms(10_000): 30mA
- lightsleep(10_000): 2.5 mA
- deepsleep(10_000): 4.5 mA

I would at least expect, that deepsleep uses less power than lightsleep. Additionally the data sheet (https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf) promises 

- 240 uA for lightsleep and
- 7 uA for deepsleep

and the above readings are far away from that.

tacker66, 2023-07-31:
Readings for a ESP32-WROVER (160MHz):

- sleep_ms(10_000): 32.7 mA
- lightsleep(10_000):  8,7 mA
- deepsleep(10_000): 7.6 mA

glenn20, 2023-08-14:
> Additionally the data sheet (https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf) promises
> 
> * 240 uA for lightsleep and
> * 7 uA for deepsleep

Those specs may be for just the esp32-s3 module itself, rather than the board. 

I agree that the first set of measurements are odd. You can see some [equivalent measurements for the FeatherS3](https://github.com/orgs/micropython/discussions/12111#discussioncomment-6585752) using a Nordic Power Profiler Kit 2:

* Normal idle operation (wifi off): **46mA**
* Light sleep:  **1.3mA**
* Deep sleep: **0.043mA**

Are you sure your device is entering deepsleep and that you don't have any thing else connected to the board which might sink current?

tacker66, 2023-08-14:
> 
> Those specs may be for just the esp32-s3 module itself, rather than the board.
>
The data sheet is for the complete board and not only for the esp32-s4 ... but might be some marketing trick ;-) I would have at least expected values somewhere in this range.
> 
> Are you sure your device is entering deepsleep and that you don't have any thing else connected to the board which might sink current?
> 
I can confirm that ... just the board on a breadboard with no peripherals.

peterhinch, 2023-08-15:
@tacker66 Are you sure the device is actually entering deepsleep? In my testing on other ESP32-S3 boards, when the deepsleep ends the chip reboots. If this does not occur, I would query whether something is keeping it awake.

glenn20, 2023-08-16:
> > Those specs may be for just the esp32-s3 module itself, rather than the board.
> 
> The data sheet is for the complete board and not only for the esp32-s3 ... but might be some marketing trick ;-) I would have at least expected values somewhere in this range.

The buck converter on your board will draw at least 200uA in quiescent current - so I can't see how the board could get close to 7uA in deepsleep. Also, from the data sheet it appears the power LED will be drawing current as well.

> > Are you sure your device is entering deepsleep and that you don't have any thing else connected to the board which might sink current?

One thing to note: according to [espressif](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/current-consumption-measurement-modules.html#how-to-choose-an-appropriate-ammeter):

- *...ordinary ammeters have a relatively high internal resistance, resulting in a significant voltage drop. This may cause the module to enter an unstable state, as it is powered by a voltage smaller than the minimum required voltage supply.*

I don't know anything about the voltcraft or how you have setup your tests, but this may be a possible explanation for the device failing to enter deepsleep properly in your test. You meter may not be reliable for measuring deepsleep currents even when the device is entering deepsleep correctly.

tacker66, 2023-08-16:
> @tacker66 Are you sure the device is actually entering deepsleep? 
@peterhinch yes, I see the current readings on the multimeter cycling throug the values stated above for 10s each. After the end of the deepsleep(10_000) the board reboots.

tacker66, 2023-08-16:
> 
> The buck converter on your board will draw at least 200uA in quiescent current - so I can't see how the board could get close to 7uA in deepsleep. Also, from the data sheet it appears the power LED will be drawing current as well.
> 
> I don't know anything about the voltcraft or how you have setup your tests, but this may be a possible explanation for the device failing to enter deepsleep properly in your test. You meter may not be reliable for measuring deepsleep currents even when the device is entering deepsleep correctly.

@glenn20 I am also suspecing something like this. But how could this explain the higher current draw in deepsleep? The Voltcraft is a 50€-class multimeter and at least in uA mode the internal resistance is so high that the board would not run. The circuitry I am using is something like that [How to Measure Current](https://www.electronics-notes.com/articles/test-methods/meters/how-to-measure-current.php).

zalexzperez, 2023-08-18:
> The buck converter on your board will draw at least 200uA in quiescent current - so I can't see how the board could get close to 7uA in deepsleep. Also, from the data sheet it appears the power LED will be drawing current as well.
> 
> > >

Where did you get that 200uA current?
The MP2322's datasheet shows 1-8uA for Quiescent Current vs. Input Voltage graph [on page 6](https://www.monolithicpower.com/en/documentview/productdocument/index/version/2/document_type/Datasheet/lang/en/sku/MP2322/).

tacker66, 2023-08-21:
@peterhinch  Measuring indirectly with a 1Ω series resistor gave the same results. So at least my multimeter seems not so bad ... 
But @glenn20 was right: the power LED is always on and from the data sheet looks like cannot be switched off.
But this IMHO would only explain a higher current in general but not a higher current in deepsleep than in lightsleep.

niftyhacking, 2023-08-21:
Check the mode of all I/O (GPIO) pins they will consume power in some
modes. I have not done the data sheet math. High Z should be lowest. Input
can have an active termination net even if the pin is no connect.
A full pin mode dump should be possible almost easy; entering deep sleep
and awakening, any state.  Idle power regulation is expected to use power.

peterhinch, 2023-08-22:
@tacker66 OK, the measurement is confirmed so it must be a hardware design issue with the board. Some component on the board is drawing more current when the chip is in deepsleep, presumably due to a change in the impedance of a pin on the chip. We could study the schematics to figure out what's going on, but I'm not sure this will help unless you are willing and able to modify the board.

It may be that your simplest practical solution is to buy a board which has been specifically designed for deepsleep and is known to work in that mode such as [this one](https://esp32s3.com/feathers3.html).

iabdalkader, 2023-08-28:
@tacker66 Could you try one of the deepsleep sketches with Arduino's firmware and see if you get similar results ? We should be able to rule out (or confirm) hardware issues easily.

tacker66, 2023-08-28:
@iabdalkader I tried to upload TimeWakeUp.ino but currently get an "No DFU capable USB device available" ... first have to fix that ...

mcwhirtj, 2025-02-02:
I spent several days on this issue before I realized it is both a) known and b) an apparently unresolved issue w/ the Nano ESP32 board.    My test results are generally the  same as others (here and elsewhere) have reported:

After removing the power-on LED and powering the board with a bench power supply, using a dead simple sketch to put the board into sleep mode on a timer wake-up, I get ~1.75mA in "deep sleep" and ~560uA in "light sleep".   The fact that "light sleep" gives a better than 2x lower current than "deep sleep" is a clear red flag that something is not right!   I tried both powering with 8V at Vin, into the on-board MP2322 step-down converter (matching my intended use case of  two 18650 cells),  and also with 3.3V supplied  directly, and saw no significant difference in the results.  The MP2322 has a spec 5uA quiescent current, so this is no surprise to me. 

I was looking at using this board for my IOT project because of it's form factor and the fact that the MP2322 step-down converter would allow me to power directly from two 18650 cells connected in parallel (~8V)  w/o having to supply my own step-down converter.  This really makes the board enticing for battery powered projects.   But I have a 500uA total quiescent current budget for this project, with as much as 200uA allocated to the controller.   In principle the ESP32 should meet that spec, but something in the design of this board is really screwing up the sleep mode use-case.   Really limits its use for battery powered IOT projects.   I'm researching STM32 board now ...